tsduck/tsduck #1434: [BUG] ? PCREXTRACT Plugin Stops Working on TS Stream Interruption and resume

TsAddicted, 2024-02-16:
Hello TSDuck Community,

I am writing to report a bug I've encountered while using the PCREXTRACT plugin in TSDuck. My workflow involves using the plugin to save PTS information from a PID to a CSV file using the following command:

tsp -v -I ip 239.0.0.1:1234 -P pcrextract -p 1024} --pts -o XXX.CSV -O ip 239.0.0.01:123456

The issue arises when the incoming TS stream over UDP is interrupted for a while and then come back. After the interruption, the PCREXTRACT plugin stops writing PTS values to the specified file and does not resume even when the stream is restored. The only workaround I've found so far is to restart the entire TSDuck process, which is not ideal for continuous operations.

Steps to reproduce:

Start extracting PTS information using the PCREXTRACT plugin with the command mentioned above.
Interrupt the incoming TS stream.
Observe that the plugin stops writing to the CSV file and does not resume upon stream restoration.
Expected behavior:
The PCREXTRACT plugin should either continue writing PTS information to the file after the stream is restored or provide a mechanism to recover from such interruptions without requiring a full process restart.

Any insights or workarounds for this issue would be greatly appreciated. Thank you for your assistance and for the great work on TSDuck.

(windows platform)
Best regards,
Jhon

lelegard, 2024-02-16:
Hi @TsAddicted 

When the streams resumes, is this a problem with the CSV output file or the transmission chain? More precisely:
- After the stream resume, is the TS resumed on the output `-O ip`?
- If you try without `-o XXX.CSV` and let `pcrextract` display the values, does the output resume?

BTW, your output address `239.0.0.01:123456` has an invalid port number. And receiving and sending on the same multicast address is a recipe for problems, especially on Windows.

TsAddicted, 2024-02-16:
Hello L
sorry the udp out address was only a placeolder and is not in the real one i m using

After the stream resume, is the TS resumed on the output -O ip?
yes
If you try without -o XXX.CSV and let pcrextract display the values, does the output resume?
i ll test

thx

TsAddicted, 2024-02-19:
hello , after update binary to last version issue disappear, if will appen again i ll inform u
i need to leave the process few days running, is possible in some way to flush the output pcr file every a number of entries or after some time/days ?

thx so much!

lelegard, 2024-02-19:
> after update binary to last version issue disappear

Nothing changed in this plugin.

The problem is somewhere else. Windows is not a good platform to monitor the content of a file while it is open for writing. Linux (or any Unix) is better. Do not expect to be able to see what is in a file while it is processed.

closed 2024-02-21 by TsAddicted