tsduck/tsduck #1440: Regulate plugin stops from working after SRT out re-connection

nangavar, 2024-03-18:
#### Bug description:
The plugin `regulate` works perfect on the first SRT out connection. But if the SRT connection goes down and reconnects (using `--multiple`), then the regulate plugin does not work anymore. I'm using regulate for regulating bitrate` -P regulate -b _value`

#### How to reproduce:

1. Setup a plugin pipe like `tsp -I ip .... -P regulate -b 12000000 -P bitrate_monitor -p 1 -O srt -l 0.0.0.0:port -m`
2. Connect a receiver to the SRT out
3. Stop the SRT reception
4. Wait a few seconds until `bitrate_monitor` reports no bitrate (15 sec or so)
5. Start the SRT reception again
6. On this 2nd attempt `bitrate_monitor` will indicate that `regulate` is not working anymore showing a lot of variation.

#### Expected behavior:
Regulate should keep doing its job no matter what happened downstream on the plugin pipe.

#### Errors and logs:
~~~
No error logs to be added.
~~~

#### Environment:
 - OS: macOS
 - OS version: 14.1.1
 - TSDuck full version: 3.36-3528
 - Installation type: Official (simply `brew install tsduck` two weeks ago)

lelegard, 2024-03-18:
Hi @nangavar 

This is not a bug, this is the normal behaviour. Here is why:

Each plugin is independent and does not interact with the others, does not know what's going on with the other plugins. In your case, the output plugin is blocked during a few seconds. It does not output anything. Because of the flow regulation, packets accumulate in the internal buffer until it saturates (16 MB by default). When the buffer is full, packets no longer pass. They wait for the flow to restart. However, your input plugin is UDP. When the application no longer reads them, UDP datagrams are dropped. This is the basis of UDP. When your output reconnects, the flow restarts. However, the `regulate` plugin has missed a lot of input packets in the meantime. Since the bitrate is computed from the PCR's and number of packets between them, the bitrate computation becomes completely wrong: the plugin sees two distant PCR values with very few packets in the middle (because of the lost input packets). Therefore, the computed bitrate is very low. Consequently, `regulate` now lets packets pass with a low bitrate, blocking incoming packets. And therefore, more UDP datagrams are missed because the flow regulation now blocks (or passes at a very slow bitrate) in the level of the `regulate` plugin. Which makes the bitrate computation even worse. And so on...

The problem is that you should not use the plugin `regulate`. This plugin is designed to regulate an unregulated input. Typically, when reading a file at full disk speed, you need to regulate its output speed if you want to broadcast it.

The UDP input is probably a live multicast which is already at the right playout speed. Why the hell would you need to regulate something which is already regulated at the right speed?

As I wrote many, many times here, never use the `regulate` plugin on an already regulate source. It will not improve the regulation (it is already regulated), it can only make things worse in case of various external effects. This is exactly what happens here.

Do you have a good reason to use `regulate`? Could you explain why, precisely?

You probably simply don't need it. Don't use it.

nangavar, 2024-03-18:
Hi @lelegard 
Thanks a lot for your explanation. It clears a lot to me the behavior. 

The reason Im using regulate is because upstream Im using FFMPEG which is not able to create a real CBR stream. It has some issues regulating the muxrate with the null packets on 0x1fff.
That makes FFMPEG really difficult to be decoded in hardware decoders, since muxrate is not enough stable, using regulate I managed to decode in every hardware decoder that I tested.

FFMPEG --udp--> regulate --srt--> HW decoder.

Apparently TSDUCK is able to control the muxrate output in much more stable way than FFMPEG.

nangavar, 2024-03-18:
You can see here the difference between the muxrate of FFMPEG before and after regulating with `regulate`

[image]

lelegard, 2024-03-19:
OK, in that case, the troublemaker to remove is UDP. Don't use it as a one-to-one local communication, it is not reliable and drops packets in case of problem.

Instead, I recommend to directly pipe `ffmpeg` into `tsp`, something like that:
~~~
ffmpeg ... | tsp -P regulate -b 12000000 -P bitrate_monitor -p 1 -O srt ...
~~~

Make sure to remove any `ffmpeg` option related to output speed regulation, just keep the `mpegts` format. The pipe will reliably deliver packets to `tsp`, stopping and restarting `ffmpeg` when the `tsp` output stops (in case of SRT connection loss for instance).

lars18th, 2024-03-19:
Hi @lelegard ,

> As I wrote many, many times here, never use the regulate plugin on an already regulate source. It will not improve the regulation (it is already regulated), it can only make things worse in case of various external effects. This is exactly what happens here.

This comment is not directly related to this issue, but I want to comment this thing:
- It could be useful if you add a WARNING while running the `regulate` plugin if you detect the the input is a network source or something else _already_ regulated. I feel this warning could help users that doesn't understand the behaviour of this plugin. The name could be quite confusing.

What you think?

lars18th, 2024-03-19:
Hi @nangavar ,

I feel your problem (not pure CBR stream) could have the root cause in a wrong use of the FFmpeg. Please read this:
https://trac.ffmpeg.org/wiki/Encode/H.264#CBRConstantBitRate
And check your use of params `--nal-hrd` and `--filler`.

After done that, the suggestion form @lelegard of using a pipe like: `ffmpeg ... | tsduck ... -O srt -l 0.0.0.0:port -m` could be the best.

lelegard, 2024-03-19:
> It could be useful if you add a WARNING while running the regulate plugin if you detect the the input is a network source or something else already regulated. 

How would you suggest to implement this? A plugin is fully independent and has no knowledge of what other plugins are (including third-party plugins) or what they are doing.

nangavar, 2024-03-19:
Hi @lars18th 
Thank you for your help. I already did that and tested many other things with some interesting findings. I worked a lot over ffmpeg in order to get the best CBR as possible. If you are interested I can share my findings here. Im already using `--nal-hrd ` and tried to get the more stable video bitrate as possible. 
But the main issue for be decoded on HW decoders turned out to be the muxrate stability, and is there were `-muxrate` on ffmpeg is not doing a good job compensating with with null packets on the padding stream 0x1fff. For some reason the final bitrate is not fully stable.

@lelegard 
Thank you for your suggestion, I tested the pipe and it is regulating even better, looks like the UDP packets dropping introduced some sparks on the final bitrate and now is absolutely stable.
I found that if I not regulate with `-muxrate` in ffmpeg the 0x1fff stream is not present on the pipe so `regulate` is not able to work properly, I had to add `-muxrate` in ffmpeg side in order to get the padding stream first. 

Regarding your suggestion it would be great to have a warning or something in regulate command, or it can be included in the manual as well. I arrived to TSDUCK project just because that plugin and then I found an amazing set of tools that are absolutely useful for my job. Im amazed with this project, congratulations guys for your amazing job :D

Attached the results of the regulate at 25M using pipe, no sparks at all 
[image]

nangavar, 2024-03-19:
All of the HW receivers were returning PCR jitter errors. But when I measured the jitter with TSDuck the jitter was within the DBV standard margins (40msec), so that pushes me to think that the problem was that those decoders were using the muxrate to create the time reference, so the jitter on the PCR were because the reference was moving. Is just a theory because stabilizing the muxrate cleared the PCR jitter error.

lelegard, 2024-03-19:
@nangavar, do you need CBR with stuffing in your final stream? This is usually necessary on broadcast streams only, where the modulation technology requires a constant bitrate.

If you work on network streams and only need the non-stuffing packets, you may remove them before output:
~~~
ffmpeg ... | tsp -P regulate -b 12000000 -P bitrate_monitor -p 1 -P filter -n -p 0x1FFF -O srt ...
~~~
The result will be VBR but without jitter.

If the final network segment, after SRT relay, is UDP, you may add `--enforce-burst` (or `-e`) to `-O srt`. To enforce 7 TS packets per UDP datagram (if the SRT relay preserves the granularity).

nangavar, 2024-03-19:
@lelegard 
Yes those HW decoders need stuffing to work properly, most of them are DVB-S satellite receivers with SRT or UDP added licenses. So probably all the TS and decoding logic is designed to work with real CBR streams.
I dont have any issue with any other software or full-network decoders streaming directly from ffmpeg. But until I started using regulate all these HW receivers were not working, stabilizing muxrate was the key.

And thank you for the suggestion, Im using` -O srt -e` option already, I just cleaned a little bit the full command for this post.