tsduck/tsduck #1441: [Q] DTU-238 on Ubuntu - Error attaching input channel 0

LilFoev, 2024-03-22:
Hi All, 

I succesfully used the DTU-238 on Windows as Input device. For example NIT extraction worked well:

`tsp  --verbose -I dektec --device 0 --io-standard demodulator --c2-bandwidth 8-MHz --preload-fifo --receive-timeout 10000 -f 262000000 --symbol-rate 6900000 --modulation 256-QAM -P tables --pid 16 --tid 0x40 --max 1 --xml NIT_dektec.xml -O drop`

Now I wanted to use tsduck on Ubuntu but unfortunately I am getting an error with the DTU-238. 

```
~$ tsdektec -r
* Error attaching input channel 0: DTAPI_E_CONNECT_TO_SERVICE (DTAPI status 4212)
```
I tried different commands, also the NIT extraction I tested succesfully under Windows, but the Error is always the same.

I also have an DTU-315 which works without any issues under Ubuntu, tested a File Playout to QAM:
`tsp -I file ~/Downloads/TS_402_MHz_7900_20240321.ts -O dektec --modulation 256-QAM --device 1 --frequency 402000000 --symbol-rate 6900000`

Do you have any advice to get the DTU-238 to work?

**System Info:**
Operating System: Ubuntu 23.10
Kernel: Linux 6.5.0-26-generic
Architecture: x86-64

**TSDuck Info:**
I installed the packages from https://tsduck.io/download:
dektec-dkms_2024.01.0_all.deb
tsduck_3.37-3670.ubuntu23_amd64.deb

**Device Info:**
```
~$ tsdektec -av

DTAPI: 6.3.0.221, Dtu: 4.15.4.92

* Device 0: DTU-238 (DTU-238 FantASI USB-2 DVB-T/T2 Input)
  Physical ports: 2
  Channels: input: 1, output: 0
  Input 0: DTU-238 port 1 (top), Demodulator (Automatic spectral inversion
      correction., Automatic symbol rate detection., VHF-band 47-470MHz,
      UHF-band 400-862MHz, Port is disabled, Uni-directional input, Transparent
      mode, MPEG-2 transport stream, Demodulation, DVB-C2 reception, DVB-T
      reception, DVB-T2 reception, ISDB-T reception, QAM-A reception)
  Subsystem id: 0x0000 (DTU-238)
  Subsystem vendor id: 0x0000
  Device id: 0x0226
  Vendor id: 0x1297
  Serial number: 00000000FC9AC143
  Firmware version: 0 (0x00000000)
  Firmware variant: 0 (0x00000000)
  USB address: 1
  Customer id: 303839
  Engineering change level: Rev 1
  Production date: 2018.01
  Part number: DTU-238
  Serial number: 4238000451

* Device 1: DTU-215 (DTU-215 USB-2 VHF/UHF Modulator)
  Physical ports: 1
  Channels: input: 0, output: 1
  Output 0: DTU-215 port 1, Modulator (VHF-band 47-470MHz, UHF-band 400-862MHz,
      Uni-directional output, MPEG-2 transport stream, Modulator output, ATSC
      8-VSB modulation, DVB-T modulation, QAM-A modulation, QAM-B modulation,
      QAM-C modulation, Adjustable output level, Continuous wave, Mute RF output
      signal, Internal RF clock reference, Internal TS rate clock reference)
  Subsystem id: 0x0000 (DTU-215)
  Subsystem vendor id: 0x0000
  Device id: 0x020F
  Vendor id: 0x1297
  Serial number: 00000000FB3BDFAE
  Firmware version: 1 (0x00000001)
  Firmware variant: 0 (0x00000000)
  USB address: 6
  Customer id: 303839
  Engineering change level: Rev 3
  Manufacture id: 03
  Production date: 2018.10
  Part number: DTU-215
  Serial number: 4215005102
  Crystal stability: 2ppm
```

Thank you

lelegard, 2024-03-22:
Hi @LilFoev 

This looks like an installation issue of Dektec drivers and services on Linux. 

According to Dektec documentation, DTAPI_E_CONNECT_TO_SERVICE when attaching an input channel means "The DTAPI service is not running".

@EricDeJong, any idea on this?

LilFoev, 2024-03-22:
Hi @lelegard 

thanks for your quick reply. 
Your comment "The DTAPI service is not running" pointed to the solution. 
I did not found the DTAPI service running and was not sure if it is even installed. So I downloaded the SDK from dektec site and installed the DtapiService.bin. After installation the service was running and no more error. Succesfully extraced the NIT with a DTU-238. 

Readme of DtapiService.bin says:
```
The DTAPI service is a Linux daemon (service) that manages DekTec's range of 
demodulator boards. The following demodulators are supported:
- DTU-234	(VSB/QAM-B receiver for USB)
- DTU-235	(DVB-T measurement receiver for USB)
- DTU-236	(QAM measurement receiver + ASI input for USB)
- DTA-2131	(Multi-standard VHF/UHF demodulator for PCI Express)
- DTA-2135	(Dual DVB-T receiver for PCI Express)
- DTA-2136	(Dual QAM-A/B/C receiver for PCI Express)
- DTA-2137	(Dual DVB-S/DVB-S2 receiver for PCI Express)
- DTA-2138	(DVB C2/T2 receiver for PCI Express)
- DTA-2139	(Twelve-channel QAM Receiver for PCI Express)
 
NOTE: this service must be running to be able to use the above boards.
```

I somehow missed the part to install this seperately, thought it is enough to install the dektec-dkms_2024.01.0_all.deb package. 

Thank you very much, I am happy now :)

closed 2024-03-22 by LilFoev