tsduck/tsduck #1443: [Q] Audio and video re-interleaving possible?

hariszukanovic, 2024-03-25:
Using mediainfo tool I can see that in my input TS stream there is a "Delay relative to video: -839 ms"
... which AFAIU means simply that while reading TS packets as they come, that is the difference between the first observed PTS of the video and first observed PTS of the audio track in question.
Can tsduck be used to interleave audio and video packets found in TS input to be closer together so that the audio and video with the similar/close PTSes are outputed next to each other in the output TS stream?

mediainfo output screenshot:
[
[image]
](url)

lelegard, 2024-03-25:
Hi @hariszukanovic 

You need a mux/remux to this and TSDuck is not a MUX. I recommend to use ffmpeg for that kind of operation.

hariszukanovic, 2024-03-29:
To my knowledge ffmpeg is not able to do this. 
If you do know an example command please share to help :)