tsduck/tsduck #1446: [Q] How to generate dbgsym deb pkg?

zerodefect, 2024-03-27:
I've been testing, and I'd like to get a good stacktrace to debug issues. To generate comprehensive debugging symbols in the binaries, I use gcc's `-g` flag:

```
export ADDITIONAL_CXX_FLAGS='-U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=3 -march=x86-64-v3 -flto -g'
make CXXFLAGS_EXTRA="${ADDITIONAL_CXX_FLAGS}" NOTEST=1 NOSTATIC=1 NOGITHUB=1 NODEKTEC=1 NOCURL=1 NOPCSC=1 NOEDITLINE=1 NOSRT=1 NOHIDES=1 NOEDITLINE=1 NORIST=1 NOVATEK=1 installer
```

The problem is that the resultant binaries are significantly larger in comparison (~7MiB vs ~80MiB). Is there any way to strip the binaries similarly to how some package managers will provide a `-dbgsym.deb` ? For the record, I'm not well-versed in the construction of Debian-like packages

Environment:
gcc-13
Ubuntu 23.10

lelegard, 2024-03-27:
Hi @zerodefect 

This is not the best place to ask this kind of question. I don't think that there are many Debian specialists to answer that question here, me included.

I would suggest to ask on one of:

- https://askubuntu.com/
- https://stackoverflow.com/

zerodefect, 2024-03-29:
Thanks Thierry. I appreciate your response.

I'll try one of the links you suggested.

closed 2024-03-29 by zerodefect