tsduck/tsduck #1447: [Q] How does EIT Actual Schedule work?

GeneralTransmission, 2024-04-01:
I'm currently trying to get my head around EIT tables and am attempting to inject an EIT into a new ts using the following;

`-P inject --poll-files --eit-normalization --eit-actual ./eit.xml --pid 0x0012 --bitrate 15000`

The test input EIT is in the following format:

`<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <EIT type="pf" version="5" current="true" actual="true" service_id="0x05D8" transport_stream_id="0x0001" original_network_id="0x0001">
    <event event_id="0x03e9" start_time="2024-04-01 12:00:00" duration="01:00:00" running_status="running" CA_mode="false">
      <short_event_descriptor language_code="eng">
        <event_name>Test Event 1</event_name>
        <text>This is a test event description 1.</text>
      </short_event_descriptor>
      <extended_event_descriptor descriptor_number="0" last_descriptor_number="0" language_code="eng">
        <text>This is a test event extended description 1.</text>
      </extended_event_descriptor>
      <content_descriptor>
        <content content_nibble_level_1="1" content_nibble_level_2="1" user_byte="0x00"/>
      </content_descriptor>
    </event>
    <event event_id="0x03ea" start_time="2024-04-01 13:00:00" duration="01:00:00" running_status="not-running" CA_mode="false">
      <short_event_descriptor language_code="eng">
        <event_name>Test Event 2</event_name>
        <text>This is a test event description 2.</text>
      </short_event_descriptor>
      <extended_event_descriptor descriptor_number="0" last_descriptor_number="0" language_code="eng">
        <text>This is a test event extended description 2.</text>
      </extended_event_descriptor>
      <content_descriptor>
        <content content_nibble_level_1="1" content_nibble_level_2="1" user_byte="0x00"/>
      </content_descriptor>
    </event>
  </EIT>
  <EIT type="80" version="5" current="true" actual="true" service_id="0x05D8" transport_stream_id="0x0001" original_network_id="0x0001">
    <event event_id="0x03e9" start_time="2024-04-01 12:00:00" duration="01:00:00" running_status="running" CA_mode="false">
      <short_event_descriptor language_code="eng">
        <event_name>Test Event 1</event_name>
        <text>This is a test event description 1.</text>
      </short_event_descriptor>
      <extended_event_descriptor descriptor_number="0" last_descriptor_number="0" language_code="eng">
        <text>This is a test event extended description 1.</text>
      </extended_event_descriptor>
      <content_descriptor>
        <content content_nibble_level_1="1" content_nibble_level_2="1" user_byte="0x00"/>
      </content_descriptor>
    </event>
    <event event_id="0x03ea" start_time="2024-04-01 13:00:00" duration="01:00:00" running_status="not-running" CA_mode="false">
      <short_event_descriptor language_code="eng">
        <event_name>Test Event 2</event_name>
        <text>This is a test event description 2.</text>
      </short_event_descriptor>
      <extended_event_descriptor descriptor_number="0" last_descriptor_number="0" language_code="eng">
        <text>This is a test event extended description 2.</text>
      </extended_event_descriptor>
      <content_descriptor>
        <content content_nibble_level_1="1" content_nibble_level_2="1" user_byte="0x00"/>
      </content_descriptor>
    </event>
  </EIT>
</tsduck>`

I then use the following to extract the tables from the captured ts:

`tsp -I file /capture.ts -P tables --pid 18 --binary-output EIT.bin -O drop`
`tstabcomp -d EIT.bin`

I then get the following as output, which doesn't look quite correct:
`<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <EIT type="pf" version="0" current="true" actual="true" service_id="0x05D8" transport_stream_id="0x0001" original_network_id="0x0001" last_table_id="0x4E">
    <event event_id="0x03E9" start_time="2024-04-01 12:00:00" duration="01:00:00" running_status="running" CA_mode="false">
      <short_event_descriptor language_code="eng">
        <event_name>Test Event 1</event_name>
        <text>This is a test event description 1.</text>
      </short_event_descriptor>
      <extended_event_descriptor descriptor_number="0" last_descriptor_number="0" language_code="eng">
        <text>This is a test event extended description 1.</text>
      </extended_event_descriptor>
      <content_descriptor>
        <content content_nibble_level_1="1" content_nibble_level_2="1" user_byte="0x00"/>
      </content_descriptor>
    </event>
    <event event_id="0x03EA" start_time="2024-04-01 13:00:00" duration="01:00:00" running_status="not-running" CA_mode="false">
      <short_event_descriptor language_code="eng">
        <event_name>Test Event 2</event_name>
        <text>This is a test event description 2.</text>
      </short_event_descriptor>
      <extended_event_descriptor descriptor_number="0" last_descriptor_number="0" language_code="eng">
        <text>This is a test event extended description 2.</text>
      </extended_event_descriptor>
      <content_descriptor>
        <content content_nibble_level_1="1" content_nibble_level_2="1" user_byte="0x00"/>
      </content_descriptor>
    </event>
  </EIT>
  <generic_long_table table_id="0xA0" table_id_ext="0x05D8" version="5" current="true" private="true">
    <section>
      00 01 00 01 00 A0 03 E9 EB F1 12 00 00 01 00 00
      80 6E 4D 34 65 6E 67 0C 54 65 73 74 20 45 76 65
      6E 74 20 31 23 54 68 69 73 20 69 73 20 61 20 74
      65 73 74 20 65 76 65 6E 74 20 64 65 73 63 72 69
      70 74 69 6F 6E 20 31 2E 4E 32 00 65 6E 67 00 2C
      54 68 69 73 20 69 73 20 61 20 74 65 73 74 20 65
      76 65 6E 74 20 65 78 74 65 6E 64 65 64 20 64 65
      73 63 72 69 70 74 69 6F 6E 20 31 2E 54 02 11 00
      03 EA EB F1 13 00 00 01 00 00 20 6E 4D 34 65 6E
      67 0C 54 65 73 74 20 45 76 65 6E 74 20 32 23 54
      68 69 73 20 69 73 20 61 20 74 65 73 74 20 65 76
      65 6E 74 20 64 65 73 63 72 69 70 74 69 6F 6E 20
      32 2E 4E 32 00 65 6E 67 00 2C 54 68 69 73 20 69
      73 20 61 20 74 65 73 74 20 65 76 65 6E 74 20 65
      78 74 65 6E 64 65 64 20 64 65 73 63 72 69 70 74
      69 6F 6E 20 32 2E 54 02 11 00
    </section>
  </generic_long_table>
</tsduck>`

The pf table looks OK except for the version number having changed for 5 to 0 (What's the reason for this? Does tsduck always renumber the EIT version with --eit-normalization?).

However instead of the schedule actual EIT table being table_id 0x50, it now appears as a generic_long_table with table_id 0xA0.
Is the "type" in the EIT XML input supposed to be type="80" or type="0x50" or something else for actual schedule?
Or is there something else I'm not doing correctly?

I should add I'm using an older build of tsduck for compatibility  (version 3.35-3215).

GeneralTransmission, 2024-04-01:
Hi @lelegard 

Thanks for your response. I will have a play around with the plugin `eitinject` however having read the description this looks to only inject the EIT on the industry standard pid 18. However i need to be able to inject a new pid with EIT on a non-standard pid as well for testing.

Regarding the current method of using the plugin `inject` see below which i hope should be reproducible.

The tsduck command:
`tsp --verbose --bitrate 38440471 --buffer-size-mb 128 \
-I null \
-P filter --max-payload-size 0 --negate -s \
-P regulate --bitrate 38440471 \
-P merge 'tsp -I ip xxx.xxx.xxx.xxx:yyyy' --incremental-pcr-restamp --no-smoothing \
-P filter --negate -p 1000 -s \
-P inject --poll-files --eit-normalization --eit-actual ./eit.xml --pid 0x0012 --bitrate 15000 -s -f \
-P file ./capture.ts \
-O file ./live.ts`

The input eit.xml file:
`<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <EIT type="pf" version="5" current="true" actual="true" service_id="0x05D8" transport_stream_id="0x0001" original_network_id="0x0001">
    <event event_id="0x03e9" start_time="2024-04-01 12:00:00" duration="01:00:00" running_status="running" CA_mode="false">
      <short_event_descriptor language_code="eng">
        <event_name>Test Event 1</event_name>
        <text>This is a test event description 1.</text>
      </short_event_descriptor>
      <extended_event_descriptor descriptor_number="0" last_descriptor_number="0" language_code="eng">
        <text>This is a test event extended description 1.</text>
      </extended_event_descriptor>
      <content_descriptor>
        <content content_nibble_level_1="1" content_nibble_level_2="1" user_byte="0x00"/>
      </content_descriptor>
    </event>
    <event event_id="0x03ea" start_time="2024-04-01 13:00:00" duration="01:00:00" running_status="not-running" CA_mode="false">
      <short_event_descriptor language_code="eng">
        <event_name>Test Event 2</event_name>
        <text>This is a test event description 2.</text>
      </short_event_descriptor>
      <extended_event_descriptor descriptor_number="0" last_descriptor_number="0" language_code="eng">
        <text>This is a test event extended description 2.</text>
      </extended_event_descriptor>
      <content_descriptor>
        <content content_nibble_level_1="1" content_nibble_level_2="1" user_byte="0x00"/>
      </content_descriptor>
    </event>
  </EIT>
  <EIT type="80" version="5" current="true" actual="true" service_id="0x05D8" transport_stream_id="0x0001" original_network_id="0x0001">
    <event event_id="0x03e9" start_time="2024-04-01 12:00:00" duration="01:00:00" running_status="running" CA_mode="false">
      <short_event_descriptor language_code="eng">
        <event_name>Test Event 1</event_name>
        <text>This is a test event description 1.</text>
      </short_event_descriptor>
      <extended_event_descriptor descriptor_number="0" last_descriptor_number="0" language_code="eng">
        <text>This is a test event extended description 1.</text>
      </extended_event_descriptor>
      <content_descriptor>
        <content content_nibble_level_1="1" content_nibble_level_2="1" user_byte="0x00"/>
      </content_descriptor>
    </event>
    <event event_id="0x03ea" start_time="2024-04-01 13:00:00" duration="01:00:00" running_status="not-running" CA_mode="false">
      <short_event_descriptor language_code="eng">
        <event_name>Test Event 2</event_name>
        <text>This is a test event description 2.</text>
      </short_event_descriptor>
      <extended_event_descriptor descriptor_number="0" last_descriptor_number="0" language_code="eng">
        <text>This is a test event extended description 2.</text>
      </extended_event_descriptor>
      <content_descriptor>
        <content content_nibble_level_1="1" content_nibble_level_2="1" user_byte="0x00"/>
      </content_descriptor>
[... 33 lines cut ...]
      <short_event_descriptor language_code="eng">
        <event_name>Test Event 1</event_name>
        <text>This is a test event description 1.</text>
      </short_event_descriptor>
      <extended_event_descriptor descriptor_number="0" last_descriptor_number="0" language_code="eng">
        <text>This is a test event extended description 1.</text>
      </extended_event_descriptor>
      <content_descriptor>
        <content content_nibble_level_1="1" content_nibble_level_2="1" user_byte="0x00"/>
      </content_descriptor>
    </event>
    <event event_id="0x03EA" start_time="2024-04-01 13:00:00" duration="01:00:00" running_status="not-running" CA_mode="false">
      <short_event_descriptor language_code="eng">
        <event_name>Test Event 2</event_name>
        <text>This is a test event description 2.</text>
      </short_event_descriptor>
      <extended_event_descriptor descriptor_number="0" last_descriptor_number="0" language_code="eng">
        <text>This is a test event extended description 2.</text>
      </extended_event_descriptor>
      <content_descriptor>
        <content content_nibble_level_1="1" content_nibble_level_2="1" user_byte="0x00"/>
      </content_descriptor>
    </event>
  </EIT>
</tsduck>`

lelegard, 2024-04-02:
> only inject the EIT on the industry standard pid 18. However i need to be able to inject a new pid with EIT on a non-standard pid

You can remap the PID using `-P remap` after `-P eitinject`.

Alternatively, I can (and will) add an option to specify another PID.

Concerning the weird section with table id 0xA0, the problem is in your XML file: `<EIT type="80" ...`.

From the TSDuck user's guide:

> D.3.5 Event Information Table (EIT)
>
> Defined by DVB in [5]. This is the DVB version of an EIT. ATSC uses a different structure for EIT.
> If type="pf", this is an EITp/f (present/following).
> If type is a 4-bit integer, this is an EITs (schedule) with TID 0x50 + type (EITs Actual) or 0x60 + type (EITs Other), depending on the actual attribute.

For EIT schedule, the type attribute shall be a value between 0 and 15. It is added to the base table id for EIT schedule actual/other. The base table id for EIT schedule actual is 80. Plus 80 equals 160, or 0x80.

It is a bug in TSDuck to accept a value above 15. Your XML file should have been rejected. I will fix this.

That being said, you can see that it is extremely difficult to generate a consistent EPG using individual EIT's from XML files. In practice, it is even impossible. I repeat: impossible.

The reason is simple. XML files can only contain full tables. The generation of a binary table over multiple sections is automatically done by TSDuck, based on available space in sections. EIT schedule, on the other hand, are not full tables. They are sparsed sections. This is why generating a consistent EPG is not possible from individual EIT structures in a XML file. This is why the plugin `eitinject` exists. The plugin `inject` can only inject full tables, meaning everything except EIT's.

This is why I asked the question: what do you want to do exactly? If you want to do some specific isolated test, `inject` _may_ be fine. If you want to inject a consistent EPG, even for test purpose, you _must_ use `eitinject`, not `inject`. And the logic is different. You do not specify tables to inject. You specify events which will be used to generate the various types of EIT's, depending on the current time.

lelegard, 2024-04-03:
Option `--pid` has been added to plugin `--eitinject` to specify an alternative EIT PID.

closed 2024-09-08 by github-actions[bot]