tsduck/tsduck #1449: How to solve this error my script was stopped after few minutes of running[Q].

udhayNuk, 2024-04-09:
tsp -v -m --synchronous-log -I hls --connection-timeout 300000 --receive-timeout 300000 http://ptu.ridsys.in/riptv/live/stream1/index.m3u8 --save-files /home/user/RIDSYSPACKAGE/riptv/live/stream1/ -O drop

@lelegard 

i have used above script for hls copying and pasting into another server for live streaming.
i runned this script after few minutes the stream was stopped how to solve this error.
please reply me.
thanking you in advance.

* hls: saving input TS to /home/user/RIDSYSPACKAGE/riptv/live/stream1/index-159578.ts
* hls: downloading from http://ptu.ridsys.in/riptv/live/stream1/index-159579.ts
* hls: MIME type: video/mp2t, expected size: 2126656 bytes
* hls: saving input TS to /home/user/RIDSYSPACKAGE/riptv/live/stream1/index-159579.ts
* hls: downloading from http://ptu.ridsys.in/riptv/live/stream1/index-159580.ts
* hls: MIME type: text/html, expected size: 275 bytes
* Warning: hls: MIME type is text/html, maybe not a valid transport stream
* hls: saving input TS to /home/user/RIDSYSPACKAGE/riptv/live/stream1/index-159580.ts
* Error: hls: synchronization lost after 93,293,055 packets, got 0x3C instead of 0x47

lelegard, 2024-04-09:
Hi @udhayNuk 

### Short answer
Use `tsp -I hls --live ...`

### Long answer
Without `--live`, does it fail _immediately_ after one or two segments, or after several minutes? This is a live playlist, a sliding playlist, a text file which is regularly updated with a new list of segments. At each update, the initial segments are removed (and the corresponding .ts file are deleted) and new segments are added at end. By default, the `hls` input plugin assumes VoD (static playlist) or Event (growing playlist). So, it reads all segments from the beginning of the playlist. In the case of a live playlist, it is possible that, in the short interval of time, after the initial playlist is read, and before the plugin analyses the playlist and starts to download the first segment, the playlist is updated and the first segment is deleted. Then, the download of the first segment fails and the returned data is an HTML error message. It can after the first or first few segments.

This is probably what happens here:
~~~
* hls: downloading from http://ptu.ridsys.in/riptv/live/stream1/index-159579.ts
* hls: MIME type: video/mp2t, expected size: 2126656 bytes
...
* hls: downloading from http://ptu.ridsys.in/riptv/live/stream1/index-159580.ts
* hls: MIME type: text/html, expected size: 275 bytes
* Warning: hls: MIME type is text/html, maybe not a valid transport stream
...
* Error: hls: synchronization lost after 93,293,055 packets, got 0x3C instead of 0x47
~~~
The segment 159579 is correctly downloaded and contains TS packets (`MIME type: video/mp2t`) but the next one, 159580, is returned as a short HTML text (`MIME type: text/html, 275 bytes`). Then, of course, `tsp` fails when processing that content because this is not TS data (`synchronization lost after 93,293,055 packets, got 0x3C instead of 0x47`).

The solution is to tell the `hls` plugin that this is a live stream using option `--live`. This means that there is no "beginning" and that the most up-to-date live content is in the last segment. In that case, the plugin starts to read the _last_ segment of the playlist, not the first one. After that segment, it will reload the playlist to discover new segments.

I tried your URL, it is publicly accessible. Without `--live` option, sometimes it works, sometimes it doesn't, depending on wether the first segment was deleted in the meantime or not. Using option `tsp --debug`, we see the returned HTTP headers and the hexa + ASCII content at the point of TS synchronization error:

~~~
$ tsp -d -I hls http://ptu.ridsys.in/riptv/live/stream1/index.m3u8 -O file /dev/null
* Debug: hls: downloading segment http://ptu.ridsys.in/riptv/live/stream1/index-166165.ts
* Debug: hls: curl retries: 0, interval: 0 ms
* Debug: hls: HTTP header: HTTP/1.1 404 Not Found
* Debug: hls: HTTP header: Date: Tue, 09 Apr 2024 09:19:34 GMT
* Debug: hls: HTTP header: Server: Apache/2.4.41 (Ubuntu)
* Debug: hls: HTTP header: Content-Length: 275
* Debug: hls: HTTP header: Content-Type: text/html; charset=iso-8859-1
* hls: downloading from http://ptu.ridsys.in/riptv/live/stream1/index-166165.ts
* hls: MIME type: text/html, expected size: 275 bytes
* Warning: hls: MIME type is text/html, maybe not a valid transport stream
* Error: hls: synchronization lost after 0 packets, got 0x3C instead of 0x47
* Debug: hls: data at loss of synchronization:
    0000:  3C 21 44 4F 43 54 59 50 45 20 48 54 4D 4C 20 50  <!DOCTYPE HTML P
    0010:  55 42 4C 49 43 20 22 2D 2F 2F 49 45 54 46 2F 2F  UBLIC "-//IETF//
    0020:  44 54 44 20 48 54 4D 4C 20 32 2E 30 2F 2F 45 4E  DTD HTML 2.0//EN
    0030:  22 3E 0A 3C 68 74 6D 6C 3E 3C 68 65 61 64 3E 0A  ">.<html><head>.
    0040:  3C 74 69 74 6C 65 3E 34 30 34 20 4E 6F 74 20 46  <title>404 Not F
    0050:  6F 75 6E 64 3C 2F 74 69 74 6C 65 3E 0A 3C 2F 68  ound</title>.</h
    0060:  65 61 64 3E 3C 62 6F 64 79 3E 0A 3C 68 31 3E 4E  ead><body>.<h1>N
    0070:  6F 74 20 46 6F 75 6E 64 3C 2F 68 31 3E 0A 3C 70  ot Found</h1>.<p
    0080:  3E 54 68 65 20 72 65 71 75 65 73 74 65 64 20 55  >The requested U
    0090:  52 4C 20 77 61 73 20 6E 6F 74 20 66 6F 75 6E 64  RL was not found
    00A0:  20 6F 6E 20 74 68 69 73 20 73 65 72 76 65 72 2E   on this server.
    00B0:  3C 2F 70 3E 0A 3C 68 72 3E 0A 3C 61              </p>.<hr>.<a
~~~
We clearly see that the TS file has disappeared from the server, with a "file not found" HTTP 404 error and HTML error message.

This does not happen with option `--live`.

### HLS server configuration error

That being said, the HLS server seems poorly configured. In practice, it is an obvious race condition when 1) the client reads a playlist, 2) the playlist is updated and segments are removed, 3) the client request the first segment, removed from the playlist.

For this reason, the HLS server shall not delete the segment files immediately after removing them from the playlist. If a client downloads a segment shortly after its name is removed from the playlist on the server side, the segment file shall still be there to serve the segment URL.

For instance, in the `hls` output plugin, the option `--live-extra-segments` specifies the amount of "zombie" segments to keep. If value 5 is specified, for instance, the last 5 segment files the name of which were removed from the playlist are still present and available for download. The oldest zombie segment file will be deleted the next time another segment name will be removed from the playlist. And so on.

[... cut ...] s results for Ridsys, a set-top box manufacturing company in Puducherry, India. However, the domain is `ridsys.com`, not `ridsys.in`.

@udhayNuk, could you please clarify your position with respect to the Ridsys company, and the status of `ridsys.in` vs. `ridsys.com`. Without more details, I may be obliged to blacklist requests referring to this site.

If you manage `ridsys.in`, please secure it. Everything is publicly accessible, including shell scripts containing system passwords in the clear.

udhayNuk, 2024-04-09:
@lelegard 
Thanking you for your reply.

udhayNuk, 2024-04-09:
@lelegard 
after adding --live option, same error occured, the stream was stopped.
please solve.

lelegard, 2024-04-09:
@udhayNuk, could you please explain the status of `ridsys.in` vs. `ridsys.com`? And the non-professional structure of `ridsys.in`?

udhayNuk, 2024-04-09:
@lelegard  ridsys.in for testing we have using. still in developing and ridsys.com is our company.
please solve my error.

lelegard, 2024-04-09:
Using your URL, I see the same problem as you without `--live`. With `--live`, I never have it.

However, the server seems a bit slow to respond. Your segments are very short, 2 seconds, and the interval between two downloads is too long, 2 and sometimes 3 seconds. So, the server seems to renew the content slightly faster than it can send it to clients.

1. Use debug mode, post the output of `tsp --timed-log --debug=2 -I hls ...`
2. How many zombie segments does the HLS server keep?

It looks like a server tuning issue: very short segments + not enough zombie segments + slow response time + insufficient output bandwith.

udhayNuk, 2024-04-10:
@lelegard 
Thanking you for replying.
Already i have used  --duration 2 --live 15,
now i have changed segment time and checking, i will check and update
tsp  -v -m --synchronous-log -I ip 233.1.1.1"1111 -P analyze -i 30 -o report.txt  -P regulate --pcr-synchronous  -O hls --duration 5 --live 7 -p index.m3u8 index.ts

lelegard, 2024-04-10:
So, your HLS server is also fed by TSDuck. In that case, add `----live-extra-segments 7' to `-O hls`. You will have as many zombie segments as live segments.

Also monitor the response time and output bandwidth of the associated HTTP server, Apache or other. But this is not a TSDuck issue.

udhayNuk, 2024-04-12:
@lelegard 
how to stream(live udp) one server to another long distance server via internet without any error,dropping and stream does not stop.
please help me to solve this.