tsduck/tsduck #1451: [Q] tsp confilt with vlc in windows server 2019

edwardlai, 2024-04-19:
tsp version 3.37-3670
vlc player 3.0.20
OS : Windows Server 1029 Standard 

tsp & vlc receive a multicast UDP stream together 

tsp process

- tsp --verbose -I ip 224.5.6.7:4567  -l 192.168.1.47 -O drop 

vlc open network stream

- udp://@224.5.6.7:4567

When tsp.exe  started  the vlc corruption  
Even tsp add the --disable-multicast-loop  not work 
Stop the tsp.exe process vlc work fine 

The OS Windows 11 tsp & vlc together work fine 

[image]

lelegard, 2024-04-19:
Hi @edwardlai 

> OS : Windows Server 1029 Standard

I assume you mean "2019", right?

This is a Windows issue. Receiving the same multicast stream on the same machine does not always work as expected, depending on the operating system, its versions, its configuration. Not only Windows (even though I usually consider Windows as _the_ problem in many situations).

Try searching for Windows issues. The problem may be a configuration issue, not necessarily Win 2019 vs. Win 11. I have seen multicast problems because of the presence of VirtualBox on the system. I saw similar reports about Hyper-V.

Anyway, we cannot really help from a TSDuck perspective. Receiving the same multicast stream on the same machine is generally looking for trouble. This is my experience. Sometimes it works, sometimes it doesn't, this is an operating system issue.

lelegard, 2024-04-23:
As I explained, in some operating systems (or configurations), receiving the same multicast stream, on the same port, from two concurrent processes on the system, may work or not. I have seen situations where each incoming UDP datagram was sent to one client only, one or the other, depending on which is waiting first. On other operating systems (or configurations), each incoming UDP datagram is correctly duplicated to each client. I cannot remember the exact conditions, it was a long time ago, but the lesson I learnt was "never listen twice on the same UDP port".

This is why you should look for multicast reception issues on Windows from multiple clients on the same stream.

edwardlai, 2024-04-24:
=> Not all the server operating system have this issue 

  - Win 10 Professional OK 
  - Win 11 Professional OK 
  - Server 2012 R2 Standard OK
  **- Server 2016 Standard  Not work 
  - Server 2019 Standard Not work** 
  - Server 2022 Standard OK

=> Server 2016 & 2019 VLC/ffplay/ffmpeg work concurrently , But incompatible with TSP