tsduck/tsduck #1452: [Q] DAB

EnoSat, 2024-04-21:
Is it possible to analyze and play DAB if TS contains only one PID with two DAB streams encapsulated?
https://mega.nz/file/ZddAVaSS#E0X_SlrChd5RxjKrodASp2cw-G9lV1GNPScT6IcGvEQ
`Streaming 2 streams
[04/21/24 07:16:53] Initialise next pseq to 35191
[04/21/24 07:16:53] EDI-AF: Packet sequence error
Radio fuer NRW   (0x10ec)
 0 : WDR 2 OWL        (0xd592) Pri subch=18 start=594 CUs= 54 PL=eep-3a bitrate=72 DAB+
 1 : WDR 2 RHEIN-RUHR (0xd792) Pri subch=16 start=486 CUs= 54 PL=eep-3a bitrate=72 DAB+
 2 : WDR 2 RUHRGEBIET (0xd692) Pri subch=17 start=540 CUs= 54 PL=eep-3a bitrate=72 DAB+
 3 : WDR Maus         (0xdf95) Pri subch= 2 start= 54 CUs= 48 PL=eep-3a bitrate=64 DAB+
 4 : WDRcosmo         (0xd496) Pri subch= 5 start=204 CUs= 48 PL=eep-3a bitrate=64 DAB+
 5 : 1LIVE diGGi      (0xdf91) Pri subch= 3 start=102 CUs= 54 PL=eep-3a bitrate=72 DAB+
 6 : WDR 4            (0xd394) Pri subch=14 start=360 CUs= 54 PL=eep-3a bitrate=72 DAB+
 7 : WDR 2 RHEINLAND  (0xd392) Pri subch= 7 start=288 CUs= 54 PL=eep-3a bitrate=72 DAB+
 8 : 1LIVE            (0xd391) Pri subch= 1 start=  0 CUs= 54 PL=eep-3a bitrate=72 DAB+
 9 : WDR 3            (0xd393) Pri subch=15 start=414 CUs= 72 PL=eep-3a bitrate=96 DAB+
10 : WDR 5            (0xd395) Pri subch= 4 start=156 CUs= 48 PL=eep-3a bitrate=64 DAB+
11 : WDR Event        (0xdf96) Pri subch= 6 start=252 CUs= 36 PL=eep-3a bitrate=48 DAB+
12 : WDR 2 BERG.-LAND (0xdc92) Pri subch=22 start=810 CUs= 54 PL=eep-3a bitrate=72 DAB+
13 : WDR 2 S�DWESTF   (0xdb92) Pri subch=21 start=756 CUs= 54 PL=eep-3a bitrate=72 DAB+
14 : WDR 2 M�NSTERL   (0xda92) Pri subch=20 start=702 CUs= 54 PL=eep-3a bitrate=72 DAB+
15 : WDR 2 AACHEN     (0xd492) Pri subch=19 start=648 CUs= 54 PL=eep-3a bitrate=72 DAB+`

lelegard, 2024-04-21:
Hi @EnoSat 

TSDuck can only analyze the transport stream. There is no support for DAB encapsulation, no way to analyze the content of the single PID containing multiple DAB streams.

EnoSat, 2024-04-21:
ok :(

closed 2024-10-06 by github-actions[bot]