tsduck/tsduck #1453: [Q] DTA-2136 on Ubuntu Server - DTAPI_E_NO_SUCH_DEVICE

LilFoev, 2024-04-26:
Hi All, 

I have a problem with the Dta Driver on Ubuntu Server and a DTA-2136 Card. 
Os Info:
- Operating System: Ubuntu 23.10
- Kernel: Linux 6.5.0-28-generic
- Architecture: x86-64

```
~$ tsdektec -r
* Error attaching input channel 0: DTAPI_E_NO_SUCH_DEVICE (DTAPI status 4114)
* Error attaching input channel 1: DTAPI_E_NO_SUCH_DEVICE (DTAPI status 4114)
```

The Card is succesfully listed:
```
~$ tsdektec -av
DTAPI: 6.3.0.221, Dta: 4.28.11.289, DtaNw: -1.-1.-1.-1, Service: 5.1.1.120

* Device 0: DTA-2136 (DTA-2136 Dual QAM-A/B/C receiver)
  Physical ports: 4
  Channels: input: 2, output: 2
  Input 0: DTA-2136 port 1 (top), QAM (Automatic spectral inversion correction.,
      Manual symbol rate., VHF-band 47-470MHz, UHF-band 400-862MHz,
      Uni-directional input, Transparent mode, MPEG-2 transport stream,
      Demodulation, QAM-A reception, QAM-B reception, QAM-C reception)
  Input 1: DTA-2136 port 2, QAM (Automatic spectral inversion correction.,
      Manual symbol rate., VHF-band 47-470MHz, UHF-band 400-862MHz,
      Uni-directional input, Get antenna signal from another port, Transparent
      mode, MPEG-2 transport stream, Demodulation, QAM-A reception, QAM-B
      reception, QAM-C reception)
  Output 0: DTA-2136 port 3, ASI (Uni-directional output, Loop-through of
      another input, ASI output signal can be inverted, MPEG-2 transport stream,
      DVB-ASI transport stream, Internal TS rate clock reference)
  Output 1: DTA-2136 port 4, ASI (Uni-directional output, Loop-through of
      another input, ASI output signal can be inverted, MPEG-2 transport stream,
      DVB-ASI transport stream, Internal TS rate clock reference)
  Subsystem id: 0x0000 (DTA-2136)
  Subsystem vendor id: 0x0000
  Device id: 0x0858
  Vendor id: 0x1A0E
  Serial number: 000000007F50C7C0
  Firmware version: 1 (0x00000001)
  Firmware variant: 0 (0x00000000)
  PCI bus: 9, slot: 0
  Customer id: 303839
  Engineering change level: Rev 2
  Manufacture id: 04
  Production date: 2013.09
  Part number: DTA-2136
  Serial number: 2136000448
  Crystal stability: 2ppm@-30/+75
  ```

I already checked several things:

Driver Information
```
# modinfo Dta
filename:       /lib/modules/6.5.0-28-generic/updates/dkms/Dta.ko.zst
license:        GPL
description:    Dta Series Driver
author:         DekTec Digital Video B.V.
license:        GPL
license:        GPL
srcversion:     340C222E60F1F2B2F6BEF85
alias:          pci:v00001A0Ed00000893sv00000000sd00000000bc*sc*i*
alias:          pci:v00001A0Ed00000886sv00000000sd00000000bc*sc*i*
alias:          pci:v00001A0Ed00000884sv00000000sd00000000bc*sc*i*
alias:          pci:v00001A0Ed00005883sv00000000sd00000000bc*sc*i*
[... 88 lines cut ...]
[    6.974069] systemd[1]: /etc/systemd/system/DtapiServiced.service:8: PIDFile= references a path below legacy directory /var/run/, updating /var/run/DtapiServiced.pid → /run/DtapiServiced.pid; please update the unit file accordingly.
[    8.920553] Dta: loading out-of-tree module taints kernel.
[    8.920565] Dta: module verification failed: signature and/or required key missing - tainting kernel
[    9.123162] Dta: The Dta driver V4.28.11.289 has loaded successfully.
[    9.123395] Dta 0000:09:00.0: enabling device (0150 -> 0152)
[    9.455851] Dta: < DtIoConfigReadFromIniFile > Error reading config file /etc/DekTec/Dta/2136/Port1.ini. Error: fffffffeh
[    9.455856] Dta: < DtIoConfigReadFromIniFile > Error reading config file /etc/DekTec/Dta/2136000448/Port1.ini. Error: fffffffeh
[    9.455858] Dta: < DtIoConfigReadFromIniFile > Error reading config file /etc/DekTec/Dta/2136/Port2.ini. Error: fffffffeh
[    9.455860] Dta: < DtIoConfigReadFromIniFile > Error reading config file /etc/DekTec/Dta/2136000448/Port2.ini. Error: fffffffeh
[    9.455862] Dta: < DtIoConfigReadFromIniFile > Error reading config file /etc/DekTec/Dta/2136/Port3.ini. Error: fffffffeh
[    9.455863] Dta: < DtIoConfigReadFromIniFile > Error reading config file /etc/DekTec/Dta/2136000448/Port3.ini. Error: fffffffeh
[    9.455865] Dta: < DtIoConfigReadFromIniFile > Error reading config file /etc/DekTec/Dta/2136/Port4.ini. Error: fffffffeh
[    9.455866] Dta: < DtIoConfigReadFromIniFile > Error reading config file /etc/DekTec/Dta/2136000448/Port4.ini. Error: fffffffeh
[    9.455900] Dta: < DtaDevicePowerUpPost > Start
[   17.543147] Dta: < DtaI2cWrite > Transfer Timeout!!
[   22.700411] Dta: < DtaI2cRead > Transfer Timeout!!
[   27.843163] Dta: < DtaI2cWrite > Transfer Timeout!!
```
These path where "DtIoConfigReadFromIniFile" wants to get .ini files from does not exist on my machine and I wonder where these files should come from? 

Initally I installed the pre build driver package "[dektec-dkms_2024.01.0_all.deb package](https://github.com/tsduck/dektec-dkms/releases/download/v2024.01.0/dektec-dkms_2024.01.0_all.deb)" but tried afterwards also the driver installation via dektec-dkms script .

I would be very grateful for any ideas. 

Thanks!

lelegard, 2024-04-27:
Hi @LilFoev 

It seems more a Dektec configuration issue than a TSDuck issue.
@EricDeJong, any idea to help on this ?

LilFoev, 2024-04-29:
Hi @lelegard, 
thanks for your response. 
I agree that this is not an issue related to tsduck. 
Tested also the DtRecord Utilitie provided by Dektec which shows the same error message. 
I reached out directly to Dektec for support. If you like we can keep that issue open and I will update as soon as I get feedback from Dektec. 

Thanks!

LilFoev, 2024-05-03:
Thanks to Dektec support I got a solution. 
The following Kernel Boot Parameter solved my problem: (Note that the "reading config file" errors can be ignored)

- sudo vi /etc/default/grub
- Add option pci=noioapicquirk to line GRUB_CMDLINE_LINUX_DEFAULT
- sudo update-grub
- sudo reboot

E.g for me the /etc/default/grub file looks now like:
```
...
#   info -f grub -n 'Simple configuration'

GRUB_DEFAULT=0
GRUB_TIMEOUT_STYLE=hidden
GRUB_TIMEOUT=0
GRUB_DISTRIBUTOR=`lsb_release -i -s 2> /dev/null || echo Debian`
GRUB_CMDLINE_LINUX_DEFAULT="pci=noioapicquirk"
GRUB_CMDLINE_LINUX=""

# If your computer has multiple operating systems installed, then you
...
```

closed 2024-05-03 by LilFoev