tsduck/tsduck #1454: [BUG] 'make installer' fails when building without some options (e.g. NOSRT)

Wahwahboy, 2024-05-04:
#### Bug description:
I get a pkg installer error after building without some options (e.g. NOSRT = 1)
As per the error, the cause is a blank line placed in the control file where NOSRT is removed by the 'sed' function.

#### How to reproduce:
Build from source without some option (e.g. make NOSRT=1)
Once complete, from tsduck repo folder:
>make installer

#### Expected behavior:
Install continues - no errors

#### Errors and logs:
dpkg-deb: error: parsing file '/home/jasonh/tsduck/pkg/installers/tmp/DEBIAN/control' **near line 9 package 'tsduck':
 blank line in value of field 'Depends'**
make[2]: *** [Makefile:29: deb-tools] Error 2
make[1]: *** [Makefile:19: installer] Error 2
make: *** [Makefile:150: installer] Error 2`

#### Environment:
**Ubuntu 22.04 running in wsl2 (Windows)**

#### Additional information:

'sed' script output from pkg/deb/Makefile (see missing line)
Result in tmp/DEBIAN/control (empty line)
```
 libssl3,

 libedit2,
 libusb-1.0-0,
 libpcsclite1,
 libcurl4,
```

lelegard, 2024-05-04:
Hi @Wahwahboy 

You must use the same options for the build of the binaries and the build of the packages.

If you used `make NOSRT=1` to build the binaries, you must use `make NOSRT=1 installer` to build the packages. Otherwise, the packages are built assuming that SRT is used and `libsrt` is replaced with the actual package with which the binaries were built, meaning an empty line since the binaries were built without SRT.

In practice, there is no need to separate the two phases, binaries and packages. Just run `make NOSRT=1 installer` from scratch, it will build the binaries and then the packages in a consistent way.

Wahwahboy, 2024-05-04:
Understood - thanks.
Apologies for raising as a bug

Amazing tool btw

closed 2024-05-04 by Wahwahboy