tsduck/tsduck #1455: [FEAT]   Add --json to dvb plugin for timed front end stats.

SonnyWalkman, 2024-05-06:
#### Is your feature request related to a problem?
 No..

#### Proposed solution:

#### Alternatives:

#### Additional information:

lelegard, 2024-05-07:
Hi @SonnyWalkman 

Good idea. Therefore implemented right away...

~~~
$ tsp -I dvb --help
...
  --json-buffer-size value
      With --json-tcp or --json-udp, specify the network socket send buffer
      size in bytes.

  --json-interval seconds
      With --json-line, --json-tcp, --json-udp, specify the interval between
      two status reports. The default is 60 seconds.

  --json-line[='prefix']
      Produce a status report in JSON format at regular intervals. Report the
      JSON text as one single line in the message logger. The optional string
      parameter specifies a prefix to prepend on the log line before the JSON
      text to locate the appropriate line in the logs.

  --json-tcp ip-address:port
      Produce a status report in JSON format at regular intervals. Report the
      JSON text as one single line in a TCP connection. The address specifies
      an IP address or a host name that translates to an IP address. The port
      specifies the destination TCP port. By default, a new TCP connection is
      established each time a JSON message is produced. Be aware that a
      complete TCP connection cycle may introduce some latency in the
      processing. If latency is an issue, consider using --json-udp.

  --json-tcp-keep
      With --json-tcp, keep the TCP connection open for all JSON messages. By
      default, a new TCP connection is established each time a JSON message is
      produced.

  --json-udp ip-address:port
      Produce a status report in JSON format at regular intervals. Report the
      JSON text as one single line in a UDP datagram. The address specifies an
      IP address which can be either unicast or multicast. It can be also a
      host name that translates to an IP address. The port specifies the
      destination UDP port. Be aware that the size of UDP datagrams is limited
      by design to 64 kB. If larger JSON contents are expected, consider using
      --json-tcp.

  --json-udp-local ip-address
      With --json-udp, when the destination is a multicast address, specify the
      IP address of the outgoing local interface. It can be also a host name
      that translates to a local address.

  --json-udp-ttl value
      With --json-udp, specifies the TTL (Time-To-Live) socket option. The
      actual option is either "Unicast TTL" or "Multicast TTL", depending on
      the destination address. Remember that the default Multicast TTL is 1 on
      most systems.
~~~
These options are the same as in other tools or plugins such as `analyze`. The code to manage them and implement their logic is in a common C++ class. This is why it was rather straightforward to implement in the `dvb` plugin.

Sample JSON message (after reformatting through `jq`):
~~~
{
  "bandwidth": 8000000,
  "ber": "0%",
  "delivery-system": "DVB-T",
  "frequency": 482000000,
  "guard-interval": "1/32",
  "hierarchy": "none",
  "high-priority-fec": "auto",
  "low-priority-fec": "auto",
  "modulation": "64-QAM",
  "packet-index": 326720,
  "per": "0%",
  "signal-locked": true,
  "snr": "20.861 dB",
  "strength": "-29.342 dB",
  "time": "2024-05-07 21:15:22",
  "transmission-mode": "8K"
}
~~~
Note that the capability to report strength or SNR depends on the driver and OS. The message above was reported on Ubuntu using a Pinnacle PCTV 72e tuner. An Hauppauge Nova-T Stick on the same system was only able to report signal-locked, no strength, no SNR.

Let us know if it works for you.

SonnyWalkman, 2024-05-07:
Hi there @lelegard ,

Much appreciated.. I was using dvb and tsanalyse together and though if the idea!

I'll give it a shot after a rebuild from source.

closed 2024-07-01 by SonnyWalkman