tsduck/tsduck #1456: [Q] How to parse SCTE-35 messages using tsduck?

arhankundu99, 2024-05-08:
Hi everyone,
I am sending a RTP stream from AWS mediaconnect to my ec2 instance and I am reading the stream using `ffmpeg` libraries.
So I have the hexadecimal representation of SCTE-35 messages from the stream and I want to parse those messages using tsduck libraries.

Example hexadecimal message: fc303a00000000000000fff00506fe794e32480024022243554549040031767fc000001499700e0c41594c3030303030303030332001010101e9b86e1a

Parsed SCTE-35 message:
```json
{
  table_id: "0xfc",
  section_syntax_indicator: false,
  private_indicator: false,
  section_length: "0x3a",
  protocol_version: "0x0",
  encrypted_packet: false,
  encryption_algorithm: "0x0",
  pts_adjustment: "0x0",
  cw_index: "0x0",
  tier: "0xfff",
  splice_command_length: "0x5",
  splice_command_type: "TimeSignal",
  SpliceCommand: {
    SpliceTime: {
      time_specified_flag: true,
      pts_time: "0x794e3248"
    }
 },
descriptor_loop_length: "0x24",
SpliceDescriptors: [
{
  splice_descriptor_tag: "SegmentationDescriptor",
  descriptor_length: "0x20",
  identifier: "0x43554549",
  segmentation_event_id: "0x4003176",
  segmentation_event_cancel_indicator: false,
  program_segmentation_flag: true,
  segmentation_duration_flag: true,
  delivery_not_restricted_flag: false,
  web_delivery_allowed_flag: false,
  no_regional_blackout_flag: false,
  archive_allowed_flag: false,
  device_restrictions: "RestrictGroup0",
  component_count: "0x0",
  ComponentTags: [],
  ComponentPtsOffsets: [],
  segmentation_duration: "0x149970",
  SegmentationUPIDs:
  [{
    segmentation_upid_type: "ADS",
    segmentation_upid_length: "0xc",
    Data: "QVlMMDAwMDAwMDAz",
    MPU: null,
    SegmentationUPIDFormat: "NotSpecified",
    Value: ""
  }],
  segmentation_type_id: "ChapterStart",
  segment_num: "0x1",
  segments_expected: "0x1",
  sub_segment_num: "0x0",
  sub_segments_expected: "0x0"
 }],
  alignment_stuffing: null,
  E_CRC_32: "0x0",
  CRC_32: "0x101e9b8"
}

```
Can anyone please share me some code examples on how to do this?
Thank you in advance.

lelegard, 2024-05-08:
Hi @arhankundu99,

There are several levels of response, command line or C++ coding.

At command line level, the command `tstabdump` analyzes binary sections in human-readable form.

Example with your hexa data (the command `xxd` decodes the hexa into binary):
~~~
$ echo fc303a00000000000000fff00506fe794e32480024022243554549040031767fc000001499700e0c41594c3030303030303030332001010101e9b86e1a \
    | xxd -r -p \
    | tstabdump

* SCTE 35 Splice Information, TID 252 (0xFC)
  Short section, size: 61 bytes
  Protocol version: 0
  Encryption: none
  PTS adjustment: 0x000000000
  CW index: 0x00 (0), tier: 0xFFF (4095)
  Command type: 0x06 (TimeSignal), size: 5 bytes
  Time: 0x0794E3248 (2,035,167,816, 22,612,975 ms)
  - Descriptor 0: SCTE 35 Segmentation (0x02, 2), 34 bytes
    Identifier: 0x43554549 ("CUEI")
    Segmentation event id: 0x04003176, cancel: 0
    Program segmentation: 1, has duration: 1, not restricted: 0
    Web delivery allowed: 0, no regional blackout: 0
    Archive allowed: 0, device restrictions: 0
    Segment duration: 1350000
    Segmentation upid type: 0x0E (ADS Information)
    Upid data (12 bytes):
      0000:  41 59 4C 30 30 30 30 30 30 30 30 33              AYL000000003
    Segmentation type id: 0x20 (Chapter Start)
    Segment number: 1, expected segments: 1
    Extraneous 2 bytes:
    0000:  01 01                                            ..
  CRC32: 0xE9B86E1A (OK)
~~~

Note here that your input data are incorrect. The segmentation descriptor contains two extraneous bytes.

See the manual analysis:
~~~
Hexa data: fc303a00000000000000fff00506fe794e32480024022243554549040031767fc000001499700e0c41594c3030303030303030332001010101e9b86e1a
   122 hexa digits = 61 bytes = section header: 3 bytes + payload: 58 bytes

Header: fc303a (len = 0x3a = 58 -> ok)
Payload:
    00000000000000fff00506fe794e3248
    Descriptor loop: 0024
      Descriptor header: 02 22 (len = 0x22 = 34)
      Descriptor payload: 43554549040031767fc00000149970 0e 0c 41594c303030303030303033 20 0101 0101
CRC32: e9b86e1a (computed ok by tstabdump)
~~~
You can see that the descriptor contains a duplicated 0101 sequence at the end.

The tool you use to display the JSON analysis is also incorrect for several reasons:

1. It incorrectly identifies the final CRC as "0x101e9b8", which is in fact the extraneous 0101 from the previous descriptor and the first 2 bytes of the real CRC. Not only the CRC was incorrectly located but it is not even checked because this value is obviously an invalid CRC for the section, with or without the extraneous 0101. In contrast, `tstabdump` recomputes the CRC and finds it "(OK)".

2. It displays `sub_segment_num: "0x0"` and `sub_segments_expected: "0x0"` coming out of nowhere since these two fields are documented in SCTE-35 as present only when `segmentation_type_id` is one of 0x34, 0x36, 0x38, 0x3A. In your case, it is 0x20. We could imagine that the extraneous 0101 comes from here. However, the two values would be reported as 0x1, not 0x0.

So, both the input data and the analysis tools are incorrect.

Now, if you want to parse the section in C++ using TSDuck and manipulate it as a C++ object, here is some sample code. The `data` block is your section content.
~~~
#include "tsduck.h"

static const uint8_t data[] = {
    0xFC, 0x30, 0x3A, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFF, 0xF0, 0x05, 0x06, 0xFE, 0x79,
    0x4E, 0x32, 0x48, 0x00, 0x24, 0x02, 0x22, 0x43, 0x55, 0x45, 0x49, 0x04, 0x00, 0x31, 0x76, 0x7F,
    0xC0, 0x00, 0x00, 0x14, 0x99, 0x70, 0x0E, 0x0C, 0x41, 0x59, 0x4C, 0x30, 0x30, 0x30, 0x30, 0x30,
    0x30, 0x30, 0x30, 0x33, 0x20, 0x01, 0x01, 0x01, 0x01, 0xE9, 0xB8, 0x6E, 0x1A,
};

int main(int argc, char* argv[])
{
    // Build a binary section with one section from the raw data.
    ts::BinaryTable tab;
    tab.addSection(ts::SectionPtr(new ts::Section(data, sizeof(data))));

    // Parse the binary table as a SpliceInformationTable.
    ts::DuckContext duck;
    ts::SpliceInformationTable splice(duck, tab);

    // Display some parsed content. Be sure to check that the parsed table is valid.
    std::cout << "Splice valid: " << splice.isValid() << std::endl;
    std::cout << "Command type: " << int(splice.splice_command_type) << std::endl;
    if (splice.splice_command_type == ts::SPLICE_TIME_SIGNAL) {
        std::cout << "Time: " << splice.time_signal.toString() << std::endl;
    }
}
~~~

Build and run:
~~~
$ clang sample.cpp -o sample $(tsconfig --cflags) $(tsconfig --libs)
$ ./sample 
Splice valid: 1
Command type: 6
Time: 0x0794E3248 (2,035,167,816, 22,612,975 ms)
~~~

As a conclusion, be sure to read the TSDuck [user's guide](https://tsduck.io/download/docs/tsduck.pdf) and [programmer's guide](https://tsduck.io/doxy/) and its [Application development](https://tsduck.io/doxy/appdev.html) section.

lelegard, 2024-05-08:
Also note that the table was correctly parsed as a C++ object because the table itself is correct, including its final CRC32. However, when browsing through the list of descriptors, the segmentation descriptor will not be parsed because its binary content is invalid (the two additional bytes).

This can be seen after the following modification in the code:
~~~
#include "tsduck.h"

static const uint8_t data[] = {
    0xFC, 0x30, 0x3A, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFF, 0xF0, 0x05, 0x06, 0xFE, 0x79,
    0x4E, 0x32, 0x48, 0x00, 0x24, 0x02, 0x22, 0x43, 0x55, 0x45, 0x49, 0x04, 0x00, 0x31, 0x76, 0x7F,
    0xC0, 0x00, 0x00, 0x14, 0x99, 0x70, 0x0E, 0x0C, 0x41, 0x59, 0x4C, 0x30, 0x30, 0x30, 0x30, 0x30,
    0x30, 0x30, 0x30, 0x33, 0x20, 0x01, 0x01, 0x01, 0x01, 0xE9, 0xB8, 0x6E, 0x1A,
};

int main(int argc, char* argv[])
{
    // Use a Duck context with initial default configuration.
    ts::DuckContext duck;

    // Build a binary section with one section from the raw data.
    ts::BinaryTable tab;
    tab.addSection(ts::SectionPtr(new ts::Section(data, sizeof(data))));

    // Parse the binary table as a SpliceInformationTable.
    ts::SpliceInformationTable splice(duck, tab);

    // Display some parsed content. Be sure to check that the parsed table is valid.
    if (splice.isValid()) {
        std::cout << "Command type: " << int(splice.splice_command_type) << std::endl;
        if (splice.splice_command_type == ts::SPLICE_TIME_SIGNAL) {
            std::cout << "Time: " << splice.time_signal.toString() << std::endl;
        }
        std::cout << "Descriptors count: " << splice.descs.size() << std::endl;
        for (size_t i = 0; i < splice.descs.size(); ++i) {
            std::cout << "Descriptor #" << i << " tag: " << int(splice.descs[i]->tag()) << std::endl;
            if (splice.descs[i]->tag() == ts::DID_SPLICE_SEGMENT) {
                ts::SpliceSegmentationDescriptor seg(duck, *splice.descs[i]);
                std::cout << "SpliceSegmentationDescriptor valid: " << seg.isValid() << std::endl;
            }
        }
    }
}
~~~

Output:
~~~
Command type: 6
Time: 0x0794E3248 (2,035,167,816, 22,612,975 ms)
Descriptors count: 1
Descriptor #0 tag: 2
SpliceSegmentationDescriptor valid: 0
~~~

The last output line shows that the deserialised descriptor is not valid, even though its tag is the one of a splice segmentation descriptor.

arhankundu99, 2024-05-08:
Thank you so much for the explanation and code examples

closed 2024-05-08 by arhankundu99