tsduck/tsduck #1459: [Q] How to play a DVB-T stream with dektec with correct arguments?

stulluk, 2024-05-11:
My Original question was here:

https://github.com/tsduck/tsduck/issues/1345#issuecomment-2105300345

The perfect answer from @lelegard  was here:

https://github.com/tsduck/tsduck/issues/1345#issuecomment-2105683725

Starting with 2nd Question: **Why tsscan does not report correct Modulation parameters ?** 
Answer: Current modern hardware and corresponding drivers does not report modulation parameters correctly. such as Guard Interval. Many variations of modulations parameters may end up with the same bitrate.

Going back to original 1st question:
So to summarize 1st question: **What are the correct cmdline arguments** to play a DVBT TS file with TSduck exactly same as StreamXpress ?

If I play ( via DTU-315 ) a valid recorded DVB-T TS file ( Such as the one in tsduck streams repository here: https://tsduck.io/streams/poland-dttv-warsaw/PL_DVB-T_2023-11-10_MUX3_522MHz.ts ) via StreamXpress from Dektec, and capture it via TBS5530 ( via tsduck / tsp ) -> Resultant TS file is here: 
http://dc7.kernelmax.com:8008/played-via-streamxpress-captured-via-tsduck.ts

[image]

The TS file is almost perfect, no visual artifacts, can be played with MPV / VLC, no issues. And in addition, StreamXpress reports the bitrate as 24,882,353 b/s , exactly same as tsduck.

[image]

But when I play the same TS with tsduck instead of StreamXpress, it reports the output bitrate as 24,882,296 b/s , and when I capture it with exactly same cmdline arguments with tsduck with TBS5530 , output TS file is here:
http://dc7.kernelmax.com:8008/played-via-tsduck-capture-via-tsduck-2.ts

The TS file has tons of continiuty errors, can not be smoothly played with MPV / VLC:

[image]

Note: I tried "--stuffing" and "--pilots" arguments but nothing changed.

If the problem is not the bitrate or 0.0002 % missing packets, then why recorded TS has such issues ?

Thanks for your guidance & answers in advance @lelegard .

lelegard, 2024-05-11:
> when I play the same TS with tsduck instead of StreamXpress, it reports the output bitrate as 24,882,296 b/s

Who is "it"? If this is the file input plugin of tsp, then I explained that this is normal because this is what is approximately measured in the input file, from the average distance between PCR's, and probably a few missing packets during the orignal capture of that file.

If that file is correctly broadcasted by Dektec StreamXpress, then there is no visible problem in that file.

>  What are the correct cmdline arguments to play a DVBT TS file with TSduck exactly same as StreamXpress ?

From the display of StreamXpress, the complete parameter set should be:
~~~
-O dektec --modulation DVB-T --frequency 522000000 \
   --bandwidth 8 --guard-interval 1/8 --transmission-mode 8K \
   --convolutional-rate 3/4 --constellation 64-QAM --cell-id 0
~~~

You may try `--power-mode high-quality`. This applies to DTU-315 and may make a difference. I no longer have any Dektec device to test and I cannot reproduce the issue.

Does StreamXpress run on the same system as `tsp -O dektec`? Or does StreamXpress run on Windows and tsp on Linux? In that case, it can be a matter of Dektec driver. You may try to run tsp on the same Windows system as StreamXpress to verify that.

stulluk, 2024-05-11:
@lelegard  sorry for confusion.

I made a great progress on this situation, and it seems to be related to my Lenovo T14 laptop or TBS5530 hardware or its kernel driver.

Let me explain.

I have two computers: One Lenovo T14 ( Laptop - has dual boot Win11 and Ubuntu 22.04 ). And I also have a AMD Ryzen 5700G based Desktop PC.

What I did is to use "tsanalyze" and play from one PC ( via DTU-315 ) and capture from another ( via TBS5530 )

Scenario-1:  Connect Dektec DTU-315 (playback) to Laptop, and connect TBS5530 (capture) to Desktop. 

In this scenario, here is the original file analysis:

```bash
==============================================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                                          |
|============================================================================================|
|  Transport Stream Id: ................. 3 (0x0003)  |  Services: ..................... 11  |
|  Bytes: .............................. 373,184,512  |  PID's: Total: ................. 59  |
|  TS packets: ........................... 1,985,024  |         Clear: ................. 59  |
|     With invalid sync: ......................... 0  |         Scrambled: .............. 0  |
|     With transport error: ...................... 0  |         With PCR's: ............. 7  |
|     Suspect and ignored: ....................... 0  |         Unreferenced: ........... 0  |
|--------------------------------------------------------------------------------------------|
|  Transport stream bitrate, based on ...................... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: .................................................. None             None  |
|  Estimated based on PCR's: .............................. 24,882,293 b/s   26,999,935 b/s  |
|  Selected reference bitrate: ............................ 24,882,293 b/s   26,999,935 b/s  |
|--------------------------------------------------------------------------------------------|
|  Broadcast time: ................................................. 119 sec (1 min 59 sec)  |
|  First TDT UTC time stamp: .......................................... 2023/11/10 17:49:28  |
|  Last TDT UTC time stamp: ........................................... 2023/11/10 17:51:08  |
|  First TOT local time stamp: ........................................ 2023/11/10 18:49:18  |
|  Last TOT local time stamp: ......................................... 2023/11/10 18:50:58  |
|  TOT country code: .................................................................. POL  |
|--------------------------------------------------------------------------------------------|
|  Srv Id           Service Name                                    Access          Bitrate  |
|  0x0001      (1)  TVP1 ............................................... C    4,764,105 b/s  |
|  0x0002      (2)  TVP2 ............................................... C    3,902,487 b/s  |
|  0x0015     (21)  TVP3 Warszawa ...................................... C    2,534,554 b/s  |
|  0x0020     (32)  TVP Historia ....................................... C    1,867,052 b/s  |
|  0x0021     (33)  TVP Sport .......................................... C    5,601,331 b/s  |
|  0x0023     (35)  TVP Info ........................................... C    3,679,702 b/s  |
|  0x0058     (88)  TVP GO ............................................. C      529,642 b/s  |
|  0x005B     (91)  TVP World .......................................... C      517,633 b/s  |
|  0x005C     (92)  TVP ABC 2 .......................................... C      529,629 b/s  |
|  0x005D     (93)  TVP Historia 2 ..................................... C      529,642 b/s  |
|  0x005E     (94)  TVP Kultura 2 ...................................... C      529,629 b/s  |
|                                                                                            |
|  Note 1: C=Clear, S=Scrambled                                                              |
|  Note 2: Unless specified otherwise, bitrates are based on 188 bytes/pkt                   |
==============================================================================================

```

And TS analysis of the captured TS file from Desktop PC :

```bash
==============================================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                                          |
|============================================================================================|
|  Transport Stream Id: ................. 3 (0x0003)  |  Services: ..................... 11  |
|  Bytes: .............................. 135,344,960  |  PID's: Total: ................. 59  |
|  TS packets: ............................. 719,920  |         Clear: ................. 59  |
|     With invalid sync: ......................... 0  |         Scrambled: .............. 0  |
|     With transport error: ...................... 0  |         With PCR's: ............. 7  |
|     Suspect and ignored: ....................... 0  |         Unreferenced: ........... 0  |
|--------------------------------------------------------------------------------------------|
|  Transport stream bitrate, based on ...................... 188 bytes/pkt    204 bytes/pkt  |
[... 101 lines cut ...]
Desktop's kernel driver for TBS5530:

```bash
stulluk ~ $  modinfo dvb_usb_tbs5530
[... cut ...]
retpoline:      Y
name:           dvb_usb_tbs5530
vermagic:       6.2.0-33-generic SMP preempt mod_unload modversions 
parm:           debug:set debugging level (1=info 2=xfer (or-able)). (int)
parm:           adapter_nr:DVB adapter numbers (array of short)
stulluk ~ $  
```

As you noticed, laptop kernel is 6.2.0-36 and Desktop's kernel is 6.2.0-33 .

I am suspecting a hardware failure here, not related to tsduck, since Scenario-1 works perfectly.

I will close this ticket after I will find out.

stulluk, 2024-05-12:
OK, used a 3rd PC ( An Intel NUC with Ubuntu 22.04 again ) and used exactly same commands, and exactly same kernel & TBS5530 drivers. 

```bash
==============================================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                                          |
|============================================================================================|
|  Transport Stream Id: ................. 3 (0x0003)  |  Services: ..................... 11  |
|  Bytes: ............................... 68,048,480  |  PID's: Total: ................. 59  |
|  TS packets: ............................. 361,960  |         Clear: ................. 59  |
|     With invalid sync: ......................... 0  |         Scrambled: .............. 0  |
|     With transport error: ...................... 0  |         With PCR's: ............. 7  |
|     Suspect and ignored: ....................... 0  |         Unreferenced: ........... 0  |
|--------------------------------------------------------------------------------------------|
|  Transport stream bitrate, based on ...................... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: .................................................. None             None  |
|  Estimated based on PCR's: .............................. 24,882,293 b/s   26,999,935 b/s  |
|  Selected reference bitrate: ............................ 24,882,293 b/s   26,999,935 b/s  |
|--------------------------------------------------------------------------------------------|
|  Broadcast time: .................................................. 21 sec (0 min 21 sec)  |
|  First TDT UTC time stamp: .......................................... 2023/11/10 17:50:08  |
|  Last TDT UTC time stamp: ........................................... 2023/11/10 17:50:08  |
|  First TOT local time stamp: ........................................ 2023/11/10 18:49:58  |
|  Last TOT local time stamp: ......................................... 2023/11/10 18:50:18  |
|  TOT country code: .................................................................. POL  |
|--------------------------------------------------------------------------------------------|
|  Srv Id           Service Name                                    Access          Bitrate  |
|  0x0001      (1)  TVP1 ............................................... C    4,624,700 b/s  |
|  0x0002      (2)  TVP2 ............................................... C    4,108,163 b/s  |
|  0x0015     (21)  TVP3 Warszawa ...................................... C    2,531,950 b/s  |
|  0x0020     (32)  TVP Historia ....................................... C    1,860,879 b/s  |
|  0x0021     (33)  TVP Sport .......................................... C    5,537,541 b/s  |
|  0x0023     (35)  TVP Info ........................................... C    3,678,793 b/s  |
|  0x0058     (88)  TVP GO ............................................. C      529,735 b/s  |
|  0x005B     (91)  TVP World .......................................... C      517,636 b/s  |
|  0x005C     (92)  TVP ABC 2 .......................................... C      529,666 b/s  |
|  0x005D     (93)  TVP Historia 2 ..................................... C      529,666 b/s  |
|  0x005E     (94)  TVP Kultura 2 ...................................... C      529,666 b/s  |
|                                                                                            |
|  Note 1: C=Clear, S=Scrambled                                                              |
|  Note 2: Unless specified otherwise, bitrates are based on 188 bytes/pkt                   |
==============================================================================================
```

I don't know what is wrong with my Lenovo T14 laptop, but this looks like a Hardware failure with the laptop.

If there are any suggestion for analysis, please let me know. I will close this ticket.

closed 2024-05-12 by stulluk