tsduck/tsduck #1461: [Q] How to parse this particular SCTE-35 message?

arhankundu99, 2024-05-14:
Hi,
I am trying to parse SCTE-35 message with hex data: `fc303a00000000000000fff00506ffe3d359f80024022243554549040001257fc000001b77400e0c41594c30303030303030303120010101012c124463`

I have the below function to get the parsed SCTE-35 message
```cpp
SCTE35Message SCTE35Parser::parse(const uint8_t* packetData, size_t packetSize) {
    SCTE35Message scte35Message;

    // Create a section from the data pointer
    ts::BinaryTable binaryTable;
    ts::SectionPtr sectionPtr = ts::SectionPtr(new ts::Section(packetData, packetSize));
    binaryTable.addSection(sectionPtr);

    ts::SpliceInformationTable sit(duckContext, binaryTable);
    ts::TablesDisplay display(duckContext);

    ts::Descriptor descriptor(static_cast<const void*>(packetData), packetSize);

    scte35Message.isValid = sit.isValid();
    scte35Message.data = playout::utils::byteToHex(packetData, packetSize);
    if (sit.isValid()) {
        // Extracting relevant details for scte message
        scte35Message.protocolVersion = sit.protocol_version;
        scte35Message.spliceCommandType = this->getSpliceCommandType(sit.splice_command_type);

        for (int i = 0; i < sit.descs.count(); i++) {
            const ts::SpliceSegmentationDescriptor ssd(duckContext, *sit.descs[i]);
            scte35Message.identifier = playout::utils::decimalToHex(ssd.identifier);
            scte35Message.noRegionalBlackout = ssd.no_regional_blackout;
            scte35Message.segmentNum = ssd.segment_num;
            scte35Message.segmentationTypeId = playout::utils::decimalToHex(ssd.segmentation_type_id);
            scte35Message.segmentationDescription = this->getSegmentationDescription(ssd.segmentation_type_id);
            scte35Message.segmentationEventId = playout::utils::decimalToHex(ssd.segmentation_event_id);
            scte35Message.segmentationUpidType = playout::utils::decimalToHex(ssd.segmentation_upid_type);
            scte35Message.programSegmentation = ssd.program_segmentation;
            scte35Message.segmentationEventCancel = ssd.segmentation_event_cancel;
            scte35Message.webDeliveryAllowed = ssd.web_delivery_allowed;
            scte35Message.upidData = playout::utils::byteBlockToHex(ssd.segmentation_upid);
            scte35Message.adId = playout::utils::hexToAscii(scte35Message.upidData);
        }   

        if (sit.splice_command_type == ts::SPLICE_TIME_SIGNAL) {
            
        }
    } else {
        std::cerr << "Invalid SCTE-35 data" << std::endl;

    }
    return scte35Message;
}
```

And a display function to display the message
```cpp
void SCTE35Parser::display(const uint8_t* packetData, size_t packetSize) {

    // Create a section from the data pointer
    ts::Section section(packetData, packetSize, ts::PID_NULL, ts::CRC32::CHECK);

    if (section.isValid()) {
   
        ts::TablesDisplay display(duckContext);
        // Display the section
        display.displaySection(section);
        std::cout << "\n";
    }
}
```

the `displaySection` function of `ts::TablesDisplay`  is displaying the message correctly
```
* unknown (0xFC), TID 0xFC (252)
Short section, size: 61 bytes
Protocol version: 0
Encryption: none
PTS adjustment: 0x000000000
CW index: 0x00 (0), tier: 0xFFF (4095)
Command type: 0x0000000000000006 (unknown), size: 5 bytes
Time: 0x1E3D359F8 (8,117,246,456, 90,191,627 ms)
- Descriptor 0: unknown (0x02, 2), 34 bytes
Identifier: 0x43554549 ("CUEI")
Segmentation event id: 0x04000125, cancel: 0
Program segmentation: 1, has duration: 1, not restricted: 0
Web delivery allowed: 0, no regional blackout: 0
Archive allowed: 0, device restrictions: 0
Segment duration: 1800000
Segmentation upid type: 0x000000000000000E (unknown)
Upid data (12 bytes):
    0000:  41 59 4C 30 30 30 30 30 30 30 30 31              AYL000000001
Segmentation type id: 0x0000000000000020 (unknown)
Segment number: 1, expected segments: 1
Extraneous 2 bytes:
 0000:  01 01                                            ..
CRC32: 0x2C124463 (OK)
```

But when I parse this using the `parse` function given above and print the `SCTE35Message`, I am getting an invalid message

```
 SCTE35Message {isValid: true, encodedData: 'fc303a00000000000000fff00506ffe3d359f80024022243554549040001257fc000001b77400e0c41594c30303030303030303120010101012c124463', protocolVersion: 0, spliceCommandType: 'TimeSignal', identifier: '0x43554549', noRegionalBlackout: true, segmentNum: 0, segmentationTypeId: '0x00', segmentationDescription: 'Unknown Segmentation Type ID', segmentationEventId: '0x00', segmentationUpidType: '0x00', programSegmentation: true, segmentationEventCancel: false, webDeliveryAllowed: true, upidData: '', adId:  }
 ```
 
 Can anyone please help me on how to parse this correctly?

lelegard, 2024-05-14:
Hi, @arhankundu99 

This is the same problem as #1456. Your splice section is globally correct but your segmentation descriptor is incorrect. It has 2 extraneous bytes. The display function does it best to display what is inside the section (including the extraneous bytes). But the parsing of the descriptor, as a C++ object, will fail because the binary data are invalid.

~~~
$ echo fc303a00000000000000fff00506ffe3d359f80024022243554549040001257fc000001b77400e0c41594c30303030303030303120010101012c124463 \
   | xxd -r -p | tstabdump -

* SCTE 35 Splice Information, TID 252 (0xFC)
  Short section, size: 61 bytes
  Protocol version: 0
  Encryption: none
  PTS adjustment: 0x000000000
  CW index: 0x00 (0), tier: 0xFFF (4095)
  Command type: 0x06 (TimeSignal), size: 5 bytes
  Time: 0x1E3D359F8 (8,117,246,456, 90,191,627 ms)
  - Descriptor 0: SCTE 35 Segmentation (0x02, 2), 34 bytes
    Identifier: 0x43554549 ("CUEI")
    Segmentation event id: 0x04000125, cancel: 0
    Program segmentation: 1, has duration: 1, not restricted: 0
    Web delivery allowed: 0, no regional blackout: 0
    Archive allowed: 0, device restrictions: 0
    Segment duration: 1800000
    Segmentation upid type: 0x0E (ADS Information)
    Upid data (12 bytes):
      0000:  41 59 4C 30 30 30 30 30 30 30 30 31              AYL000000001
    Segmentation type id: 0x20 (Chapter Start)
    Segment number: 1, expected segments: 1
    Extraneous 2 bytes:
    0000:  01 01                                            ..
  CRC32: 0x2C124463 (OK)
~~~

Additionally, there are several other issues.

First, you have probably not correctly installed the TSDuck library (`make install` or .rpm or .deb package file). We can see that in your display function. The structure of the section is correctly displayed but the name of the section and the descriptor, as well as the splice command type are set as "unknown". The names are in configuration files. If you did not properly install the library and its development environment, you won't see any meaningful name, and other weird behaviours may happen.

The following code is also invalid:
~~~
ts::Descriptor descriptor(static_cast<const void*>(packetData), packetSize);
~~~
Your binary data contain a section, not a descriptor. So, you cannot do anything with this variable. You don't use it anyway, so I assume it was a former incorrect attempt.

After this code, you use `ssd` without checking that it is valid (`ssd.isValid()`).
~~~
const ts::SpliceSegmentationDescriptor ssd(duckContext, *sit.descs[i]);
~~~
A descriptor can be anything, not necessarily a splice segmentation descriptor. So, you have to check that the descriptor was properly parsed as the expected type.

Furthermore, in your case, event though your descriptor is a segmentation one, it is incorrect because of the extraneous bytes. So, the parsing fails. You would see it by checking `ssd.isValid()`. Therefore, the variable `ssd` does not contain any correct descriptor (probably the result of the default constructor).

arhankundu99, 2024-05-14:
But using the code I mentioned above, the SCTE message is getting parsed correctly for valid hex message.
Valid hex message: `'fc303300000000000000fff00506fe1099fde8001d021b43554549040002377f800e0c41594c303030303030303132210101627c1224`

The `display` function prints the below message
```
 Short section, size: 54 bytes
Protocol version: 0
Encryption: none
PTS adjustment: 0x000000000
Command type: 0x0000000000000006 (unknown), size: 5 bytes
Time: 0x01099FDE8 (278,527,464, 3,094,749 ms)
- Descriptor 0: unknown (0x02, 2), 27 bytes
Identifier: 0x43554549 ("CUEI")
Segmentation event id: 0x04000237, cancel: 0
Program segmentation: 1, has duration: 0, not restricted: 0
Web delivery allowed: 0, no regional blackout: 0
Archive allowed: 0, device restrictions: 0
Segmentation upid type: 0x000000000000000E (unknown)
Upid data (12 bytes):
    0000:  41 59 4C 30 30 30 30 30 30 30 31 32              AYL000000012
Segmentation type id: 0x0000000000000021 (unknown)
Segment number: 1, expected segments: 1
CRC32: 0x627C1224 (OK)
```

And the SCTEMessage I get from `parse` function is this
```
SCTE35Message {isValid: true, encodedData: 'fc303300000000000000fff00506fe1099fde8001d021b43554549040002377f800e0c41594c303030303030303132210101627c1224', protocolVersion: 0, spliceCommandType: 'TimeSignal', identifier: '0x43554549', noRegionalBlackout: false, segmentNum: 1, segmentationTypeId: '0x21', segmentationDescription: 'Chapter End', segmentationEventId: '0x4000237', segmentationUpidType: '0x0e', programSegmentation: true, segmentationEventCancel: false, webDeliveryAllowed: false, upidData: '41594c303030303030303132', adId: AYL000000012 }
```

lelegard, 2024-05-15:
> But using the code I mentioned above, the SCTE message is getting parsed correctly for valid hex message.

Precisely because the second message is correct and the first one incorrect, one is correctly parsed and the other one is not.

Using the trick `echo hexa-text | xxd -r -p | tstabdump`, you can easily see this. Because of the extraneous two bytes in the segmentation descriptor, the parsing fails.

I don't understand what is your problem exactly. Failing to parse an incorrect message and succeeding in parsing a correct message is exactly what is expected.

arhankundu99, 2024-05-15:
No sorry for the confusion, what I meant to say is that even though the hex message was invalid, the display function was still able to print the message.
I was wondering if I can somehow parse that data in my struct using splice information table class.

lelegard, 2024-05-16:
The display and deserialize functions are two different kinds of "parse".

The display function tries its best to display was can be parsed and give some clue on what is incorrect. Depending on the kind of corruption, it can be complete garbage or useful information such as "2 extraneous bytes". This is debug tool.

The deserialization function, on the other hand, cannot report incorrect data. There would be no way to check which field is correct or not. So, deserialization must reject incorrect data. No alternative.

arhankundu99, 2024-05-16:
Makes sense, thank you so much for taking time out!

closed 2024-05-16 by arhankundu99