tsduck/tsduck #1463: [Q]  User-specified vs Estimated bitrate

edepa, 2024-05-18:
Good day,

I am new to TSDuck, and I am using its output 
to measure the performance of a video service in my lab.

I confess that I am also new to video streaming; 
my interest lies in measuring power use of video service infrastructure. 
This disclaimer might jsutify the basic level of my question.

I am streaming a video across a LAN using ffmpeg and
measuring its parameters using tsp and the analyze plugin.
tsp's output is a series of files, 
where each file reports measurements 
over a 5-second interval of measurement.

In each such output file, I see:
(1) a user-specified bitrate and 
(2) a bitrate that is "Estimated based on PCR's". 

Both bitrates vary from file to file.
I don't understand what is "user-specified", 
nor what the analyze plugin is measuring
to report two distinct measurements.
I have searched and found out about
the method used to produce the "Estimate based on PCR's", but
I am still unable to understand the "user-specified" bitrate.

With thanks,

Etienne

lelegard, 2024-05-19:
Hi @edepa 

First, to establish a context, there are two kinds of transport streams, as far as bitrate is concerned: CBR, constant bitrate, and VBR, variable bitrate.

Broadcast streams, the original usage for transport streams, are always CBR because the modulation schemes always produce a constant bitrate. TSDuck was originally developed in a CBR-only context when I worked for a TV operator (satellite, cable, terrestrial) and needed test tools.

Video streaming over IP has no such modulation constraints. The streams can be VBR, meaning that the bitrate changes all the time. In a VBR stream, the bitrate can be only an average over a given period of time. In your case, the TS can be CBR or VBR, it depends on the ffmpeg encoding parameters.

Second, TSDuck is a test tool which is supposed to work on any context, any type of input, live streams, offline stored files or any kind of input the speed of which is different from the playout speed. Therefore, in generic modules such as the analysis one, TSDuck never relies on wall-clock time to compute bitrates.

Now, how can we analyze a the bitrate of a stream? As you noticed, counting the number of packets (and therefore bits) between PCR timestamps is a good way of getting the average bitrate between these two timestamps. However, this method has some limitations:

- The stream may be incorrect, corrupted, inaccurate, incorrectly multiplexed, etc. and PCR are not reliable.
- Very few PID's have PCR. Typically, there is only one PCR stream per service, usually the video PID or a dedicated one.
- To compute the average bitrate of PID's without PCR, we need to have an idea of the TS global bitrate and then we count packets in each PID.

So, as you noticed, the TS bitrate which is "estimated based on PCR's" is an average of all bitrates which were computed from all PID's containing PCR's.

- On a perfectly captured broadcast CBR stream, without loss of packets, this value is very close to the theoretical bitrate value from the modulation parameters. However, the theoretical value is precise to the bit (or even "sub-bit") while the evaluation is based on an integral number of TS packets = 8 x 188 = 1504 bits. So, there is always a measurement granularity of 1504 bits. The shorter the TS is, the larger difference you will find between the evaluated and theoretical value.
- On a VBR stream, the evaluation is not precise, essentially because when the bitrate is variable, there is no single bitrate value to extract from the stream or any PID.

Because this method is more or less precise, depending on the context, there is a possibility for some "higher authority" (ie. the "user") to specify a bitrate which is assumed to be more precise. In that case, this TS bitrate value is used to compute the individual bitrates of all PID's.

When the analysis is triggered from the command `tsanalyze`, this value is taken from the optional `--bitrate` parameter (hence, the name "user-defined"). If this parameter is unused, then the user-specified bitrate is simply "None".

When the analysis is triggered from the `analyze` plugin in command `tsp`, it depends on the input. Somehow, the input plugin evaluates the "user-defined" bitrate. This bitrate is propagated from plugin to plugin.

In decreasing order of precedence:
- There is a global option `tsp --bitrate` which says to all plugins "this is the right global bitrate to consider, ignore any other".
- In "reliable" input plugins which are able to compute an exact input bitrate, this one is used. This includes the `dvb` plugin which computes the theoretical bitrate from the modulation parameters or the `dektec` plugin which reads the actual bitrate from a hardware real-time device.
- On other inputs (`file`, `ip`, etc), when the input has no intrinsic bitrate, the `tsp` framework recomputes an evaluated bitrate based on PCR's from time to time. So, in essence, this is the same thing as the PCR-based evaluation in the analysis module. However, because the measurement points are not synchronized between `tsp` and the `analyze` plugin, because of granularity artefacts, the two values are not always identical, although relatively close.
- Additionally, be aware that intermediate plugins may remove packets (PID or service extraction, filtering, etc). In that cases, the global TS bitrate is modified (reduced) after these plugins. For performance reasons, the bitrate is not re-evaluated after each plugin. Consequently, the propagated input bitrate becomes incorrect. To solve this, the plugin `pcrbitrate` is a way to re-evaluate the bitrate of the TS at that point in the plugin chain, and propagates it downstream, ignoring the input bitrate after that point. Because the user is supposed to understand what he does, when a plugin needs to know the TS bitrate but a previous plugin broke the bitrate by removing packets, then an instance of `pcrbitrate` shall be inserted between the two.

At the end, the analysis module has a "selected reference bitrate" which either the user-specified one or the PCR-based one. In the context of `tsp`

Hope this helps...

edepa, 2024-05-20:
Thank you @lelegard , it was indeed helpful. 
Thank you also for the time you took to write in detail!

Please excuse my late acknowledgement; 
I was reading your reply and hoping to open an intelligent conversation, but 
there is so much to read and assimilate that
if I were to wait until I reassure myself that I have taken it all in,
I would probably only reply in a few days' time.

I hope to revert later this week 
with a comment affirming full understanding of your note.

closed 2024-10-27 by github-actions[bot]