tsduck/tsduck #1466: PSIP tables for ATSC

caritechsolutions, 2024-05-23:
Hi I'm new to tsduck and I'm sure it can help me create what I'm trying to do but I need a little guidance. 

The Plan is to use tsduck to generate a stream to which I can mux together with my content at an ATSC modulator. I want to get working epg on the television sets.

My plan was to extract the tables needed from a working stream so I could have a template to which I could write code to manipulate then insert these updated table back into a null stream with tsduck for me to mux with my content.

I managed to get some output from the working stream using the following:

`tsp -I ip 224.100.1.2:60500 -P tables --xml-output tables.xml`

my question is how would i inject my modified "tables.xml" into a udp stream?

I want to build on this command: `tsp -I null -P inject tables.xml`

Can you help point me in the right direction?

caritechsolutions, 2024-05-24:
I made some progress with the following command:
`tsp -vvvvv --bitrate 1600000 -I null -P inject --bitrate 600000 -p 8187 tables.xml=1000 -O ip 224.100.5.5:49500`

however I am having trouble controlling the output bitrate, it's flooding the interface. I will keep at it but I'm logging my progress here for others and for me to find later.

caritechsolutions, 2024-05-24:
Ok more progress here is what i have so far as a command line:
`tsp -vvvvvv --bitrate 600000 -I null -P regulate -P inject --bitrate 300000 -p 8187 tables2.xml=1000 -O ip  224.100.5.5:49500`
The regulate plugin is what i was missing so now i have control of the bitrates. 
My next step is to look into eitinject and how to update the eit tables

caritechsolutions, 2024-05-24:
Now I'm a little stuck. Current command is as follows:
`tsp -vvvvvv --bitrate 600000 -I null -P regulate -P inject --bitrate 300000 -p 8187 tvct_table.xml=1000 -P eitinject --bitrate 300000 --terrestrial --time system -f eit_table.xml=1000 -O ip  224.100.5.5:49500`

The issue I am having is I'm not seeing and EIT tables being generated. I'm not sure what rules need to be correct for this to happen. Can anyone point me in the right direction?

lelegard, 2024-05-24:
Hi @caritechsolutions 

You are progressing fast, congratulations 😃 

Concerning EIT's, this is a different story. First, there are two kinds of EIT: DVB EIT and ATSC EIT. Although they serve the same purpose (EPG), they are very different. The plugin `eitinject` processes DVB EIT's. It generates and sends them according to the specific rules of ETSI TS 101 211 (one of the DVB standards). Unlike other tables, the DVB EIT's are made of sparse (non contiguous) sections and need a special treatment. With `eitinject`, the input files are just collections of events, in the form or EIT's for simplicity but they are just input containers for events. The plugin stores these events and generates EIT's on the fly, according to DVB operation rules: as time progresses, the same event will be embedded in different types of EIT sections. Input events can be constantly updated by some program management system.

I am not familiar with ATSC EIT and their operation rules. There is currently no specific plugin to dynamically generate ATSC EIT's.

One last detail, `tsp -vvvvvv` is the same as `tsp -v` or `tsp --verbose`. The boolean options are not cumulative. If you want more details, you can use `tsp -d` or `tsp --debug`. Deeper debug levels can be reached with `-d2` or `--debug=2`, then 3, etc.

caritechsolutions, 2024-05-24:
ok thank you very much for the feedback. I found a work around to generate the ATSC EIT using the inject plugin as follows:

`tsp -v --control-local 127.0.0.1 --control-port 8000 --bitrate 600000 -I null -P regulate -P inject --bitrate 300000 -p 8187 tvct_table.xml=1000 -P inject --bitrate 300000 -p7424 eit_table.xml=1000 -O ip  224.100.5.5:49500`

Now my thoughts are since my video stream is completely separate from this one and only muxed at the modulator i could update the files with code because i need to change the system time in the following line of pid 8187:

`<STT protocol_version="0" system_time="1,400,555,990" GPS_UTC_offset="18" DS_status="false">`

then i can restart the plugin via code using the following command so the change is reflected in the stream:

`tspcontrol --tsp 127.0.0.1:8000 restart -s 2`

I can do the same with the EIT restarting the 3rd plugin if there is a change to the EIT data. That's what I'm thinking to try next, I was wondering if there is a better way to generate the system time dynamically. Hopefully my progress helps others!

lelegard, 2024-05-24:
Restarting a plugin on a regular basis is probably not a good idea. We don't know all potential consequences on the long term.

The dynamics of the various protocols are better handled through dedicated plugins. DVB is much more active than ATSC. There are plugins to dynamically generate EIT (`eitinject`) or continuously update the system time in the TS (`timeref`). Due to the lack of interest and lack of contributors, there is no equivalent for ATSC. If you are volunteer to implement the corresponding plugins, you are welcome.

The problem with ATSC is that it violates the MPEG rules and conventions on sections operations, for instance:
1. Use long sections for transient tables while they should use short sections. For instance, for the system time, the DVB TDT is a short section while the ATSC STT is a long section.
2. MPEG states that long sections should be identical when they have the same version number. ATSC violates that rule, changing the content of the STT while the version does not change. This breaks the section demux of a fully compliant MPEG receiver which triggers an analysis on version number change.
3. ATSC mixes distinct tables, with distinct life cycles on the same PID. For instance, repeatedly cycled static tables such as the MGT or xVCT are mixed in PID 0x1FFB with regularly updated tables such as the STT, making the update of a given table quite difficult in a plugin. DVB clearly separate dynamic PID's (time, running status) from static PID's (service description, network description). 

In short, the ATSC signalization was defined by people who did not understand the rules and spirit of MPEG sections, unlike DVB and ISDB. Maybe this is one of the reasons why there are not much interest in it. Maybe also this is one of the many reasons why ATSC 3 gave up the TS format, the ATSC TS specification was so broken that it was better to scrap it and restart with something completely different.

caritechsolutions, 2024-05-24:
I agree with you 100%,. Unfortunate for me, I live in the Caribbean and we shop in the USA so ATSC is what we have here. EPG is not mission critical for my project just really nice to have. I will run so test in my lab and report back my results here, where could I start if I wanted to help with plugins? Can I modify the current ones you mentioned?

Regards
Richard Rawlins

caritechsolutions, 2024-06-07:
Hi Lelegard,
                   So i have a working prototype that is rough but i can inject tables and get epg on an atsc 1.0 decoder. What I would like some help with is the following. I have been using a server which extracts the EPG as XMLTV from the output of my IRDs but it's super buggy and I have to restart it randomly to have the correct data presented for my code. So I decided to dig myself deeper down this rabbit hole by using tsduck to export from the udp out of the IRD pid 0x12 and save it as an xml.
                   
                   Here is the command I'm using `tsp -I ip 224.100.1.5:49500 -P tables -p 18 --xml-output eit01_table.xml`
 I however don't seem to be extracting the full amount of data from the transport and I'm sure it's due to my limited knowledge in this area. Can you point me in the right direction?

lelegard, 2024-06-07:
I am not familiar with ATSC EIT. Are they full tables, with no missing section? Or are they sparse sections, with holes in the section sequences, as DVB EIT?

I the latter case, you cannot directly extract them as XML since they are not full tables and XML can only be complete tables.

Alternatively, you may collect all sections individually and save them as binary or text. Use `-P tables --all-sections`. With `--binary-output outfile.bin` to save the binary content.

You may try `--pack-all-sections --xml-output outfile.xml` to "transform" each EIT section in a single-section table. This will break the section numbers but you will get all section as XML tables.

closed 2024-11-17 by github-actions[bot]