tsduck/tsduck #1469: [BUG] tsWinUtils.h missing when building on aarch64

omikron88, 2024-06-08:
When building tsduck on Rasbian 12 (Pi5), it ends with

setpath.cpp:24:10: fatal error: tsWinUtils.h: No such file or directory
   24 | #include "tsWinUtils.h"

#### How to reproduce:

git clone https://github.com/tsduck/tsduck.git
cd tsduck
scripts/install-prerequisites.sh
make

#### Environment:
 - OS: Linux aarch64
 - OS Raspbian 12 as downloaded from Raspberry Pi site

lelegard, 2024-06-09:
Hi @omikron88 

The tool `setpath` is a Windows-only tool. It is used in the TSDuck installer for Windows. It is not compiled on non-Windows systems, never with `make`.

In the directory `src/utils`, the makefile contains this:

~~~
EXECS := $(addprefix $(BINDIR)/,$(filter-out setpath $(if $(NOTEST)$(NOSTATIC),tsprofiling,),$(sort $(notdir $(basename $(wildcard *.cpp))))))
~~~

This means that `setpath` is explicitly filtered out of the list of executables to build.

You must have typed some distinct command. Please provide the complete log of your `make` command.

(and, by the way, because of the extra-long compilation time, always provide some level of parallelism, such as `make -j4` or `make -j10`, depending on your number of cores and memory size)

omikron88, 2024-06-10:
Thank you. It works fine now.
Compilation on Raspberry Pi5 is quite fast with make -j4

lelegard, 2024-06-10:
> Compilation on Raspberry Pi5 is quite fast with make -j4

And on my iMac M3, a full `make -j10` completes in 2mn 15s...
Parallelism is essential on large builds.

closed 2024-11-10 by github-actions[bot]