tsduck/tsduck #1473: [Q] replacing one PID in a service

petrn, 2024-06-18:
Hello,

I have a MPTS stream (or alternatively a SPTS stream) and I want to replace a single PID with a different content.

I am using this method:

```
tsp --synchronous-log --timed-log \
    -I ip --buffer-size $((1024*1024)) --local-address 172.26.12.7 232.1.1.112:5000 \
    -P zap 0x31B \
    -P svrename --name 'CNN test EXE' 0x31B \
    -P filter --pid 4372 --negate --stuffing \
    -P merge --ignore-conflicts --no-psi-merge --drop 0-4371 --drop 4372-0x1fff --pass 4372 --bitrate 130000 --transparent --max-queue 5000 --no-smoothing \
        "tsp -I ip --buffer-size $((1024*1024)) --local-address 172.26.25.7 231.100.9.52:5000" \
    -O ip --rtp  --packet-burst 7 --enforce-burst --pcr-pid 0x1109 --local-addr 172.26.25.7 239.239.239.240:5000
```

so I first select only 1 service from the MPTS, rename it, filter out the old content and then merge in the replacement content.
Is this the right way to do it? Is there a better way? What should be the bitrate value in the merge plugin - the rate before the filtering or after the filtering? My problem is that the new data are inserted, but they are late by several minutes, but I need to replace them synchronously.

The PID being replaced are DVB subtitles and this is for testing changes before deploying to production - the output stream goes into a modulator for checking.

lelegard, 2024-06-18:
Hello @petrn 

Your input TS has probably no stuffing (only broadcast has stuffing). You replace PID 4372 with stuffing. So, the free space in the TS is exactly the previous PID, no less, no more.  If the new content has a slightly higher bitrate, it does not fit and late packets accumulate.

First, you need to evaluate, at least approximately, how much more you need when replacing the old PID by the new one. Then, add artificial stuffing at input using option `tsp --add-input-stuffing`.

If you don't really know but you think that there is not a significant difference (if you don't replace SD with 4K), simply double the input bitrate to have margin. Because you use IP input, prefer chunks of 7 TS packets:  `tsp --add-input-stuffing 7/7`.

Then, just before the final `-O ip`, remove the stuffing in excess with `-P filter -n -p 0x1FFF`.

That is for the capacity. Then, this is a brutal merge, not a synchronized muxing. If the merge PID contains data, that should be fine. If it contains video, or even audio, you may have syncrhonization issues.

petrn, 2024-06-19:
Hello Thierry,

thanks a lot for the hint. The actual situation is that the input stream has plenty of stuffing (it is a DVB-T2 broadcast stream), but all that stuffing is removed by the zap plugin. As ```--add-input-stuffing``` is a global option, it will not help here, as any stuffing added here will be removed by the zap.

The subtitles stream we generate is effectively CBR, so replacing the old PID with the new one should be possible.

Would you please answer the question if the ```--bitrate``` on the merge plugin applies to the rate before applying the filter for the merge filter or after it?

lelegard, 2024-06-19:
I overlooked the plugin `zap`, sorry. In that case, use `-P zap --stuffing` to keep more insertion opportunities. And filter out the remaining stuffing before the output plugin.

petrn, 2024-06-19:
Yes, I did that and the delay has disappeared.

Tt seem the ```--bitrate``` needs to be the filtered one. The stream being merged has no PCR as it only has the subtitles, that is why the ```--bitrate``` needs to be present and correct.

closed 2024-11-24 by github-actions[bot]