tsduck/tsduck #1477: UDPReceiver Usage

mihaicodrean, 2024-06-28:
Hi!

I'm basically trying to achieve the following, suggested [here](https://github.com/tsduck/tsduck/issues/593#issuecomment-640199332):
> can also use the class UDPReceiver to read UDP input and class TSFile to write output TS file.

However, the UDPReceiver is not getting any packets apparently. I'm replaying as follows:
`ffmpeg.exe -re -i file.ts -c copy -f mpegts "udp://239.101.101.1:1234"`

And that's something I can render with `ffplay udp://239.101.101.1:1234` just fine.

Here's the stripped out code as I'm using the UDPReceiver, with the error handling code left out:
```
ts::UDPReceiver receiver;
receiver.setParameters(ts::IPv4SocketAddress::AnySocketAddress, true);
receiver.open(*this);
receiver.setReceiveTimeoutArg(100ms);
receiver.addMembershipAll(ip, ts::IPv4Address::AnyAddress, *this); // ip holds the multicast address
while (true)
{
   auto result = receiver.receive(packet.data(), packet.size(), actualSize, sender, destination, nullptr, *this);
   if (result)
       ... // never gets here
}
```
I'm not getting any errors from the calls above, nor from the parent class implementing ts::Report.

Can you please indicate what I might be doing wrong? For one, I found it odd that at some point I was forced to call setParameters, for the internal _receiver_specified to be set, because that method is marked for unicast use, not multicast:
> Set application-specified parameters to receive unicast traffic

lelegard, 2024-06-28:
Hi,

A `UDPRreceiver` is a subclass of `UDPSocket` with a few additional properties such as command line options definition and parsing. Most UDP parameters are taken from the command line. 

You may not use the command line parameters but, in that case, you need to specify all necessary UDP parameters. 

Here, you don't even specify on which port you are listening. So, you can't receive anything.

mihaicodrean, 2024-07-01:
Thanks for the reply.

> Here, you don't even specify on which port you are listening. So, you can't receive anything.

I was under the impression that I was supposed to filter by the port myself, due to the type of the 1st parameter in setParameters, IPv4SocketAddress, and its name, localAddress. I see now that it's actually an endpoint, so I've tried like this, but still not packets received:
```
receiver.setParameters(ts::IPv4SocketAddress(ts::IPv4Address::AnyAddress, 1234), true);
```
Could it be that the issue is setsockopt with IPPROTO_IP/IP_ADD_SOURCE_MEMBERSHIP, say misbehaving with "any address" for the source? In fact, in my test, I really don't care about the source of the multicast traffic, so IP_ADD_MEMBERSHIP could be used instead.

mihaicodrean, 2024-07-01:
The strange thing is that I don't get this logging from addMembership, that is called from addMembershipAll:
```
    if (local.hasAddress()) {
        report.verbose(u"joining multicast group %s from local address %s", {groupString, local});
    }
    else {
        report.verbose(u"joining multicast group %s from default interface", {groupString});
    }
```
I'll try to add extra logging in UDPSocket, maybe that will point to the issue.

mihaicodrean, 2024-07-01:
Bingo! I had to instruct ffmpeg to send the packets on one of the interfaces joined for multicast:

`ffmpeg.exe -re -i file.ts -c copy -f mpegts "udp://239.101.101.1:1234?localaddr=192.168.0.5"`

Now I get the packets.

mihaicodrean, 2024-07-01:
So the packets make it, but are filtered:
```
rejecting packet, destination: 239.101.101.1:1234, expecting: 0.0.0.0:1234
```
That's due to:
```
receiver.setParameters(ts::IPv4SocketAddress(ts::IPv4Address::AnyAddress, 1234), true);
```
If I try to specify the actual multicast address there, then I get:
```
error binding socket to local address: The requested address is not valid in its context.
```
So I guess I'm back to the question: should I really be using setParameters for multicast, since its comment mentions unicast?

lelegard, 2024-07-01:
Glad you made it work. 

By experience, when you have an issue between a sender and a receiver, the fist thing to do is running Wireshark. After analysis of what's on the wire (or the absence of it in that case), you immediately know where to look at.

mihaicodrean, 2024-07-01:
> Glad you made it work.

I'm actually almost there - please see my previous comment.

> By experience, when you have an issue between a sender and a receiver, the fist thing to do is running Wireshark. After analysis of what's on the wire (or the absence of it in that case), you immediately know where to look at.

Agreed, although WireShark has issues sometimes if replaying on "localhost" on Windows. :)

mihaicodrean, 2024-07-01:
It could be something like this:
```
//----------------------------------------------------------------------------
// Set application-specified parameters to receive multicast traffic.
//----------------------------------------------------------------------------

void ts::UDPReceiver::setMulticastParameters(const IPv4SocketAddress& multicastEndpoint, bool reusePort, size_t bufferSize)
{
    _receiver_specified = true;
    _use_ssm = false;
    _dest_addr = multicastEndpoint;
    //_local_address = localAddress;
    _reuse_port = reusePort;
    _recv_bufsize = bufferSize;
}
```

lelegard, 2024-07-03:
Hi @mihaicodrean 

The class `UDPReceiver` was initially designed to work with command line parameters. The `setParameters` method was added later to reuse the class on unicast servers without command line parameters. But it is not generic enough.

I plan to separate `UDPReceiver` and `UDPReceiverArgs`, just like `WebRequest` and `WebRequestArgs`. Then, you can use `UDPReceiverArgs` either as a C++ class where you set all your parameters or directly from the command line.

mihaicodrean, 2024-07-03:
That will be helpful, yes!

lelegard, 2024-07-04:
`UDPReceiverArgs` has been separated from `UDPReceiver`. The set of parameters can be defined/loaded from the command line or manually crafted by an application.

mihaicodrean, 2024-07-11:
Thanks! Setting the UDPReceiverArgs::destination and passing that into setParameters does the trick, and I can receive the multicast packets.

Do you have a rough ETA for the release of 3.38?

lelegard, 2024-07-11:
> Do you have a rough ETA for the release of 3.38?

There is no established planning. Probably somewhere around end of August, after the history rewrite of the repo (see #1472).

In the meantime, you may safely use the current repo state. There is no specific validation process for official releases. We rely on the continuous integration process which is applied on all pushes.

mihaicodrean, 2024-07-11:
Got it, thanks again!

closed 2024-07-11 by mihaicodrean