tsduck/tsduck #1480: Fast & lightweight way to detect the live stream type, e.g. audio & video

mihaicodrean, 2024-07-11:
Hi,

Is there a lightweight analyzer to detect the stream types for a live stream, just to be able to say if it contains video & audio?

I've seen that the TSAnalyzer::analyzePMT checks the PMT::Stream::stream_type, and even uses the StreamTypeIsAudio & StreamTypeIsVideo functions, but not sure if it's appropriate for the task at hand.

Thanks.

lelegard, 2024-07-11:
If you do not need a complete in-depth analysis, no need to use a `TSAnalyzer`. To be notified as soon as possible, use a `SignalizationDemux`. In your implementation of `SignalizationHandlerInterface`, just provide a `handlePMT` to get all PMT's, as soon as they are available. In the received `PMT` object, browse all `streams` and use methods `isAudio` or `isVideo` on each of them.

Checking the 8-bit stream type alone is not enough. It only works for some codecs (typically those which are standardized by MPEG). For other codecs, the stream type is something like "private data". You need to look for some special audio or video descriptor in the stream description. This is what methods `isAudio` and `isVideo` are doing.

mihaicodrean, 2024-07-11:
Sounds perfect, thanks for the fast & detailed reply! I'll try it out.

mihaicodrean, 2024-07-11:
It works like a charm, thanks again!

closed 2024-07-15 by mihaicodrean