tsduck/tsduck #1482: [Q] set top box

SonnyWalkman, 2024-07-14:
Is there a way to continually listen on a frequency for -

1. Build a tsp -I dvb ... -P analyse -i 5 -O drop
Await a valid sig lock and TS stream at which analyse sig for 5 seconds for the PAT for PMT and return the first service from the MPTS or only service in SPTS.

2. Restart tsp -I dvb .... -P zap <service id> -O ip ... 
use service id taken from 1. analyse file and begin streaming the service to udp as SPTS. All other services are ignored.

3. Once unlock is found. Re- No signal detected await 5 seconds kill tsp. Build a fresh tsp pipeline 1. Awaiting for lock then find service if and loop again.

lelegard, 2024-07-15:
Hi @SonnyWalkman 

There is no way to "wait for a valid signal lock" using the various API. You have to poll for it.

The rest is just a matter of shell scripting. What you could do is endlessly loop on something like:

- `rm -f ts.json`
- `tsp -I dvb --receive-timeout 5000 ... -P until --seconds 3 -P analyze --json -o ts.json -O drop`
- If `ts.json` does not exist, or is empty, wait a few seconds and loop back (this is polling for signal lock).
- `jq ... ts.json` to extract the service id or other relevant info you need.
- `tsp -I dvb --receive-timeout 5000 ... -P zap ... -O ip ...`
- Loop back (if you get there, this means that the signal drop was detected by the receive timeout`

Adjust the scenario and timeouts to your needs.

TVforME, 2024-07-16:
Hi @lelegard,

Thanks for your prompt reply..
I'll develop a script to do what you have described and let you know how I go.

Thanks for your valued knowledge.  I could look at using femon or femon2 to monitor frontend lock? I'll play around and see what works.

lelegard, 2024-07-16:
>  I could look at using femon or femon2 to monitor frontend lock? 

`femon` is a Linux application which does [polling and sleep](https://github.com/tbsdtv/dvb-apps/blob/e6ff7da39edd44fe6bb14bcba29c0fc5a02957a8/util/femon/femon.c#L136) on frontend status. It does not _wait_ for a signal lock.

TVforME, 2024-08-06:
Hi @lelegard,

My apologies I've resurfaces as @TVforME  and @SonnyWalkman have two github accounts.  I've been successful in polling the adapter frontend for stats using ioctl call to the dvb  api3 interface.. Thought I'd let you know and thanks for your comments.

closed 2024-08-06 by SonnyWalkman