tsduck/tsduck #1484: C++ package manager support

igem0n, 2024-07-19:
Hello guys!
I'm interested in using TsDuck for my project that is going to be crossplatform and I'm seeking for a handy and universal way to include it as a dependency into my cmake file. So support of any c++ package managers like **Conan** or **CPM** would be really great!
Thanks in advance!

lelegard, 2024-07-19:
There is no such thing at the moment. Mostly due to the lack of knowledge on how it works. We need a contribution from someone with sufficient familiarity with these package managers. 

At some point, I tried to learn "modern cmake" and how to apply it to TSDuck instead of the make / msbuild duality. But there were some requirements that cmake experts considered mandatory that were unacceptable for TSDuck. So, I dropped the effort.

igem0n, 2024-07-23:
Conan actually can work with prebuilt binaries also, so even without cmake it's possible. I'll try to create such package for my own use, as I want all the dependencies in my project to be available with Conan anyway and will share it with you, so you will be able to contribute to ConanCentral if you want

lelegard, 2024-07-23:
Ok, thanks !

The main problem I got with other build managers is the requirement to list exactly all source and header files in the project. In a project such as TSDuck, ~2000 files, new files coming in frequently, one single maintainer, this is an unacceptable requirement. The existence of a file at a given place in the source tree is a sufficient information. All explicit lists of source files are uselessly redundant. This works fine with GNU make and MSBuild. This is more or less possible with cmake but cmake zealots insist on the fact that this is bad practice. People advocating redundancy in the structure of a project have obviously no experience with large dynamic projects with few maintainers.

TVforME, 2024-08-06:
Hi @lelegard,

Have you considered meson and ninja.  I've used make however, meson is making way over Cmake, make these days.
Worth a look into..

lelegard, 2024-08-11:
> Have you considered meson and ninja?

@TVforME, not really. It took years to stabilize the build system for TSDuck using GNU Make and MSBuild. There is a lot of complexity in the combination of systems, system versions, compilers, compiler versions, etc. All upper-level build managers eventually end up generating makefiles and msbuild files (aka VS project files). So, why reinventing the wheel?

Moreover, the tutorials of all these tools demonstrate how magically easy it is to manage a "hello world" example. But TSDuck is way more complicated. Difficulties and peculiar situations appear at each step. My experience with CMake was quite disappointing and resulted in a waste of time. I do not feel like doing this again with another build tool.

I am open to new contributions. I am ready to evaluate a contribution of a TSDuck build system using meson or any other tool, as long as it works with the currently supported [20 operating systems, 3 compilers and 8 CPU architectures](https://tsduck.io/download/docs/tsduck-dev.html#_tested_systems), and support [all dependency options](https://tsduck.io/download/docs/tsduck-dev.html#buildopt). If the result is satisfactory and easier to maintain, I will be glad to integrate it. But I am personally not ready to invest weeks or months to replace something that works with something that _may_ work.

stazio, 2025-01-13:
I had some issues getting CLion to be content with the current build system (their Makefile support is very particular) and as a result I did end up writing a very very crude `meson.build`. It's available in https://github.com/stazio/tsduck/commit/d419936d385290bdd48613b518c119faf668b8fa#diff-a416f714db9122a987524b8ad1574c64f46693deaaecc53a3898b139046c481cR47. It currently disables whatever it can just to get it off the ground (i.e. SRT,RIST, Dektec, etc...). If there's some interest in this, I could look into expanding the functionality.

Meson is a very two-edged sword where its very powerful and provides good solutions to most common build/project problems, but its also very opinionated so god forbid you try and do something they don't want you to do.

SonnyWalkman, 2025-01-13:
Hi there @stazio 
I believe there is merit in continuing seeing you put effort in give the meson build system a shot.

closed 2025-06-15 by github-actions[bot]