tsduck/tsduck #1486: Solve my doubt[Q]

udhayNuk, 2024-07-26:
tsp -v -m \
    -I hls --live --connection-timeout 900000 --receive-timeout 900000 http://192.168.0.1/live/stream10/index.m3u8 \
    -P regulate --pcr-synchronous \
    -O hls --duration 3 --live 10 -p index.m3u8 index.ts 
    
how to use this option    --infinite, i want to run this stream without stopping, please tell me suggestion in other methods to stream
@lelegard please reply me.

lelegard, 2024-07-26:
There is no `--infinite` option with the `hls` input plugin. Live streams never end by definition. If there is a network error or a server failure, `tsp` will terminate. If you want to continue and retry on error, the simplest way is to use a shell script and loop back to the same command when it terminates.

udhayNuk, 2024-07-26:
@lelegard thank u for replying me, i have got this error how to solve this.
 
* Error: hls: synchronization lost after 93,293,055 packets, got 0x3C instead of 0x47
using tsresync --continue in lower script the input can receive tsresync and process it or not?

tsp -I hls --live --connection-timeout 900000 --receive-timeout 900000 http://192.168.0.101/live/stream3/index.m3u8 -P regulate --pcr-synchronous | tsresync --continue | tsp -v -m -O hls --duration 3 --live 10 -p index.m3u8 index.ts

lelegard, 2024-07-26:
The loss of synchronization is most likely due to the http server not returning a segment correctly. 

Use `tsp --debug` to see the http headers with the potential error status.

udhayNuk, 2024-07-27:
@lelegard 
tsswitch  -v -m 
-I hls --live --connection-timeout 900000 --receive-timeout 900000 http://192.168.0.1/live/stream10/index.m3u8
-O hls --duration 3 --live 10 -p index.m3u8 index.ts

by using this any problem will occur for live streams, give me some suggestions.