tsduck/tsduck #1487: How to hold the mpegts stream for few seconds?

Avencheng, 2024-08-01:
How to hold the stream for 5 seconds without changing frams's PTS ? 
I'm not too sure which command i should use, I have tried the following command
```
aven.cheng@tgcusers-MacBook-Pro ~ % tsp -I ip 235.35.3.5:3535 -P timeshift --time 5000 -O ip 235.35.3.5:3536

* Warning: timeshift: unknown initial bitrate, discarding packets until a valid bitrate can set the buffer size
```
But i found that frames's PTS are not the same anymore. The PTS of each frame is reduced by 1.2 seconds.

lelegard, 2024-08-08:
Hi @Avencheng 

The plugin `timeshift` delays a fixed number of TS packets. Using `--time` needs to know the TS bitrate to convert that time into a number of packets.

On CBR streams (broadcast), there is a strict equivalence between time and packets. On VBR streams (as most IP streams are), this can be inaccurate. The initial measured bitrate, based on PCR values, is used to convert the time into a number of packets. That number of packets will be buffered all the way down. If the initially measured bitrate is not significant, there could be some inaccuracy in the time shift.

You can safely ignore the warning message about "unknown initial bitrate". With IP input, the initial buffer load is very small (typically one UDP datagram). The plugin is just saying that it needs more TS packets to correctly evaluate the bitrate and compute the timeshift buffer size in TS packets. Before that, packets are dropped. If you see output packets, this means that the bitrate was quickly evaluated.

How do you conclude that PTS were modified? The `timeshift` plugin does no modify the TS packets, it does not change the PTS values. The packets are just delayed.

Avencheng, 2024-08-09:
Oops! you are right! the PTS is unchanged. I think the way I checked the result was wrong before.
Thank you for your detailed explanation, that's very helpful.

closed 2025-01-12 by github-actions[bot]