tsduck/tsduck #1488: [Q] Minimal TSduck build. What files do I need to build a compact install for Raspberry Pi CM4 (Pi 5)?

TVforME, 2024-08-06:
Hello all and @lelegard,

I’m back with another TSduck-related query, which might have been covered before, but after scanning past issues, I think my question is somewhat unique. Additionally, the link to the "how-to-build TSduck" in the README is broken, so any guidance there would be helpful.

Background
I am using TSduck for my Repeater receiver frontend project. This project is essentially a software-derived STB with a web-hosted OSD, used in my Repeater Project to receive DVB-T and DVB-S signals from amateur TV experimenters, like myself. My call sign is VK3DG in Australia. The OSD is alpha-keyed over the incoming video feed, similar to a standard STB, but with a much cooler, customizable display using CSS, HTML, and JavaScript.

While there are existing software solutions like TVheadend, DVBlast, and MuMudvb, none quite fit my needs. Therefore, I developed my own solution, using TSduck as the underlying engine. I monitor frontend stats using custom code to poll each adapter via ioctl through the dvb api3 API.

My project is programmed in Go, which integrates well with C++ and C libraries. I am currently developing bindings for TSduck to avoid making exec calls and sifting through stdio buffers. While this method works, it is not ideal.

This is my first major repository, so I appreciate your understanding as I continue to enhance it with additional logging and features. It has been running continuously for months on both a Linux laptop (i5 Gen 9) with 2 USB DVB adapters (DVB-T) and a Raspberry Pi CM4 with 2 TBS 6522H PCIe adapters on a PCIe x1 switch. So far, everything is working well, thanks to TSduck.

Here is my repo:  [Repeater Receiver](https://github.com/TVforME/Repeater-receiver)

**My Question**
Can I build a minimal TSduck toolkit with just the components I need: TSP, DVB, IP output RTP, zap, until, and analyze? I attempted to build with options against make (e.g., NOVATEK=1, NOHIDES=1, NOJAVA=1), however, VATEK seems to be pulled in during the build, causing errors. Not sure if it a bug or my understanding of the TSduck build process?

If a minimal build is not possible, what files are necessary for TSduck to function? 

A list of required versus optional files would be helpful. My aim is to develop a tcz package for PiCore64, which is a IOT OS Linux running entirely in RAM, so minimising the footprint for TSduck is crucial for quick loading of the package at startup. If I have a minimal build, it be more than enough to get me going. 

I do not need HiDES, DEKTEC, VATEK, or Crypto components since my usage involves using the tsp processor and a few of my chosen plugins. 
`tsp -I dvb and -O ip  and the plugins analyze, zap, and until.`

Additionally, having a Go binding to interact with the TSduck API directly is a future goal for me. I had a bit of a fiddle, however, ran into issue with C++17 and configuring CGO to use C++17 over the default build essentials. I’ll save that for another day.

**Appreciation**
I want to extend my thanks to @lelegard for creating such an excellent toolkit and for his ongoing efforts in bug fixes, feature additions, and maintenance. I encourage anyone using my code to show appreciation by making a small donation to Thierry’s cause. I have done so multiple times and will continue to do so. 

Maybe I could convince Theirry to look into adding a Teletextinject plugin for TSDuck  @SonnyWalkman

Thanks again for your assistance.

lelegard, 2024-08-08:
Hi @TVforME

> the link to the "how-to-build TSduck" in the README is broken, so any guidance there would be helpful.

Sorry for that. The documentation has been updated in Asciidoc format and the build instructions are now a [section of the developer's guide](https://tsduck.io/download/docs/tsduck-dev.html#building). I will update the README asap.

> My project is programmed in Go, which integrates well with C++ and C libraries. I am currently developing bindings for TSduck to avoid making exec calls and sifting through stdio buffers. 

If you could make these Go bindings aligned with the [Python and Java bindings](https://tsduck.io/download/docs/tsduck-dev.html#jpbindings), in terms of features, that would be a great contribution. I have currently no Go knowledge. Go and Rust would be my next two languages to learn but I have unfortunately no time for that.

> Can I build a minimal TSduck toolkit with just the components I need: TSP, DVB, IP output RTP, zap, until, and analyze?

This is not possible because it would not be useful. TSDuck has been designed as a general purpose library. All commands and plugins are very small wrappers on top of the library. Removing unused commands and plugins would not change the library size and would not significantly reduce the total binary size.

> I attempted to build with options against make (e.g., NOVATEK=1, NOHIDES=1, NOJAVA=1),

These options are not designed to reduce the code size. They are designed to remove specific dependencies on some external libraries which may not be installed on a given system.

> however, VATEK seems to be pulled in during the build, causing errors.

Did you try on the latest repo state? I recall something like that on a previous version and that was fixed. I can build with `NOVATEK=1` on the current repo state.

> If a minimal build is not possible, what files are necessary for TSduck to function?

Hard to tell. There are around 2000 C++ source files in TSDuck. Tracking all possible interdependencies and supporting options to selectively remove them, in a consistent and reliable way, on all platforms, would be too complicated and error prone. TSDuck is a lab tool for laptop or server systems. There is no support for small embedded devices.

A very large subset of the code base (around 700 source files) is dedicated to the tables and descriptors. Many tables and most descriptors can be removed without beaking the build. But there is no precise list. In short, any table or descriptor which is not directly referenced as a C++ class elsewhere can be removed. However, the corresponding structures will no longer be correctly parsed and become unusable as XML or JSON structure.

> ran into issue with C++17

The TSDuck code base has been recently modernized and now requires C++17. There are requirements on the minimum version of each compiler. [See there for more details](https://tsduck.io/download/docs/tsduck-dev.html#cpprequnix).

TVforME, 2024-08-12:
Hi @lelegard,
 Thanks again for you relay Theirry. 

> Sorry for that. The documentation has been updated in Asciidoc format and the build instructions are now a [section of the developer's guide](https://tsduck.io/download/docs/tsduck-dev.html#building). I will update the README asap.

I assumed your work in porting to ascii.doc there was always going to be something forgotten or missed in a large change. No issues at all.

> If you could make these Go bindings aligned with the [Python and Java bindings](https://tsduck.io/download/docs/tsduck-dev.html#jpbindings), in terms of features, that would be a great contribution. [...]

I'll look a getting the tsp --version working. I believe there has been other discussions about adding a Go binding? I'm using TSduck in my receiver project using call the command line through exec. It works however, to be able to interact with the lib directly has benefits. I've read the developers info on TSduck  and I'm beginning to understand how it code is structures for the base and TSduck plugins. I'll try to implement my teletext inject plugin that myself and @SonnyWalkman as been banging on about there is already a working solution. vbit2 handles the generation of pages, taking a dir of teletext pages, converting each page from a tti file to T42 binary, then creating a full magazine of pages in specific order. pages prioritised send page 100 more offed then adding PID and PCR timestamps before sending through stdio..  A plugin solution removes the output to stdio and merges into the TS directly in a TSduck style plugin.  Probably 10 people on the planet want to use Teletext in their project?

 vbit2 is coded in C++ therefore there maybe opportunity to take parts of this working design and turn into a Plugin?  vbit2 developer supported my request to ADD  the TS option `--format ts`   which add PID and PCR timestamps to the teletext stream to be compliant with DVB. The add-on saved the extra code (python) to add the PCR.  I have vbit2 feeding into _P merge (tsp ...)  into the TS with modification to the PAT adding the service info etc.
Despite, it is working, I'd prefer a  TSduck plugin way of implementation.   -P teletextinject  [options]  -i (path to pages dir)  
Teletext my seem dead however, I'm just one to keep it alive in DVB.

 If you have a teletext decode why not teletext encode? I understand its time.  Under the teletext folder include tsTeletexMux.cpp and split the parity and hamming84 to be common libs? Add additional libs to create the PES and PCR addition etc.  I'll look at it further once I understand how TSduck operates under the hood.

> These options are not designed to reduce the code size. They are designed to remove specific dependencies on some external libraries which may not be installed on a given system.

My intention was not to reduce the size of the install. For my specific build I had no requirement for HiDes, Dektec and VATEK hardware on the box, why install the drivers if there never a need to use this hardware?   Not clear how to implement the flags correctly without diving into the Make system to see how the flags change the build conditions? I don't believe NOJAVA is a flag option? 

> Hard to tell. There are around 2000 C++ source files in TSDuck. Tracking all possible interdependencies and supporting options to selectively remove them, in a consistent and reliable way, on all [...]

> A very large subset of the code base (around 700 source files) is dedicated to the tables and descriptors. Many tables and most descriptors can be removed without breaking the build. But there is no precise list. In short, any table or descriptor which is not directly referenced as a C++ class elsewhere can be removed. However, the corresponding structures will no longer be correctly parsed and become unusable as XML or JSON structure.

My simple solution is to build and copy across the so files needed. Test and, if it breaks, then add what is missing till it works. Let you know where I get with this approach? Trial by Elimination in reverse. Add what I believe is base and core till is works?

> The TSDuck code base has been recently modernized and now requires C++17. There are requirements on the minimum version of each compiler. [See there for more details](https://tsduck.io/download/docs/tsduck-dev.html#cpprequnix).

Thanks for the link.
The above was my issue with attempting to use cgo to build function binds to the TSduck C++ library for Go.  Ran into issues with Linux build environment as it was for using C++12 even with setting the correct flags to use C++17.  Left it for the time being.  I'll get back to it once I have time.
Seriously the whole C++ build process is complicated and hard to navigate if like me aren't a fulltime C++ dev. C and C++ still haven't got the handling of strings right. Let's not go there, for a debate. 
Reason why I've taken to developing and learning in Go. All of the compilation and linking is hidden and it just works. No include files mumbo jumbo. The program dependency system is brilliant.
I know I keep banging on about Go however, I had more success in building an application over trying to learn all the undelaying C++ coding style with a slight hit to performance. Just so much is already under the hood.
Go being Statically Typed modern language with value pointer etc is checked at compilation time.  Few simple commands to build for different architectures. I just love it coming from C embedded system and all the rage of OO with C++  C# etc did my head in. Structures are my thing. Classes and overrides and inherency I find over complicated.

lelegard, 2024-08-22:
> If you have a teletext decode why not teletext encode?

There is no generic teletext decoder, just teletext _subtitles_ extraction. My selfish original goal was to record movies from TV on DVD (after the death of the venerable VHS cassettes) and a SubRip file was necessary to burn the subtitles into the video using ffmpeg, before authoring the DVD.

> I don't believe NOJAVA is a flag option?

That's right. However, if the JDK is not installed (no Java compiler), the Java bindings are not compiled. No error is generated.

Adding `NOJAVA` was not considered necessary, unlike NOPCSC, NOSRT and the like, because compiling the Java bindings does not create a specific dependency in the TSDuck binaries. 

Let's take the example of SRT or RIST. If `libsrt` or `librist` is not installed on the build system, the corresponding features will be silently ignored, without error, just like Java. However, if these libraries are present on the build system but maybe not on the deployment systems, we have a problem because the binaries will reference a library which may not be present, and running TSDuck will fail. This is why `NOSRT` and `NORIST` are defined, to avoid creating dependencies on these libraries, _even if they are present on the build system_.

With Java, the situation is different. The structure of JNI (Java Native Interface) is such that it creates no binary dependency. So, if the JDK is present in the build system and TSDuck is built with the Java bindings, there will be no error on deployment systems without JRE. You just can't run Java applications using TSDuck (but without JRE you won't _any_ Java application anyway).

> My simple solution is to build and copy across the so files needed. Test and, if it breaks, then add what is missing till it works. Let you know where I get with this approach? Trial by Elimination in reverse. Add what I believe is base and core till is works?

It can be time consuming. I would recommend to evaluate if you really need that. Excessive pre-optimization is usually not a good idea. Did you try the default build? Did you encounter a specific problem? Did you evaluate the cost of this problem? 

> Reason why I've taken to developing and learning in Go. All of the compilation and linking is hidden and it just works. No include files mumbo jumbo. The program dependency system is brilliant.

Probably, as long as you stay into a known domain. Many tools or languages are like that. It's wonderfully easy to build and port an "hello world" application. Problems start with the interface to external libraries where you have to reimplement all the bindings. At the beginning of my career, I worked for 15 years in Ada, Ada83, then Ada95. Very reliable, robust, hidden complexity of compilation and link, etc. As long as you wrote 100% of the software and did not rely on external non-Ada components. New languages appear and history restarts...

> Structures are my thing. Classes and overrides and inherency I find over complicated.

That is exactly what I was thinking 20 years ago, including in TSDuck v1 (which was developed in C). Until I realized that all the complicated things I did with structures were an unsafe and error-prone reimplementation of OOD. Moving to C++ and OOD, most of the low-level code disappeared, the development speed doubled, the debug time disappeared, etc.

TVforME, 2024-08-23:
Ho @lelegard,

> There is no generic teletext decoder, just teletext subtitles extraction. My selfish original goal was to record movies from TV on DVD (after the death of the venerable VHS cassettes) and a SubRip file was necessary to burn the subtitles into the video using ffmpeg, before authoring the DVD.

I've used ffmpeg and know about the subrip format. In addition to SubRip I believe ffmpeg if compiles against libzvbi library can decode and encode teletext however, I'm yet to see anyone accomplish encoding with ffmpeg? 
Seems to be "why would you want to subject"   There are a few Teletext experimenters willing to keep the long-forgotten format alive on DVB.  As you well know, Teletext is a valid auxiliary data service in the DVB transport stream.

I've been using GStreamer exclusively in my development.  For anything TS related GStreamer provides a few plugin to handle TS. One I've come to have a love/hate relationship with is the  mpegtsmux plugin. Looking through the code base for mpegtsmux indicates it able to encode a 45 byte teletext PES and adds the PCR taken for the video source. 

[Link to teletext prepare code](https://gitlab.freedesktop.org/gstreamer/gstreamer/-/blob/main/subprojects/gst-plugins-bad/gst/mpegtsmux/gstbasetsmuxttxt.c)

I'm yet to play with getting a GStreamer pipeline to accept teletext and actually multiplex in with video and audio SPTS?
Only early days however, encoding Teletext to an existing service and  IS the preferred method.  If I can find and solution and get working, then I'd be less requesting a teletext encode plugin from you OR once I'm familiar, develop such plugin for TSduck.

> I don't believe NOJAVA is a flag option?

Thanks for clearing up the NOJAVA and other flags in the build process. I wasn't sure make system taking into account for dependencies on the system during the build phase. Good to know. 

> Reason why I've taken to developing and learning in Go. All of the compilation and linking is hidden and it just works. No include files mumbo jumbo. The program dependency system is brilliant.

I completely understand why C++ over  vanilla C. Everyone has their "flavour" of development and C++ is yours. 
I've tried Rust and couldn't get my head around it's complex borrows and mutex's guess Go is for me.

Thanks for you ongoing support.

closed 2025-01-26 by github-actions[bot]