tsduck/tsduck #1490: Warning: History rewrite of the repository

lelegard, 2024-08-10:
Dear TSDuck developers and contributors,

This note is an advanced warning about a big bang in the https://github.com/tsduck/tsduck repository. Only developers and contributors are impacted, more generally anyone who cloned or forked the TSDuck source code repository.

The big bang consists in a _history rewrite_ of the Git repository. The reasons for this operation have been explained in issue #1472.

The operation is planned for __Saturday August the 17th, 2024__.

Feel free to comment or report problems in this thread before or after the operation.

## How to handle the change ?

Nothing should be done before the operation. You can perform the change any time after the history rewrite is completed, after August the 17th.

### TSDuck users who install prebuilt packages

There is nothing to do. Users who never accessed the source code repository are not impacted by the change.

### TSDuck users who rebuild TSDuck without modification

If you have cloned the main TSDuck repository and use it to rebuild TSDuck, without modification in the source code, the simplest method is to delete your local workarea and clone the repository again. Since the updated repository is much smaller, the new clone operation is fast.

~~~
rm -rf tsduck
git clone https://github.com/tsduck/tsduck.git
~~~

### TSDuck contributors with forked repositories

If you do not have any modification in your forked repositories, maybe the simplest method is to delete your forked repository on Github and  delete all the corresponding cloned workareas in your local systems. Then, fork the TSDuck repository again and clone your forked repository.

Otherwise, if you want to preserve your repositories, there are two required steps:

- Fix your forked repository on GitHub.
- Fix your local cloned workareas for that forked repository.

#### Fix your forked repository on GitHub

On https://github.com, go to your forked repository, typically something like https://github.com/USER/tsduck.

On branch `master`, you will see some message such as _"This branch is 3747 commits ahead of 3756 commits behind tsduck/tsduck:master"_. Click on "Sync fork" and select "Discard 3747 commits". Note that this is just an example, the number of commits may be different.

If you have other personal branches, they cannot be fixed from the Github Web interface. They must be fixed from the local workarea.

#### Fix your local cloned workareas

On each workarea which has been cloned from your forked repository, we assume that you did not modify the `master` branch and implemented your changes in dedicated personal branches. In this case, you need to synchronize the `master` branch and manually rebuild your personal branches.

I have not found a simple, generic and reliable way of fixing the specific branches. If you have recommendations and advices, please post them in this thread.

If you have old dedicated branches which were already merged into the main TSDuck repository, I recommend to simply delete them, locally and on your remote forked repository. There is no need to keep them.

For all dedicated branches with pending work, the basic idea is to rebuild them manually:

- Checkout the dedicated branch.
- Make a recursive copy of the workarea into a temporary directory, outside the repository.
- Checkout `master` back.
- Delete the local branch and its remote counterpart in the fork repository.

Then, synchronize the `master` branch. The idea is to reset the workarea to the very first commit of the repository and resynchronize from the remote forked repository:

~~~
git checkout master
git status # verify that the working tree is clean
git reset --hard 62f6f9cab1bdd96eb8b415c01b6c276ba92285ef
git pull
~~~

Finally, for each dedicated branch for which you saved a copy:

- Recreate the branch and checkout it.
- Cherry pick the files you modified or created from the local copy.
- Commit the files in the new branch.

<hr/>

The immediately following posts in this thread list all contributors and fork owners in order to notify them of this announce. The list is split over several posts to comply with the notification limitations of Github issues.

lelegard, 2024-08-10:
For notification (see the first post of this thread):

@1div0 @5l1v3r1 @851896022 @a038537 @a83912a @abainbridge @ajithdesilva @akshhack @alevinsn @AlfonZ42 @Alx-I @amkoxie @amoscmc @anto35 @antoineagscom @anylee2021 @aplathan @aqu1ntero @arunlodhi @As-You-Like @asdlei99 @ASTRO-Strobel @athoik @atvcaptain @aulion @Avetri @axiu-fly @axm06051 @barsnick @basicmaster @bbc @bentoinfor @bentoti @bigbao9494 @bilalk @BillOatmanWork @blowfish64 @bluetarpmedia @bojanra @bpalanimuthu @braghetto @bravesoftdz @breezetemple @BrinedFish @brotherbus @bruuteforce @calabashdad @casalazarh @caveman99 @cedinu

lelegard, 2024-08-10:
For notification (see the first post of this thread):

@ceelona @chibbert @cmassiot @CongraTV @ConnectionMaster @david-maison-TrustInSoft @djsdevel @dutchboontje @duyouhua @ElyKar @emcodemall @enricovittorini @ensdlpyt @EricDeJong @ethouris @evandropoa @fabsconnect @fairywh @felipewd @fennewald @freeflyfish @freesatdevops @gangbanlau @ganqiuye @garethfenn @gdams @GERD0GDU @ggnull35 @github-jinwei @Goga1992 @haifeng2170 @Hakai-Shin @HammingByte @harmonicinc-com @hit2000 @hygortc @imjoy @inrg @ioxuy @izharm @jakub-zwolakowski @jco1904 @jdlyga @jindapol2562 @jlyga3 @jmsn @jyyoung89 @Kamilcuk @kevleyski @Kirk-007

lelegard, 2024-08-10:
For notification (see the first post of this thread):

@kishorekumar-anchala @kmdewaal @kth-xx @Kudos5 @kxp @l3akage @lance-lmwang @lars18th @leanmeanaudio @leobin @LiminWang @limkokholefork @Littlelies @liuchuxiong @locke12456 @Lordevss @lumovan @macczy @MadManLabs @MarkMaYJ @MartinFretigne @Marware @masnagam @matveysudomoykin @maxha651 @maxkao @mekhti @mich181189 @MikeGogia @minyuwang @mirakc @misiu-git @mkll @MortazaHassani @moveman @musingwei @mypopydev @Navezjt @neevek @newrain7803 @npcomplete13 @nstuardod @nuclidjp @nxtreaming @oottppxx @oviano @paraeco @paulhiggs @petervictory @phillipreeder

lelegard, 2024-08-10:
For notification (see the first post of this thread):

@piaoxue85 @piotrserafin @poijas @potterxu @pvdheijden @Qencode-Live-Streaming-Project @RaviAkshintala @rborgese @reckless-sergio @rishabhFC @rkrishna1 @rnshah9 @RodrigoPrintes @rodrigovb96 @RonWuYi @rtk4616 @rtpriya @rubenochiavone @sahwar @samarasteph @Saurabh-Evertz @selvakumaran @shaheryaramir @shivambhatia5432 @sichunqin @smartclip @spike008t @spnngl @SS-FF @st-channel24 @standback @Strefa35 @sukesh-ak @svlobanov @Taadeus @tarelda @tectronics @thiennguu @TUISTERa @tysos @ubaldus @ufo-chang @UNUWARE @vacing @vcofre @vector-x-labs @verb5 @VisionAdvanceTech @vtns @vvhh2002

lelegard, 2024-08-10:
For notification (see the first post of this thread):

@wangzf100 @wdelstrother @willdelstrother @witxjp @wongfei2009 @xuxiaowei007 @yrpark99

lelegard, 2024-08-16:
Dear all,

The Big Bang has happened... The history has been rewritten and pushed on GitHub. And the Earth seems to be still present.

For reference (or in case of trouble), the original history has been saved in repository https://github.com/tsduck/tsduck-archive-before-history-rewrite which must be considered as read-only.

Feel free to discuss the consequences of this history rewrite in this thread.

paulhiggs, 2024-08-17:
Thanks Thierry for the update.

This morning, my fork ([https://github.com/paulhiggs/tsduck](https://github.com/paulhiggs/tsduck)) shows “This branch is 3772 commits ahead of, 3756 commits behind tsduck/tsduck:master.”

Does that sound right – should I go ahead as sync against tsduck/tsduck?

lelegard, 2024-08-17:
Hi Paul, yes, this is what I explained in the first post of this thread. Since I am sure you have quite a lot of pending development on other branches, I am afraid you will have to cherry pick your changes and copy them back in fresh new branches after you complete the synchronisation of 1) your forked repo and 2) its clone on your local system. Of course, I recommend that you back up your local repo with unmerged code before doing this. Just in case.

lelegard, 2024-08-22:
TSDuck version 3.38-3822 has been released yesterday, based on the new rewritten history.

I hope that no one suffered from the history rewriting operation. Lessons being learnt, this kind of operation will likely never happen again.

lelegard, 2024-10-12:
Now closing in the absence of negative feedback

closed 2024-10-12 by lelegard