tsduck/tsduck #1491: [Q] Rust bindings

SteveMcFarlin, 2024-08-10:
Has anyone attempted to create rust bindings (via bindgen)? I wanted to ask before attempting to do this. Thanks.

lelegard, 2024-08-11:
Hi @SteveMcFarlin

I am not aware of any Rust bindings for TSDuck, at least not publicly.

Currently, in addition to the native complete C++ interface, TSDuck has [Java and Python bindings](https://tsduck.io/download/docs/tsduck-dev.html#jpbindings) only, for a subset of high-level features. There was a proposed [JavaScript binding](https://github.com/tsduck/tsduck/pull/875) but the build integration was unfortunately never completed by the contributors.

The main requirement for valuable bindings in another language is object-oriented capabilities. TSDuck as a strictly object-oriented design, internally and in the user API. TSDuck defines abstract classes (or _interfaces_ in Java parlance) and uses them to dispatch demuxing, events, logging, plugin interface, etc. The application implements subclasses through which it is notified of everything.

The Java and Python bindings define the corresponding classes so that an application in these languages can subclass them and get notified. The `sample` subdirectory in the TSDuck source tree contains Java and Python small applications which intentionally have the same structure to illustrate the similarities.

I am not knowledgeable enough in Rust to evaluate the feasibility of equivalent bindings. From what I have read on Rust, the language has no class or inheritance features (feel free to correct me if I am wrong). It only has _traits_. Even though the two concepts are vaguely related, inheritance and traits are very different. C++ has both and they are used for distinct purposes. Similarly, Java and Ada have _generics_ and inheritance for distinct purposes.

Therefore, designing valuable Rust bindings for TSDuck would require an experienced Rust expert to adapt the API concepts. Maybe it is possible, I don't know. It you are volunteer, I would be glad to help. Just let me know.

Of course, for a dedicated usage, it is still possible to write specialized C++ _functions_ and interface them with Rust. It can help for a dedicated Rust project but this can hardly be defined as Rust bindings for TSDuck.

SteveMcFarlin, 2024-08-11:
I took a look into autocxx, and also browsed the core of TSDuck. I honestly think your last point is the path forward for usage of TSDuck in a Rust application.

Yes you are correct that rust has no notion of a class as most OO languages implement (Essentially C structs is all). While trait objects do implement V-Tables, it is far from a 1-1 mapping to a C++ class. At this point I think it _may_ be possible to implement the API bindings in Rust, but would probably not be overall useful and quite possibly a pain to work with. 

Thank you for the response, and for writing and maintaining such a great library.

closed 2024-08-12 by SteveMcFarlin