tsduck/tsduck #1492: Install tsduck on Ubuntu 24.04

Avencheng, 2024-08-12:
I'm trying to install tsduck on Ubuntu 24.04.
I used the following steps to install:
1. `git clone https://github.com/tsduck/tsduck.git`
2. `bash tsduck/scripts/install-prerequisites.sh`
3. `make -C tsduck installer`(following are the messages after i executed this command)
```
:
:
  [CXX] tsscan.cpp
  [CXX] tssmartcard.cpp
  [CXX] tsstuff.cpp
  [CXX] tsswitch.cpp
  [CXX] tstabcomp.cpp
  [CXX] tstabdump.cpp
  [CXX] tstables.cpp
  [CXX] tsterinfo.cpp
  [CXX] tstestecmg.cpp
  [CXX] tsvatek.cpp
  [CXX] tsversion.cpp
  [CXX] tsxml.cpp
  [GEN] tsduck-completion.bash
  [LD] /home/test/tsduck/bin/release-x86_64-cuda/tscrc32
  [LD] /home/test/tsduck/bin/release-x86_64-cuda/tsdate
  [LD] /home/test/tsduck/bin/release-x86_64-cuda/tsdektec
  [LD] /home/test/tsduck/bin/release-x86_64-cuda/tsdump
  [LD] /home/test/tsduck/bin/release-x86_64-cuda/tsfixcc
  [LD] /home/test/tsduck/bin/release-x86_64-cuda/tsfuzz
  [LD] /home/test/tsduck/bin/release-x86_64-cuda/tshides
/usr/bin/ld: warning: libedit.so.0, needed by /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libedit.so.0, needed by /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so, not found (try using -rpath or -rpath-link)
  [LD] /home/test/tsduck/bin/release-x86_64-cuda/tslatencymonitor
/usr/bin/ld: warning: libedit.so.0, needed by /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so, not found (try using -rpath or -rpath-link)
  [LD] /home/test/tsduck/bin/release-x86_64-cuda/tslsdvb
/usr/bin/ld: warning: libedit.so.0, needed by /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so, not found (try using -rpath or -rpath-link)
  [LD] /home/test/tsduck/bin/release-x86_64-cuda/tsp
  [LD] /home/test/tsduck/bin/release-x86_64-cuda/tspacketize
/usr/bin/ld: warning: libedit.so.0, needed by /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `read_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `using_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `read_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `using_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `write_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `write_history/usr/bin/ld'
: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `stifle_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `stifle_history'
/usr/bin/ld: warning: libedit.so.0, needed by /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `readline'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `add_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `readline'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `add_history'
/usr/bin/ld: warning: libedit.so.0/usr/bin/ld, needed by /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so, not found (try using -rpath or -rpath-link)
: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `read_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `using_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `write_history'
collect2: error: ld returned 1 exit status
make[5]: *** [../../Makefile.inc:619: /home/test/tsduck/bin/release-x86_64-cuda/tsdump] Error 1
make[5]: *** Waiting for unfinished jobs....
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `stifle_history'
collect2: error: ld returned 1 exit status
make[5]: *** [../../Makefile.inc:619: /home/test/tsduck/bin/release-x86_64-cuda/tsdate] Error 1
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `readline'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `add_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `read_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `using_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `write_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `stifle_history'
collect2: error: ld returned 1 exit status
make[5]: *** [../../Makefile.inc:619: /home/test/tsduck/bin/release-x86_64-cuda/tscrc32] Error 1
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `readline'
/usr/bin/ld: warning: libedit.so.0, needed by /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so, not found (try using -rpath or -rpath-link)
[... 45 lines cut ...]
make[5]: *** [../../Makefile.inc:619: /home/test/tsduck/bin/release-x86_64-cuda/tslatencymonitor] Error 1
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `read_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `using_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `write_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `stifle_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `read_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `readline'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `add_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `using_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `write_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `stifle_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `readline'
collect2: error: ld returned 1 exit status
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `add_history'
make[5]: *** [../../Makefile.inc:619: /home/test/tsduck/bin/release-x86_64-cuda/tsp] Error 1
collect2: error: ld returned 1 exit status
make[5]: *** [../../Makefile.inc:619: /home/test/tsduck/bin/release-x86_64-cuda/tslsdvb] Error 1
make[4]: *** [Makefile:14: default] Error 2
make[3]: *** [Makefile:42: default] Error 2
make[2]: *** [Makefile:24: deb-tools] Error 2
make[1]: *** [Makefile:19: installer] Error 2
make: *** [Makefile:155: installer] Error 2
make: Leaving directory '/home/test/tsduck'
```
Please let me know how can i fix it? Thanks!

lelegard, 2024-08-12:
@Avencheng 

There is something wrong on your system with the installation of packages for `libedit`. Did you get any error message when running `install-prerequisites.sh`?

On Ubuntu 24.04, you should have packages `libedit-dev` and `libedit2`. The dev package is probably installed, otherwise you would get an error message during compilation, failing to find the header files. However, there is a link error "libedit.so.0 not found".

The problem is that Ubuntu 24.04 contains `libedit.so.2`, not `libedit.so.0`. The reason why the linker tries to look for `libedit.so.0` instead of `.2` is unclear. This is a system configuration issue, not a TSDuck issue.

I would suggest to update the system (`apt update; apt upgrade`), remove and reinstall libedit, etc.

Avencheng, 2024-08-13:
Hi ! Thanks for your suggestion. I have updated the system (apt update; apt upgrade)

But I'm not quite sure how to remove the previous installation?  I just move the tsduck folder to the trash bin...
and redo the following steps:
1. `git clone https://github.com/tsduck/tsduck.git`
2. `bash tsduck/scripts/install-prerequisites.sh`(following are the messages)
```
(base) test@cuda:~$ bash tsduck/scripts/install-prerequisites.sh
Packages: git g++ cmake flex bison dos2unix curl tar zip linux-libc-dev dpkg-dev python3 libssl-dev asciidoctor qpdf libedit-dev pcscd libpcsclite-dev libcurl4 libcurl4-openssl-dev librist-dev libsrt-openssl-dev libusb-1.0-0-dev default-jdk doxygen graphviz
[sudo] password for test: 
Get:1 file:/var/cuda-repo-ubuntu2204-12-5-local  InRelease [1,572 B]
Get:1 file:/var/cuda-repo-ubuntu2204-12-5-local  InRelease [1,572 B]                       
Hit:2 http://security.ubuntu.com/ubuntu noble-security InRelease                           
Hit:3 http://archive.ubuntu.com/ubuntu noble InRelease                                     
Get:4 http://archive.ubuntu.com/ubuntu noble-updates InRelease [126 kB]          
Get:5 https://ppa.launchpadcontent.net/graphics-drivers/ppa/ubuntu noble InRelease [24.1 kB]
Hit:6 http://archive.ubuntu.com/ubuntu noble-backports InRelease     
Fetched 150 kB in 2s (70.9 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libcurl4t64' instead of 'libcurl4'
git is already the newest version (1:2.43.0-1ubuntu7.1).
g++ is already the newest version (4:13.2.0-7ubuntu1).
cmake is already the newest version (3.28.3-1build7).
flex is already the newest version (2.6.4-8.2build1).
bison is already the newest version (2:3.8.2+dfsg-1build2).
dos2unix is already the newest version (7.5.1-1).
curl is already the newest version (8.5.0-2ubuntu10.2).
tar is already the newest version (1.35+dfsg-3build1).
zip is already the newest version (3.0-13build1).
linux-libc-dev is already the newest version (6.8.0-40.40).
dpkg-dev is already the newest version (1.22.6ubuntu6).
python3 is already the newest version (3.12.3-0ubuntu1).
libssl-dev is already the newest version (3.0.13-0ubuntu3.2).
asciidoctor is already the newest version (2.0.20-1).
qpdf is already the newest version (11.9.0-1.1build1).
libedit-dev is already the newest version (3.1-20230828-1build1).
pcscd is already the newest version (2.0.3-1build1).
libpcsclite-dev is already the newest version (2.0.3-1build1).
libcurl4t64 is already the newest version (8.5.0-2ubuntu10.2).
libcurl4-openssl-dev is already the newest version (8.5.0-2ubuntu10.2).
librist-dev is already the newest version (0.2.10+dfsg-2).
libsrt-openssl-dev is already the newest version (1.5.3-1build2).
libusb-1.0-0-dev is already the newest version (2:1.0.27-1).
default-jdk is already the newest version (2:1.21-75+exp1).
doxygen is already the newest version (1.9.8+ds-2build5).
graphviz is already the newest version (2.42.2-9ubuntu0.1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Successfully installed asciidoctor-pdf-2.3.18
Parsing documentation for asciidoctor-pdf-2.3.18
Done installing documentation for asciidoctor-pdf after 0 seconds
Successfully installed rouge-4.3.0
Parsing documentation for rouge-4.3.0
Done installing documentation for rouge after 1 seconds
2 gems installed
```
3. `make -C tsduck installer`(following are the messages)
```
(base) test@cuda:~$ make -C tsduck installer
make: Entering directory '/home/test/tsduck'
rm -rf /home/test/tsduck/pkg/installers/tmp
make -C /home/test/tsduck -j28 NOTEST=true
: 
:
:
[... 189 lines cut ...]
/usr/bin/ld: warning: libedit.so.0, needed by /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `read_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `using_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `write_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `stifle_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `readline'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `add_history'
collect2: error: ld returned 1 exit status
make[5]: *** [../../Makefile.inc:619: /home/test/tsduck/bin/release-x86_64-cuda/tsp] Error 1
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `read_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `using_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `write_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `stifle_history'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `readline'
/usr/bin/ld: /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so: undefined reference to `add_history'
collect2: error: ld returned 1 exit status
make[5]: *** [../../Makefile.inc:619: /home/test/tsduck/bin/release-x86_64-cuda/tspacketize] Error 1
make[4]: *** [Makefile:14: default] Error 2
make[3]: *** [Makefile:42: default] Error 2
make[2]: *** [Makefile:24: deb-tools] Error 2
make[1]: *** [Makefile:19: installer] Error 2
make: *** [Makefile:155: installer] Error 2
make: Leaving directory '/home/test/tsduck'
```
It still has the same problem, do you know how to deal with it?

lelegard, 2024-08-13:
Please post the output of the following commands (with my outputs as reference):

~~~
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04 LTS
Release:	24.04
Codename:	noble
$
$ dpkg -l libedit\*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                   Version              Architecture Description
+++-======================-====================-============-======================================================
ii  libedit-dev:amd64      3.1-20230828-1build1 amd64        BSD editline and history libraries (development files)
ii  libedit2:amd64         3.1-20230828-1build1 amd64        BSD editline and history libraries
ii  libeditorconfig0:amd64 0.12.7-0.1           amd64        coding style indenter across editors - library
$
$ ls -l /usr/lib/x86_64-linux-gnu/libedit*
-rw-r--r-- 1 root root 368304 Apr  8 18:02 /usr/lib/x86_64-linux-gnu/libedit.a
lrwxrwxrwx 1 root root     25 Apr  1 20:31 /usr/lib/x86_64-linux-gnu/libeditorconfig.so.0 -> libeditorconfig.so.0.12.7
-rw-r--r-- 1 root root  26616 Apr  1 20:31 /usr/lib/x86_64-linux-gnu/libeditorconfig.so.0.12.7
lrwxrwxrwx 1 root root     17 Apr  8 18:02 /usr/lib/x86_64-linux-gnu/libedit.so -> libedit.so.2.0.72
lrwxrwxrwx 1 root root     17 Apr  8 18:02 /usr/lib/x86_64-linux-gnu/libedit.so.2 -> libedit.so.2.0.72
-rw-r--r-- 1 root root 220472 Apr  8 18:02 /usr/lib/x86_64-linux-gnu/libedit.so.2.0.72
$ 
~~~

I can help but you have to realize that you have a system issue on your Ubuntu, this is not a TSDuck issue. If you cannot find the solution, you will have to get help from Ubuntu guys. The core problem is the following message:

~~~
/usr/bin/ld: warning: libedit.so.0, needed by /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so, not found (try using -rpath or -rpath-link)
~~~

And the question is: where does the reference to `libedit.so.0` come from? The problem being the trailing `.0` because `libedit.so.0` does not exist and no one should request it. The default `libedit` is `libedit.so.2`.

lelegard, 2024-08-13:
In your log, `libtsduck.so` seems to have been built already. The errors occur when linking the executables. So please report the output of:
~~~
$ ldd /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so
~~~

For reference, this is my output:
~~~
	linux-vdso.so.1 (0x00007fff0afb3000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x0000782387b8d000)
	libcrypto.so.3 => /lib/x86_64-linux-gnu/libcrypto.so.3 (0x0000782385c00000)
	libpcsclite.so.1 => /lib/x86_64-linux-gnu/libpcsclite.so.1 (0x0000782387b81000)
	libedit.so.2 => /lib/x86_64-linux-gnu/libedit.so.2 (0x0000782387b47000)
	libcurl.so.4 => /lib/x86_64-linux-gnu/libcurl.so.4 (0x000078238613f000)
	libsrt.so.1.5 => /lib/x86_64-linux-gnu/libsrt.so.1.5 (0x0000782385b37000)
	librist.so.4 => /lib/x86_64-linux-gnu/librist.so.4 (0x0000782385b04000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x0000782385800000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x0000782385717000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x0000782385ad7000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x0000782385400000)
	/lib64/ld-linux-x86-64.so.2 (0x0000782387bc5000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x0000782385aa4000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007823856e3000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x0000782387b2d000)
	libnghttp2.so.14 => /lib/x86_64-linux-gnu/libnghttp2.so.14 (0x0000782386114000)
	libidn2.so.0 => /lib/x86_64-linux-gnu/libidn2.so.0 (0x0000782385a82000)
	librtmp.so.1 => /lib/x86_64-linux-gnu/librtmp.so.1 (0x00007823856c5000)
	libssh.so.4 => /lib/x86_64-linux-gnu/libssh.so.4 (0x0000782385654000)
	libpsl.so.5 => /lib/x86_64-linux-gnu/libpsl.so.5 (0x0000782385640000)
	libssl.so.3 => /lib/x86_64-linux-gnu/libssl.so.3 (0x0000782385356000)
	libgssapi_krb5.so.2 => /lib/x86_64-linux-gnu/libgssapi_krb5.so.2 (0x0000782385302000)
	libldap.so.2 => /lib/x86_64-linux-gnu/libldap.so.2 (0x00007823852a5000)
	liblber.so.2 => /lib/x86_64-linux-gnu/liblber.so.2 (0x0000782385630000)
	libzstd.so.1 => /lib/x86_64-linux-gnu/libzstd.so.1 (0x00007823851eb000)
	libbrotlidec.so.1 => /lib/x86_64-linux-gnu/libbrotlidec.so.1 (0x0000782385622000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007823851cf000)
	libmbedcrypto.so.7 => /lib/x86_64-linux-gnu/libmbedcrypto.so.7 (0x000078238514a000)
	libcjson.so.1 => /lib/x86_64-linux-gnu/libcjson.so.1 (0x0000782385618000)
	libcap.so.2 => /lib/x86_64-linux-gnu/libcap.so.2 (0x000078238513d000)
	libmd.so.0 => /lib/x86_64-linux-gnu/libmd.so.0 (0x000078238512e000)
	libunistring.so.5 => /lib/x86_64-linux-gnu/libunistring.so.5 (0x0000782384f81000)
	libgnutls.so.30 => /lib/x86_64-linux-gnu/libgnutls.so.30 (0x0000782384d87000)
	libhogweed.so.6 => /lib/x86_64-linux-gnu/libhogweed.so.6 (0x0000782384d3f000)
	libnettle.so.8 => /lib/x86_64-linux-gnu/libnettle.so.8 (0x0000782384cea000)
	libgmp.so.10 => /lib/x86_64-linux-gnu/libgmp.so.10 (0x0000782384c66000)
	libkrb5.so.3 => /lib/x86_64-linux-gnu/libkrb5.so.3 (0x0000782384b9d000)
	libk5crypto.so.3 => /lib/x86_64-linux-gnu/libk5crypto.so.3 (0x0000782384b71000)
	libcom_err.so.2 => /lib/x86_64-linux-gnu/libcom_err.so.2 (0x0000782385612000)
	libkrb5support.so.0 => /lib/x86_64-linux-gnu/libkrb5support.so.0 (0x0000782384b64000)
	libsasl2.so.2 => /lib/x86_64-linux-gnu/libsasl2.so.2 (0x0000782384b4a000)
	libbrotlicommon.so.1 => /lib/x86_64-linux-gnu/libbrotlicommon.so.1 (0x0000782384b27000)
	libp11-kit.so.0 => /lib/x86_64-linux-gnu/libp11-kit.so.0 (0x0000782384983000)
	libtasn1.so.6 => /lib/x86_64-linux-gnu/libtasn1.so.6 (0x000078238496d000)
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x0000782384966000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x0000782384953000)
	libffi.so.8 => /lib/x86_64-linux-gnu/libffi.so.8 (0x0000782384947000)
~~~

You can see that the reference to `libedit.so` is `.2`, not `.0`.

The first references, up to `libc`, are explicitly required by `libtsduck`. All the rest are recursive dependencies of the first libs.

Avencheng, 2024-08-13:
I really appreciate your help.
following are the outputs:
```
(base) test@cuda:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04 LTS
Release:	24.04
Codename:	noble
(base) test@cuda:~$ dpkg -l libedit\*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                   Version              Architecture Description
+++-======================-====================-============-======================================================
ii  libedit-dev:amd64      3.1-20230828-1build1 amd64        BSD editline and history libraries (development files)
ii  libedit2:amd64         3.1-20230828-1build1 amd64        BSD editline and history libraries
ii  libeditorconfig0:amd64 0.12.7-0.1           amd64        coding style indenter across editors - library
(base) test@cuda:~$ 
(base) test@cuda:~$ ls -l /usr/lib/x86_64-linux-gnu/libedit*
-rw-r--r-- 1 root root 368304 Apr  9 00:02 /usr/lib/x86_64-linux-gnu/libedit.a
lrwxrwxrwx 1 root root     25 Apr  2 02:31 /usr/lib/x86_64-linux-gnu/libeditorconfig.so.0 -> libeditorconfig.so.0.12.7
-rw-r--r-- 1 root root  26616 Apr  2 02:31 /usr/lib/x86_64-linux-gnu/libeditorconfig.so.0.12.7
lrwxrwxrwx 1 root root     17 Apr  9 00:02 /usr/lib/x86_64-linux-gnu/libedit.so -> libedit.so.2.0.72
lrwxrwxrwx 1 root root     17 Apr  9 00:02 /usr/lib/x86_64-linux-gnu/libedit.so.2 -> libedit.so.2.0.72
-rw-r--r-- 1 root root 220472 Apr  9 00:02 /usr/lib/x86_64-linux-gnu/libedit.so.2.0.72
(base) test@cuda:~$ ldd /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so
	linux-vdso.so.1 (0x00007ffcb0df9000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x0000797a3c5b3000)
	libcrypto.so.3 => /lib/x86_64-linux-gnu/libcrypto.so.3 (0x0000797a3a600000)
	libpcsclite.so.1 => /lib/x86_64-linux-gnu/libpcsclite.so.1 (0x0000797a3c5a7000)
	libedit.so.0 => not found
	libcurl.so.4 => /lib/x86_64-linux-gnu/libcurl.so.4 (0x0000797a3ab3f000)
	libsrt.so.1.5 => /lib/x86_64-linux-gnu/libsrt.so.1.5 (0x0000797a3a537000)
	librist.so.4 => /lib/x86_64-linux-gnu/librist.so.4 (0x0000797a3c572000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x0000797a3a200000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x0000797a3a117000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x0000797a3c545000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x0000797a39e00000)
	/lib64/ld-linux-x86-64.so.2 (0x0000797a3c5e6000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x0000797a3a504000)
	libnghttp2.so.14 => /lib/x86_64-linux-gnu/libnghttp2.so.14 (0x0000797a3ab14000)
	libidn2.so.0 => /lib/x86_64-linux-gnu/libidn2.so.0 (0x0000797a3c521000)
	librtmp.so.1 => /lib/x86_64-linux-gnu/librtmp.so.1 (0x0000797a3a4e6000)
	libssh.so.4 => /lib/x86_64-linux-gnu/libssh.so.4 (0x0000797a3a0a6000)
	libpsl.so.5 => /lib/x86_64-linux-gnu/libpsl.so.5 (0x0000797a3a4d0000)
	libssl.so.3 => /lib/x86_64-linux-gnu/libssl.so.3 (0x0000797a39d56000)
	libgssapi_krb5.so.2 => /lib/x86_64-linux-gnu/libgssapi_krb5.so.2 (0x0000797a3a052000)
	libldap.so.2 => /lib/x86_64-linux-gnu/libldap.so.2 (0x0000797a39cf9000)
	liblber.so.2 => /lib/x86_64-linux-gnu/liblber.so.2 (0x0000797a3a4c0000)
	libzstd.so.1 => /lib/x86_64-linux-gnu/libzstd.so.1 (0x0000797a39c3f000)
	libbrotlidec.so.1 => /lib/x86_64-linux-gnu/libbrotlidec.so.1 (0x0000797a3a4b0000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x0000797a3a494000)
	libmbedcrypto.so.7 => /lib/x86_64-linux-gnu/libmbedcrypto.so.7 (0x0000797a39bba000)
	libcjson.so.1 => /lib/x86_64-linux-gnu/libcjson.so.1 (0x0000797a3a48a000)
	libcap.so.2 => /lib/x86_64-linux-gnu/libcap.so.2 (0x0000797a3a47d000)
	libunistring.so.5 => /lib/x86_64-linux-gnu/libunistring.so.5 (0x0000797a39a0d000)
	libgnutls.so.30 => /lib/x86_64-linux-gnu/libgnutls.so.30 (0x0000797a39813000)
	libhogweed.so.6 => /lib/x86_64-linux-gnu/libhogweed.so.6 (0x0000797a397cb000)
	libnettle.so.8 => /lib/x86_64-linux-gnu/libnettle.so.8 (0x0000797a39776000)
	libgmp.so.10 => /lib/x86_64-linux-gnu/libgmp.so.10 (0x0000797a396f2000)
	libkrb5.so.3 => /lib/x86_64-linux-gnu/libkrb5.so.3 (0x0000797a39629000)
	libk5crypto.so.3 => /lib/x86_64-linux-gnu/libk5crypto.so.3 (0x0000797a3a022000)
	libcom_err.so.2 => /lib/x86_64-linux-gnu/libcom_err.so.2 (0x0000797a3a01c000)
	libkrb5support.so.0 => /lib/x86_64-linux-gnu/libkrb5support.so.0 (0x0000797a3961c000)
	libsasl2.so.2 => /lib/x86_64-linux-gnu/libsasl2.so.2 (0x0000797a39602000)
	libbrotlicommon.so.1 => /lib/x86_64-linux-gnu/libbrotlicommon.so.1 (0x0000797a395df000)
	libp11-kit.so.0 => /lib/x86_64-linux-gnu/libp11-kit.so.0 (0x0000797a3943b000)
	libtasn1.so.6 => /lib/x86_64-linux-gnu/libtasn1.so.6 (0x0000797a39425000)
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x0000797a3a013000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x0000797a39412000)
	libffi.so.8 => /lib/x86_64-linux-gnu/libffi.so.8 (0x0000797a39404000)
(base) test@cuda:~$ 
```

lelegard, 2024-08-13:
And if you don't find anything relevant, try this:

~~~
$ find /usr/lib -name '*.so' -o -name '*.so.*' | xargs strings -f | grep libedit.so.0
~~~

On my Ubuntu 24.04 system, it prints nothing. If you find something, you have a problem.

Avencheng, 2024-08-13:
```
(base) test@cuda:~$ find /usr/lib -name '*.so' -o -name '*.so.*' | xargs strings -f | grep libedit.so.0
(base) test@cuda:~$ 
```
My one prints nothing, too.

lelegard, 2024-08-13:
That is really weird:

~~~
(base) test@cuda:~$ ldd /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so
...
	libedit.so.0 => not found
~~~

Also try this to get the full build command:
~~~
$ rm /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so
$ make -j10 V=1
~~~

You need to understand where the explicit reference to the non-existent file `libedit.so.0` come from.

Avencheng, 2024-08-13:
Here are the outputs:
```
(base) test@cuda:~$ rm /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so
(base) test@cuda:~$ make -j10 V=1
make: *** No targets specified and no makefile found.  Stop.
(base) test@cuda:~$ cd tsduck
(base) test@cuda:~/tsduck$ make -j10 V=1
  [DTAPI] using /home/test/tsduck/bin/LinuxSDK/DTAPI/Lib/GCC7.5.0/DTAPI64.o
cd /home/test/tsduck/bin/release-x86_64-cuda/objs-libtsduck; \
g++ -Wl,-soname=libtsduck.so,-z,noexecstack         -pthread  -Wl,-rpath,'$ORIGIN',-z,noexecstack tsAACDescriptor.o tsAC3Attributes.o tsAES128.o tsAES256.o tsAFExtensionsDescriptor.o tsAIT.o tsARIBCharset.o tsARIBCharsetData.o tsARIBCharsetEncoding.o tsATSCAC3AudioStreamDescriptor.o tsATSCEAC3AudioDescriptor.o tsATSCEIT.o tsATSCMultipleString.o tsATSCStuffingDescriptor.o tsATSCTimeShiftedServiceDescriptor.o tsAV1VideoDescriptor.o tsAVCAccessUnitDelimiter.o tsAVCAttributes.o tsAVCHRDParameters.o tsAVCParser.o tsAVCSequenceParameterSet.o tsAVCTimingAndHRDDescriptor.o tsAVCVUIParameters.o tsAVCVideoDescriptor.o tsAVS3VideoDescriptor.o tsAbortInterface.o tsAbstractAVCAccessUnit.o tsAbstractAudioVideoAttributes.o tsAbstractDatagramInputPlugin.o tsAbstractDefinedByStandards.o tsAbstractDeliverySystemDescriptor.o tsAbstractDemux.o tsAbstractDescrambler.o tsAbstractDescriptor.o tsAbstractDescriptorsTable.o tsAbstractDownloadContentDescriptor.o tsAbstractDuplicateRemapPlugin.o tsAbstractHEVCAccessUnit.o tsAbstractHTTPInputPlugin.o tsAbstractLogicalChannelDescriptor.o tsAbstractLongTable.o tsAbstractMultilingualDescriptor.o tsAbstractNetworkAddress.o tsAbstractNumber.o tsAbstractOutputStream.o tsAbstractPacketizer.o tsAbstractPreferredNameIdentifierDescriptor.o tsAbstractPreferredNameListDescriptor.o tsAbstractReadStreamInterface.o tsAbstractSignalization.o tsAbstractTable.o tsAbstractTablePlugin.o tsAbstractTransportListTable.o tsAbstractVVCAccessUnit.o tsAbstractVideoAccessUnit.o tsAbstractVideoData.o tsAbstractVideoStructure.o tsAbstractWriteStreamInterface.o tsAccessUnitIterator.o tsAdaptationFieldDataDescriptor.o tsAncillaryDataDescriptor.o tsAnnouncementSupportDescriptor.o tsApplicationDescriptor.o tsApplicationIconsDescriptor.o tsApplicationNameDescriptor.o tsApplicationRecordingDescriptor.o tsApplicationSharedLibrary.o tsApplicationSignallingDescriptor.o tsApplicationStorageDescriptor.o tsApplicationUsageDescriptor.o tsAreaBroadcastingInformationDescriptor.o tsArgMix.o tsArgs.o tsArgsWithPlugins.o tsAssociationTagDescriptor.o tsAsyncReport.o tsAsyncReportArgs.o tsAudioComponentDescriptor.o tsAudioLanguageOptions.o tsAudioPreselectionDescriptor.o tsAudioStreamDescriptor.o tsAustraliaLogicalChannelDescriptor.o tsAuxiliaryVideoStreamDescriptor.o tsBAT.o tsBCD.o tsBIT.o tsBasicLocalEventDescriptor.o tsBetterSystemRandomGenerator.o tsBinaryTable.o tsBitRateRegulator.o tsBitrateDifferenceDVBT.o tsBlockCipher.o tsBlockCipherAlertInterface.o tsBlockCipherProperties.o tsBoardInformationDescriptor.o tsBoolPredicate.o tsBouquetNameDescriptor.o tsBroadcasterNameDescriptor.o tsBuffer.o tsByteBlock.o tsC2BundleDeliverySystemDescriptor.o tsC2DeliverySystemDescriptor.o tsCAContractInfoDescriptor.o tsCADescriptor.o tsCAEMMTSDescriptor.o tsCAIdentifierDescriptor.o tsCASFamily.o tsCASMapper.o tsCASSelectionArgs.o tsCAServiceDescriptor.o tsCAT.o tsCDT.o tsCIAncillaryDataDescriptor.o tsCIT.o tsCOM.o tsCPCMDeliverySignallingDescriptor.o tsCPDescriptor.o tsCPIdentifierDescriptor.o tsCRC32.accel.o tsCRC32.o tsCVCT.o tsCableDeliverySystemDescriptor.o tsCableEmergencyAlertTable.o tsCaptionServiceDescriptor.o tsCarouselCompatibleCompositeDescriptor.o tsCarouselIdentifierDescriptor.o tsCellFrequencyLinkDescriptor.o tsCellListDescriptor.o tsCerrReport.o tsChannelFile.o tsCharset.o tsCodecType.o tsCommandLine.o tsCommandLineHandler.o tsComponentDescriptor.o tsComponentNameDescriptor.o tsConditionalPlaybackDescriptor.o tsConfigFile.o tsConfigSection.o tsConsoleState.o tsContentAdvisoryDescriptor.o tsContentAvailabilityDescriptor.o tsContentDescriptor.o tsContentIdentifierDescriptor.o tsContentLabellingDescriptor.o tsContinuityAnalyzer.o tsCopyrightDescriptor.o tsCountryAvailabilityDescriptor.o tsCryptoLibrary.o tsCueIdentifierDescriptor.o tsCyclingPacketizer.o tsDCCArrivingRequestDescriptor.o tsDCCDepartingRequestDescriptor.o tsDCCSCT.o tsDCCT.o tsDES.o tsDIILocationDescriptor.o tsDSMCCStreamDescriptorsTable.o tsDSNGDescriptor.o tsDTGGuidanceDescriptor.o tsDTGHDSimulcastLogicalChannelDescriptor.o tsDTGLogicalChannelDescriptor.o tsDTGPreferredNameIdentifierDescriptor.o tsDTGPreferredNameListDescriptor.o tsDTGServiceAttributeDescriptor.o tsDTGShortServiceNameDescriptor.o tsDTSDescriptor.o tsDTSHDDescriptor.o tsDTSNeuralDescriptor.o tsDTVProperties.o tsDVBAC3Descriptor.o tsDVBAC4Descriptor.o tsDVBCISSA.o tsDVBCSA2.o tsDVBCharTable.o tsDVBCharTableSingleByte.o tsDVBCharTableUTF16.o tsDVBCharTableUTF8.o tsDVBCharset.o tsDVBDTSUHDDescriptor.o tsDVBEnhancedAC3Descriptor.o tsDVBHTMLApplicationBoundaryDescriptor.o tsDVBHTMLApplicationDescriptor.o tsDVBHTMLApplicationLocationDescriptor.o tsDVBJApplicationDescriptor.o tsDVBJApplicationLocationDescriptor.o tsDVBServiceProminenceDescriptor.o tsDVBStuffingDescriptor.o tsDVBT2ParamsEvaluator.o tsDVBTimeShiftedServiceDescriptor.o tsDataBroadcastDescriptor.o tsDataBroadcastIdDescriptor.o tsDataComponentDescriptor.o tsDataContentDescriptor.o tsDataStreamAlignmentDescriptor.o tsDebugPlugin.o tsDefaultAuthorityDescriptor.o tsDeferredAssociationTagsDescriptor.o tsDektecArgsUtils.o tsDektecControl.o  [... cut ...]
collect2: error: ld returned 1 exit status
make[2]: *** [../../Makefile.inc:619: /home/test/tsduck/bin/release-x86_64-cuda/tsbitrate] Error 1
make[1]: *** [Makefile:14: default] Error 2
make: *** [Makefile:42: default] Error 2
(base) test@cuda:~/tsduck$ ldconfig -p | grep libedit
	libeditorconfig.so.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libeditorconfig.so.0
	libedit.so.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libedit.so.2
	libedit.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libedit.so
```
The main error still the same.

lelegard, 2024-08-13:
We may have something here. In your full build command for `libtsduck.so`, there is a suspect and unexpected option `-L/home/test/anaconda3/lib`.

What is the directory `/home/test/anaconda3`?

Please post the output of the following commands:
~~~
find /home/test/anaconda3/lib -name 'libedit*'
echo $PATH
echo $LD_LIBRARY_PATH
which curl-config
curl-config --cflags
curl-config --libs
~~~

Avencheng, 2024-08-13:
Oh ! Anaconda is the application that I use to manage virtual environment.
I ran the installation in the`(base)` environment. That might be the problem.
Here are the outputs:
```
(base) test@cuda:~/tsduck$ find /home/test/anaconda3/lib -name 'libedit*'
echo $PATH
echo $LD_LIBRARY_PATH
which curl-config
curl-config --cflags
curl-config --libs
/home/test/anaconda3/lib/libedit.a
/home/test/anaconda3/lib/pkgconfig/libedit.pc
/home/test/anaconda3/lib/libedit.so.0.0.72
/home/test/anaconda3/lib/libedit.so
/home/test/anaconda3/lib/libedit.so.0
/home/test/anaconda3/bin:/home/test/anaconda3/condabin:/usr/local/cuda/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
:/usr/local/cuda/lib64
/home/test/anaconda3/bin/curl-config
-I/home/test/anaconda3/include
-L/home/test/anaconda3/lib -lcurl

```

Avencheng, 2024-08-13:
I have deactivated the `(base)` environment and ran the following command again, the errors are no longer occurring.
```
$ rm /home/test/tsduck/bin/release-x86_64-cuda/libtsduck.so
$ make -j10 V=1
```
And i ran `make -C tsduck installer`, no errors, either.
Now i have two `.deb` under the following path.
```
test@cuda:~$ ls tsduck/pkg/installers
README.md  tsduck_3.38-3817.ubuntu24_amd64.deb  tsduck-dev_3.38-3817.ubuntu24_amd64.deb
```
Should i run the following command to install them?
```
$ sudo apt install tsduck_3.38-3817.ubuntu24_amd64.deb
$ sudo apt install tsduck-dev_3.38-3817.ubuntu24_amd64.deb
```
Sorry for lots of questions and thanks for the help.

Avencheng, 2024-08-14:
Thanks! Now i can use tsduck on my Ubuntu 24.04.

closed 2024-08-14 by Avencheng