tsduck/tsduck #1496: [Q] How do I add Audio Description to a TS file?

Mewankyrshan, 2024-08-18:
I read about Audio and Audio Description and from what I have understood, they are differnet and have their own descriptior tags(since audio description has to play at the same time as audio and its not a different audio choice), how do i add audio description .aac file to a ts file? I searched online everywhere and i cannot find sample files so the next best thing is to generate my own ts file. I need this file to test a media playback code to check if it can parse and play Audio Description.(I did not write the code, its some old company code so i am not sure how it works, but i need to check if i can use it or not).

lelegard, 2024-08-18:
Hi @Mewankyrshan 

If you want to add an audio track to an existing service, you need a MUX. TSDuck is not a MUX.

TSDuck is able to update the signalization and add all required descriptors in the PMT for instance. However, the muxing of the new audio track, from a .aac file or whatever, needs another tool, typically ffmpeg or equivalent.

Mewankyrshan, 2024-08-20:
I could add the audio track to my ts file using ffmpeg, thanks a lot. And I'm guessing I should use tsduck to update the pmt of the new audio track?

lelegard, 2024-08-20:
> I should use tsduck to update the pmt of the new audio track?

Exactly.

SteveMcFarlin, 2024-08-21:
I was just about to ask a similar question, and I was also leaning on needed a muxer. I am guessing from the conversation above this is not possible:

We extract an the audio data from a TS 'chunk' (hls TS output from TS duck). We then process this into several audio files, and wanted to add them to the original TS 'chunk'. While I have found ways to do this with TSDuck it is not ideal, or simply does not work. e.g.

```
tsp -I file --interleave KSTS_video.ts KSTS_audio.ts -O file KSTS_merge.ts
```

The problem we have with FFmpeg is it does not map any SCTE data property or at all. So I guess my question would be is this flow maybe possible.

Extract and process the data. Combine the processed data with the original file into a new file. Use TS duck to extract and insert any non AV data into the new file? If this is not possible or would be super cumbersome, is there any TS muxer other than FFmpeg that you can recommend?

lelegard, 2024-08-22:
> The problem we have with FFmpeg is it does not map any SCTE data property or at all.

@SteveMcFarlin, you are right, ffmpeg is known to destroy everything in the TS which is not video or audio from the processed service. There is a solution for that:

1. Encapsulate all PID's to preserve into a tunnel PID which is signaled as a data PID of the service.
2. Pipe through ffmpeg, doing the audio/video processing. It won't alter the data PID.
3. Pipe through TSDuck again to restore the encapsulated PID's from the tunnel.

In the TSDuck user's guide, there is [an example which encapsulates the EIT PID to preserve it through ffmpeg](https://tsduck.io/download/docs/tsduck.html#_preserving_eits_through_ffmpeg_processing). Based on this example, you can encapsulate any number of PID's into the tunnel. Please carefully read the description of the [encap](https://tsduck.io/download/docs/tsduck.html#encap-ref) and [decap](https://tsduck.io/download/docs/tsduck.html#_decap) plugins.

SteveMcFarlin, 2024-08-22:
Thank you so much.

closed 2025-02-02 by github-actions[bot]