tsduck/tsduck #1497: [Q] best way to minimize buffering between input and output

wjcarpenter, 2024-08-18:
I use TSDuck as a simple proxy for an MPEG-TS stream. It's to overcome some problems that have nothing to do with TSDuck, but TSDuck happens to help with working around the problems. (Not important for this question, but details [here](https://community.getchannels.com/t/androidhdmi-for-channels-ah4c-a-virtual-channel-tuner-using-hdmi-encoder-s-streaming-stick-s/38976/339?u=wjcarpenter).)

One minor annoyance is that when I connect to the TSDuck HTTP port, there are several seconds of buffered video accumulated and read before my intended stream. The input source is sending all the time.

Here is my current command line:
```
tsp --realtime=true --max-flushed-packets=50 --buffer-size-mb=0.0005 -I http --infinite --ignore-errors http://hank.carpenter.org/ts/1_0 -P history -O http --ignore-bad-request --multiple-clients --server 8888
```
Even with that 0.0005mb buffer, I get about 20 seconds of buffered video before the real video starts. Here's an example scenario to help with understanding what I'm describing....

- Imagine the stream is displaying a clock.
- At 9:00, connect to the proxy and read the stream, stopping at 9:05.
- At 11:00, connect to the proxy and read the stream.
- The consumed stream would show the clock as "9:05" for about 20 seconds, and it would then jump to "11:00".

What I am trying to figure out is how to eliminate or minimize those buffered packets. Is there a way to say "when someone connects to the HTTP port, flush the ring buffer so they will only see fresh packets"?

lelegard, 2024-08-19:
Hi @wjcarpenter 

>  Is there a way to say "when someone connects to the HTTP port, flush the ring buffer so they will only see fresh packets"?

There is no such option.

It is a misconception to consider the chain of plugins as a collaborative system. Each plugin is fully independent and ignore what the others are, what they do, and even if they exist or not. This is how any combination of plugins is allowed. This flexibility is a strength in a lab environment to quickly setup any test configuration. However, this is a weakness in a production environment where you have a specialized global behaviour in mind.

This is why TSDuck has always been presented as a lab tool for experimentation, never as a production tool.

By the way, `--buffer-size-mb=0.0005` would mean 500 bytes, only 2 TS packets. This is counter-productive and would prevent most configurations from working: the threading and buffer processing overhead would likely overcome the useful processing time, only one packet would be processed at a time and no parallelism between plugins would be possible. Therefore, a minimum buffer size of 100 TS packets is enforced, meaning approximately 150 kb. The corresponding playout time depends on the bitrate.

wjcarpenter, 2024-08-19:
Thanks for your detailed response. I did wonder why reducing the buffer size to something ridiculously small did not have further latency reductions that earlier reduced buffer sizes produced.

lelegard, 2024-08-20:
The [answer is always in the code](https://github.com/tsduck/tsduck/blob/c9becda662c977a2e58bd50c97ed9811e8427b84/src/libtsduck/plugins/apps/tsTSProcessor.cpp#L100)

wjcarpenter, 2024-08-20:
> The answer is always in the code

Ha, so it is. So much code, so little time.

BTW I was able to reduce the latency from 20 seconds to 6 seconds by reducing max flush packets from 50 to 5. I imagine that means a performance hit, but it's ok for my scenario.

closed 2025-01-19 by github-actions[bot]