tsduck/tsduck #1498: [Q] Cannot compile in VS 2022 17.11.0

enricovittorini, 2024-08-20:
i am trying to compile https://github.com/tsduck/tsduck/blob/master/sample/sample-plugin/tsplugin_sample.cpp 

on Windows 10 with VS 17.11.0 
_MSC_FULL_VER = 194134120

i get LNK1120

Note I had to comment //tsp->info(u"got %d packets", counter); that was giving another error

```
tsversion --version=all
tsversion: TSDuck - The MPEG Transport Stream Toolkit - version 3.37-3670
Built Mar 19 2024 - 10:39:37
Using MSVC 19.39.33522.00, C++ std 2017.03
System: Windows (Windows 6.2 Build 9200), on Intel x86-64, 64-bit, little-endian, page size: 4096 bytes
Acceleration: CRC32: no
Bitrate: 64-bit floating-point value
Dektec: DTAPI: 6.3.0.221, Service: 3.8.6.95
VATek: libvatek version 3.01
Cryptographic library: Microsoft BCrypt
Web library: WinInet
SRT library: libsrt version 1.5.3
RIST library: librist version v0.2.10, API version 4.4.0
```

lelegard, 2024-08-20:
> i get LNK1120

Hi @enricovittorini, what is the error message, precisely?

enricovittorini, 2024-08-20:
@lelegard 
Here it is the error:
```
1>tsplugin_sample.cpp
1>MSVCRTD.lib(exe_main.obj) : error LNK2019: unresolved external symbol main referenced in function "int __cdecl invoke_main(void)" (?invoke_main@@YAHXZ)
1>C:\tsduck\eviduck\x64\Debug\eviduck.exe : fatal error LNK1120: 1 unresolved externals
```

enricovittorini, 2024-08-20:
sorry @lelegard 

Error	LNK1120	1 unresolved externals	eviduck	C:\tsduck\eviduck\x64\Debug\eviduck.exe	1

lelegard, 2024-08-20:
The error about `tsp->info` comes from an inconsistency between the installed version (3.37 final) and the current state of the git repository where the sample code is extracted (3.38-****). You wouldn't have had this error if you grabbed the same sample from the 3.37 source archive.

Let me explain why.

Up to TSDuck 3.37, all reporting methods used a profile using: `(format-string, initializer_list)`. Therefore, the call syntax was something like that (note the `{}` braces):
~~~
info(u"got %d packets", {counter});
info(u"strings %s and %s", {str1, str2});
~~~

In the cycle of 3.38, many things were modernized, to adapt to a more modern C++ style. The previous reporting methods with`(format-string, initializer_list)` are kept for compatibility but marked as "deprecated". New methods were added using variadic templates. They are identical to the previous methods but the `{}` braces are no longer needed in the call. This is a more natural programming style:
~~~
info(u"got %d packets", counter);
info(u"strings %s and %s", str1, str2);
~~~

In the source tree of 3.38, all code and sample applications were updated to use the new style. However, they cannot be compiled with TSDuck 3.37 because the new methods did not exist.

You have two options:
- Keep TSDuck 3.37 but grab the sample sources from 3.37.
- Install the [latest 3.38 prerelease](https://tsduck.io/download/prerelease/)

If you start a new development, I would recommend the second option. The latest prerelease is very close to the final 3.38. I waited for the big history rewrite (see #1490) before releasing the version 3.38. It is expected soon.

enricovittorini, 2024-08-20:
@lelegard thanks.

I now copied the code from 3.37 (https://github.com/tsduck/tsduck/blob/v3.37-3670/sample/sample-plugin/tsplugin_sample.cpp)

Now i get:

```
1>MSVCRTD.lib(exe_main.obj) : error LNK2019: unresolved external symbol main referenced in function "int __cdecl invoke_main(void)" (?invoke_main@@YAHXZ)
1>C:\tsduck\eviduck\x64\Debug\eviduck.exe : fatal error LNK1120: 1 unresolved externals
```

lelegard, 2024-08-20:
I don't understand this one. If I install the 3.37 pack and compile its samples using VS 17.11.0, it builds correctly.

However, you apparently don't compile the sample plugin because this sample project builds a DLL and your project attempts to build an EXE. What are you trying to do? Build a plugin (hence a DLL) or and application (EXE) or a project containing both?

If you build a plugin only, make sure that your project is configured to build a DLL. If you try to build an EXE from the source code of a DLL, it won't link since there is no symbol `main` in a DLL (which is precisely your error: "unresolved external symbol main").

enricovittorini, 2024-08-20:
thanks @lelegard 
i build the dll now and moved it in the C:\Program Files (x86)\TSDuck\bin.

if i run the test tsp -I null -P until --packet 200 -P sample --count -O drop i do not have any output

enricovittorini, 2024-08-20:
my mistake sorry

closed 2024-08-20 by enricovittorini