tsduck/tsduck #1500: [Q] Fuzz Plugin:

enricovittorini, 2024-08-21:
tsduck: 3.38-3822
Windows 10

starting from the tspluin_sample i added a line to check for "sync byte error".

```
ts::ProcessorPlugin::Status ts::SamplePlugin::processPacket(TSPacket& pkt, TSPacketMetadata& pktData)
{
    if (doCount) {
        counter++;
    }

    //Sync_byte_error
    if (!pkt.hasValidSync()) {
        info(u"ETR-P1: Not valid Sync but on PID %d", pid);
    }

 return TSP_OK;

}
```

When I run the command without the fuzzy plugin i have no issue reported on the output.
`tsp -v -I file test.ts  -P sample -O drop`

When I run the following command, sync error are reported on pids other than the 255 specified
`tsp -v -I file test.ts -P fuzz -c 1/20  --sync-byte --pid 255 -P sample -O drop`

lelegard, 2024-08-22:
> sync error are reported on pids other than the 255 specified

`-P fuzz --pid 255` means that only packets with _input_ PID 255 will be altered. However, the corruption may happen anywhere in the packet, including in the 2-byte field containing the PID value. You corrupt an average of 1 byte out of 20. So, for a sufficiently large file, the probability to corrupt the sync byte and one of the two PID bytes in the same packet is not negligible.

enricovittorini, 2024-08-22:
@lelegard thanks. I thought that with the Sync Byte option only the sync byte was corrupted. Misinterpreted the manual.

Thanks

closed 2024-08-22 by enricovittorini