tsduck/tsduck #1503: [Q] ETR 101290 - P1 - 1.3.a

enricovittorini, 2024-08-25:
Was trying to practice the dev of a plugin  with the implementation of ETR101290 
I started from form the tsplugin_sample.cpp.

I would like to have a clue on how to implement the following two points:

1. Sections with table_id 0x00 do not occur at least every 0,5 s on PID 0x0000.
2. Section with table_id other than 0x00 found on PID 0x0000.

```

`ts::ProcessorPlugin::Status ts::SamplePlugin::processPacket(TSPacket& pkt, TSPacketMetadata& pktData)
{
   
    //Get PID
    uint16_t pid{};
    pid = pkt.getPID();

    //ETR-P1-1.2: Sync_byte_error
    if (!pkt.hasValidSync()) {
        info(u"ETR-P1: Not valid Sync but on PID %d", pid);
    }

    //PAT PID
    if (pid == MPEG_PAT_PID) {

        //ETR-P1-1.3a: TSC control on PAT
        if (pkt.isScrambled()) {
            info(u"ETR-P1: PAT pid tsc is not 00");
        }

    }

    return TSP_OK;
}
```

thanks!

lelegard, 2024-08-26:
Hi @enricovittorini 

To grab sections, 
- implement `SectionHandlerInterface` in your plugin
- use a `SectionDemux`
- set `AllPIDs` as PID filter
- in your `processPacket`, pass all packets to the demux
- your `handleSection` is invoked on every new section.

There are several existing plugins with a similar pattern.

To manage the time between sections:
- Create a map of contexts per PID/TID pair
- Keep the `firstTSPacketIndex()` of the last `Section` in the TID/PID (when `sectionNumber()` is 0).
- Compute the distance in packets between two tables.
- Use the `tsp->bitrate()` to convert into a duration using `PacketInterval()`

Also note that the predefined PID's and TID's values are already defined in `tsTS.h` and `tsPSI.h`.

enricovittorini, 2024-08-26:
@lelegard thanks. could you guide me a little bit on the creation of the map of Context per PID/TID value?

So far the code is below:

// Sample TSDuck plugin (packet processing).
// Simply count packets if --count is specified.

#include "tsduck.h"

//----------------------------------------------------------------------------
// Plugin definition
//----------------------------------------------------------------------------

namespace ts {
     class SamplePlugin:
        public ProcessorPlugin,
        //private TableHandlerInterface,
        private SectionHandlerInterface
        //private TSAnalyzer
        
     
                
    {
        TS_NOBUILD_NOCOPY(SamplePlugin);
        //TS_PLUGIN_CONSTRUCTORS(SamplePlugin);
    public:
        // Implementation of plugin API
        SamplePlugin(TSP*);
        virtual bool getOptions() override;
        virtual bool start() override;
        virtual bool stop() override;
        virtual BitRate getBitrate() override;
        virtual Status processPacket(TSPacket&, TSPacketMetadata&) override;

    private:
        // Command line options, stay unchanged after getOptions():
        bool doCount;  // Option --count

        uint64_t _ts_pkt_cnt = 0;
      

        // Processing data:
        PacketCounter counter;  // Actual packet counter.
        bool pat_tid{ false };

        SectionDemux _demux{ duck, nullptr, this };

        // Implementation of SectionHandlerInterface
        virtual void handleSection(SectionDemux&, const Section&) override;       
    };

}

TS_REGISTER_PROCESSOR_PLUGIN(u"sample", ts::SamplePlugin);

//----------------------------------------------------------------------------
// Constructor
//----------------------------------------------------------------------------

ts::SamplePlugin::SamplePlugin(TSP* tsp_) :
    ProcessorPlugin(tsp_, u"Sample packet processor", u"[options]"),
    
    doCount(false),
    counter(0)

{
    option(u"count", 'c');
    help(u"count", u"Count packets");

}

//----------------------------------------------------------------------------
// Get command line options
//----------------------------------------------------------------------------

bool ts::SamplePlugin::getOptions()
{
    verbose(u"sample plugin: get options");

    doCount = present(u"count");
    return true;
}

//----------------------------------------------------------------------------
// Start method
//----------------------------------------------------------------------------

bool ts::SamplePlugin::start()
{
    verbose(u"sample plugin: start");

    counter = 0;
    _demux.reset();
    _demux.setPIDFilter(AllPIDs);

    return true;
   
}

//----------------------------------------------------------------------------
// Stop method
//----------------------------------------------------------------------------

bool ts::SamplePlugin::stop()
{
    verbose(u"sample plugin: stop");

    // Close resources, display final report, etc.
    if (doCount) {
        info(u"got %d packets", counter);
    }

    return true;
}

//----------------------------------------------------------------------------
// New bitrate computation method, return zero if unknown
//----------------------------------------------------------------------------

ts::BitRate ts::SamplePlugin::getBitrate()
{
    // If the plugin recomputes the bitrate, implement here.
    // Otherwise, return zero, or simply don't override getBitrate().
    return 0;
}

//----------------------------------------------------------------------------
// Implementation of SectionHandlerInterface
//----------------------------------------------------------------------------
void ts::SamplePlugin::handleSection(SectionDemux&, const Section& section) {
    info(u"Section found %d", section.isValid() );
}

//----------------------------------------------------------------------------
// Packet processing method
//----------------------------------------------------------------------------

ts::ProcessorPlugin::Status ts::SamplePlugin::processPacket(TSPacket& pkt, TSPacketMetadata& pktData)
{

    //Get PID
    uint16_t pid{};
    pid = pkt.getPID();
   
    //ETR-P1-1.2: Sync_byte_error
    if (!pkt.hasValidSync()) {
        info(u"ETR-P1: Not valid Sync but on PID %d", pid);
    }

    _demux.feedPacket(pkt);

    //PAT PID
    if (pid == PID_PAT) {
        

        //ETR-P1-1.3a: TSC control on PAT
        if (pkt.isScrambled()) {
            info(u"ETR-P1: PAT pid tsc is not 0: %d", pkt.getScrambling());
        }

       
    }

    return TSP_OK;
}

lelegard, 2024-08-26:
I can give you advises, hints or directions, but I cannot debug your code.

Concerning a sample map of PID contexts, you can have a look at plugin `pcrverify`. It is one of the many plugins which keep a map of contexts per PID (or TID, or PID/TID pair, or whatever). The contexts are built and updated as packets arrive (or sections, or whatever).

That's the whole idea of a plugin: you accumulate information in the private fields of the plugin instance and you manipulate that information on each packet. Don't forget to reinitialize the accumulated information on `start()`.

enricovittorini, 2024-08-26:
thanks @lelegard 

I think I managed to have something:

```

// Sample TSDuck plugin (packet processing).
// Simply count packets if --count is specified.

#include "tsduck.h"

//----------------------------------------------------------------------------
// Plugin definition
//----------------------------------------------------------------------------

namespace ts {
     class SamplePlugin:
        public ProcessorPlugin,
        private SectionHandlerInterface
                
    {
        TS_NOBUILD_NOCOPY(SamplePlugin);
        //TS_PLUGIN_CONSTRUCTORS(SamplePlugin);
    public:
        // Implementation of plugin API
        SamplePlugin(TSP*);
        virtual bool getOptions() override;
        virtual bool start() override;
        virtual bool stop() override;
        virtual BitRate getBitrate() override;
        virtual Status processPacket(TSPacket&, TSPacketMetadata&) override;

    private:
        // Command line options, stay unchanged after getOptions():
        bool doCount;  // Option --count

        uint64_t _ts_pkt_cnt = 0;
      

        // Processing data:
        PacketCounter counter;  // Actual packet counter.
        bool pat_tid{ false };

        // Description of one PID
        struct PIDContext
        {
            PIDContext() = default;
            uint64_t     table_count{ 0 };
            PacketCounter     first_pkt{ 0 };
        };

        std::map<PID, PIDContext> _stats{};              // Per-PID statistics

    

        

        SectionDemux _demux{ duck, nullptr, this };

        // Implementation of SectionHandlerInterface
        virtual void handleSection(SectionDemux&, const Section&) override;       
    };

}

TS_REGISTER_PROCESSOR_PLUGIN(u"sample", ts::SamplePlugin);

//----------------------------------------------------------------------------
// Constructor
//----------------------------------------------------------------------------

ts::SamplePlugin::SamplePlugin(TSP* tsp_) :
    ProcessorPlugin(tsp_, u"Sample packet processor", u"[options]"),
    
    doCount(false),
    counter(0)

{
    option(u"count", 'c');
    help(u"count", u"Count packets");

}

//----------------------------------------------------------------------------
// Get command line options
//----------------------------------------------------------------------------

bool ts::SamplePlugin::getOptions()
{
    verbose(u"sample plugin: get options");

    doCount = present(u"count");
    return true;
}

//----------------------------------------------------------------------------
// Start method
//----------------------------------------------------------------------------

bool ts::SamplePlugin::start()
{
    verbose(u"sample plugin: start");

    counter = 0;
    _demux.reset();
    _demux.setPIDFilter(AllPIDs);
    _stats.clear();

    return true;
   
}

//----------------------------------------------------------------------------
// Stop method
//----------------------------------------------------------------------------

bool ts::SamplePlugin::stop()
{
    verbose(u"sample plugin: stop");

    // Close resources, display final report, etc.
    if (doCount) {
        info(u"got %d packets", counter);
    }

    return true;
}

//----------------------------------------------------------------------------
// New bitrate computation method, return zero if unknown
//----------------------------------------------------------------------------

ts::BitRate ts::SamplePlugin::getBitrate()
{
    // If the plugin recomputes the bitrate, implement here.
    // Otherwise, return zero, or simply don't override getBitrate().
    return 0;
}

//----------------------------------------------------------------------------
// Implementation of SectionHandlerInterface
//----------------------------------------------------------------------------
void ts::SamplePlugin::handleSection(SectionDemux&, const Section& section) {
    

    PID  pid = section.sourcePID();
    PIDContext& pc(_stats[pid]);
    

    // Section# 0 is used to track tables
    if (section.tableId() == TID_PAT && section.sectionNumber() == 0) {

            PID pid = section.sourcePID();
            TID tid = section.tableId();
            PacketCounter pk = section.firstTSPacketIndex();
            
            // if not the first time I see a table
            if (_stats[pid].first_pkt != 0) {
                  
                  tsp->bitrate();

                  PacketCounter pkt_int = section.firstTSPacketIndex() - _stats[pid].first_pkt;

                  std::chrono::milliseconds interval = PacketInterval(tsp->bitrate(), pkt_int);
                  info(u"Interval: %f", interval);

                  _stats[pid].first_pkt = section.firstTSPacketIndex();
            }
            else {
                _stats[pid].first_pkt = section.firstTSPacketIndex();
            }
        
    }
}

//----------------------------------------------------------------------------
// Packet processing method
//----------------------------------------------------------------------------

ts::ProcessorPlugin::Status ts::SamplePlugin::processPacket(TSPacket& pkt, TSPacketMetadata& pktData)
{

    //Get PID
    uint16_t pid{};
    pid = pkt.getPID();
   
    //ETR-P1-1.2: Sync_byte_error
    if (!pkt.hasValidSync()) {
        info(u"ETR-P1: Not valid Sync but on PID %d", pid);
    }

    _demux.feedPacket(pkt);

    //PAT PID
    if (pid == PID_PAT) {
        

        //ETR-P1-1.3a: TSC control on PAT
        if (pkt.isScrambled()) {
            info(u"ETR-P1: PAT pid tsc is not 0: %d", pkt.getScrambling());
        }

       
    }

    return TSP_OK;
}

```

enricovittorini, 2024-08-29:
@lelegard  what thinking that it is not enough to keep track of the forst packet with section =0.

the ETR 1.3.a is defined as: Sec_tions with table_id 0x00 do not occur at least every 0,5 s on PID 0x0000_

_PAT_error_2
The reworded description of the error in PAT_error_2 refers to the possibility that the Program Association Table may consist of several (consecutive) sections with the same table_id 0x00_

I was thinking that for Each PID I should keep track of each section_number and check that the packet interval for each section  is within the standard range. 

The structure I had in mind is:

```
        //Section Context: keep track of the last packet of each section in a table
        struct SectionContext
        {
            SectionContext() = default;
            PacketCounter     sec_pkt{ 0 };
            std::chrono::milliseconds interval{ 0 };
        };

         // Description of one PID
        struct PIDContext
        {
            PIDContext() = default;
            uint64_t     table_count{ 0 };
            PacketCounter     first_pkt{ 0 };
            std::chrono::milliseconds interval{ 0 };
            uint8_t num_sections{ 0 };
            uint8_t version{ 0 };
            //SectionContext _sec_stats{};
            std::map<TID, std::map<uint8_t, SectionContext> > _sec_stats{}; // track all sections within this PID for each TableID
        };

        std::map<PID, PIDContext> _stats{};
```

what do you think?

enricovittorini, 2024-08-31:
hello @lelegard 
is  the handleTable called by the demux for a PMT only if that PMT is referenced in the PAT?

I would like to understand what would happen in a scenario where a PMT pid is present in the TS but not referenced in the PAT.

thanks!

lelegard, 2024-08-31:
> is the handleTable called by the demux for a PMT only if that PMT is referenced in the PAT?

The class `SectionDemux` is just a... section demux, nothing else. Its role is to demux sections on a set of PID's. It does not interpret the sections and does not make any decision based on their content.

So, if you initialize `demux.setPIDFilter(ts::AllPIDs)`, it will demux all sections and tables from all PID's in the TS. Event if it seems overkill, there is no issue about that. When a PID does not contain sections, it contains PES packets. By design, a PES payload cannot be interpreted as sections (the PES start code 00 00 01 is not a valid TS payload for sections). Therefore, you will get all sections and tables from all PID's, without interpretation.

Your natural interpretation as PMT being located from the PAT is a higher level of application. Typically, this is done in class `SignalizationDemux` for instance. This class extracts the logical signalization of the TS. It works at a higher level. It is not a `SectionDemux`, it _contains_ a `SectionDemux`. The internal demux is initially set with PID 0 for the PAT, and optionally all the predefined PID's for MPEG, DVB, ATSC, ISDB. Then, as tables are collected, more PID's are demuxed. Typically, after a PAT is received, all specified PMT PID's are added to the `SectionDemux` in order to collect those PMT's. In such a context, a PMT which is not signaled in the PAT won't be found, because we interpret the logical structure of the signalization.

But `SectionDemux` is just a low-level demux which works on PID's only.

closed 2025-02-09 by github-actions[bot]