tsduck/tsduck #1505: [Q] automatic DVB-T2 detection?

cristianpuddu, 2024-08-28:
Hi, tonight RAI in Italy finally converted one frequency from DVB-T to DVB-T2. 
With Tsduck + Raspberry Pi + Pi Tuner TV HAT, when I perform a scan with Tsscan or TSP -I dvb ..., I am forced to use --delivery-system DVB-T2.
 Only this way is it detected; otherwise, TSDuck doesn't find anything.
 Is this normal? Isn't it possible to automatically detect the modulation type?
 Thanks

lelegard, 2024-08-28:
Hi @cristianpuddu 

We cannot really "detect" the modulation type. When the tuner tries a frequency, the application had set specific modulation parameters. For DVB-T (and probably DVB-T2), it can auto-detect some parameters such as FEC or guard interval. But this parameter detection remains inside a given modulation type.

To "detect" DVB-T vs. T2, we would need to try each frequency twice: once using DVB-T, once using DVB-T2. That is a possible modification but it would double the scan time.

When you use `tsscan --delivery-system DVB-T2`, I assume it only detects DVB-T2 streams, not DVB-T streams, right?

cristianpuddu, 2024-08-28:
yes @lelegard , unfortunatly tsscan --delivery-system DVB-T2 detects only DVB-T2 streams.
 possible solution could be an option that makes Tsscan directly perform two scans....

lelegard, 2024-09-01:
Hi @cristianpuddu 

The option `--delivery-system` can now be specified more than once in `tsscan`. In that case, each frequency is scanned several times, using the specified delivery systems, in the specified order, until a TS is found. As expected, the scan time doubles on all unused frequencies since we timeout twice (for two delivery systems).

See an example below of a scan of the French terrestrial network from the Paris Eiffel Tower transmitter. We see that UHF channels 22, 25, 28, 30, 32, 35, 42 have a DVB-T signal and UHF channel 24 has a DVB-T2 signal.

~~~
$ tsscan --verbose --uhf-band --service-list --show-modulation --delivery-system dvb-t --delivery-system dvb-t2
scanning channel 21, 474,000,000 Hz (DVB-T)
scanning channel 21, 474,000,000 Hz (DVB-T2)
scanning channel 22, 482,000,000 Hz (DVB-T)
* UHF channel 22 (482 MHz), locked: no, strength: -69 dB, SNR: 23.25 dB, BER: 65%
  Transport stream id: 3, 0x0003
  Delivery system: DVB-T
  Carrier frequency: 482,000,000 Hz
  Modulation: 64-QAM
  Guard interval: 1/32
  Bandwidth: 8,000,000 Hz
  Transmission mode: 8K
  Hierarchy: none

  LCN Name           Provider ServId TSId   ONetId Type PMTPID
  --- -------------- -------- ------ ------ ------ ---- ------
    4 CANAL+         CNH      0x0301 0x0003 0x20FA 0x19 0x0384 
   26 LCI            CNH      0x0308 0x0003 0x20FA 0x16 0x01F4 
   41 PARIS PREMIERE CNH      0x0309 0x0003 0x20FA 0x16 0x0258 
   42 CANAL+ SPORT   CNH      0x0303 0x0003 0x20FA 0x19 0x012C 
   43 CANAL+ CINEMA  CNH      0x0302 0x0003 0x20FA 0x19 0x00C8 
   45 PLANETE+       CNH      0x0304 0x0003 0x20FA 0x19 0x0190 
      CANAL+         CNH      0x030A 0x0003 0x20FA 0x19 0x0064 
                     CNH      0x03F2 0x0003 0x20FA 0x0C 0x0578 
                     CNH      0x03F3 0x0003 0x20FA 0x0C 0x05DC 
                     CNH      0x03F4 0x0003 0x20FA 0x0C 0x0640 
      DATASYSTEM R7  CNH      0x03F5 0x0003 0x20FA 0x0C 0x06A4 
                     CNH      0x03F6 0x0003 0x20FA 0x0C 0x0708 

scanning channel 23, 490,000,000 Hz (DVB-T)
scanning channel 23, 490,000,000 Hz (DVB-T2)
scanning channel 24, 498,000,000 Hz (DVB-T)
scanning channel 24, 498,000,000 Hz (DVB-T2)
* UHF channel 24 (498 MHz), locked: no, strength: -53 dB, SNR: 25.5 dB, BER: 67%
  Transport stream id: 9, 0x0009
  Delivery system: DVB-T2
  Carrier frequency: 498,000,000 Hz
  Spectral inversion: off
  Modulation: 64-QAM
  Guard interval: 1/32
  Bandwidth: 8,000,000 Hz
  Transmission mode: 8K
  Hierarchy: none

  LCN Name         Provider ServId TSId   ONetId Type PMTPID
  --- ------------ -------- ------ ------ ------ ---- ------
   52 France 2 UHD GR9A     0x0901 0x0009 0x20FA 0x20 0x0BCC 
   53 France 3 UHD GR9A     0x0902 0x0009 0x20FA 0x20 0x0BEA 

scanning channel 25, 506,000,000 Hz (DVB-T)
* UHF channel 25 (506 MHz), locked: no, strength: -53 dB, SNR: 27.5 dB, BER: 67%
  Transport stream id: 2, 0x0002
  Delivery system: DVB-T
  Carrier frequency: 506,000,000 Hz
  Spectral inversion: off
  Modulation: 64-QAM
  Guard interval: 1/32
  Bandwidth: 8,000,000 Hz
  Transmission mode: 8K
  Hierarchy: none

  LCN Name   Provider ServId TSId   ONetId Type PMTPID
  --- ------ -------- ------ ------ ------ ---- ------
[... 159 lines cut ...]
scanning channel 58, 770,000,000 Hz (DVB-T)
scanning channel 58, 770,000,000 Hz (DVB-T2)
scanning channel 59, 778,000,000 Hz (DVB-T)
scanning channel 59, 778,000,000 Hz (DVB-T2)
scanning channel 60, 786,000,000 Hz (DVB-T)
scanning channel 60, 786,000,000 Hz (DVB-T2)
scanning channel 61, 794,000,000 Hz (DVB-T)
scanning channel 61, 794,000,000 Hz (DVB-T2)
scanning channel 62, 802,000,000 Hz (DVB-T)
scanning channel 62, 802,000,000 Hz (DVB-T2)
scanning channel 63, 810,000,000 Hz (DVB-T)
scanning channel 63, 810,000,000 Hz (DVB-T2)
scanning channel 64, 818,000,000 Hz (DVB-T)
scanning channel 64, 818,000,000 Hz (DVB-T2)
scanning channel 65, 826,000,000 Hz (DVB-T)
scanning channel 65, 826,000,000 Hz (DVB-T2)
scanning channel 66, 834,000,000 Hz (DVB-T)
scanning channel 66, 834,000,000 Hz (DVB-T2)
scanning channel 67, 842,000,000 Hz (DVB-T)
scanning channel 67, 842,000,000 Hz (DVB-T2)
scanning channel 68, 850,000,000 Hz (DVB-T)
scanning channel 68, 850,000,000 Hz (DVB-T2)
scanning channel 69, 858,000,000 Hz (DVB-T)
scanning channel 69, 858,000,000 Hz (DVB-T2)
~~~

cristianpuddu, 2024-09-01:
txn @lelegard, I will try the new version these days. Is it already available in the repository?

lelegard, 2024-09-01:
Yes, this is the latest commit in the repo.

jof-cw, 2024-09-01:
Good night all. Excuse me @, would you share a recent recording of channel 24 in DVB-T2?.

Thanks in advance.
Jesus

cristianpuddu, 2024-09-03:
Hi  @lelegard, unfortunately, I'm struggling to compile the new version on the Raspberry Pi, but I'm unable to do so and I'm encountering numerous errors. I might open a few more issue reports if necessary. I had old version 3.28.2509

lelegard, 2024-09-03:
Could you specify which type of RPi, which distro, which version, which GCC version.

Please also post the commands and error messages.

Version 3.28 is very old and TSDuck went through a lot of code refactoring in the meantime to adapt to more modern C++. See also the [building guide](https://tsduck.io/download/docs/tsduck-dev.html#unixbuild)

cristianpuddu, 2024-09-05:
Hi @lelegard,
I’ve finally solved the issue! There were several problems.
I was using an old version of Raspberry OS (Buster). I tried to update GCC and use LLVM/Clang, but I kept getting various errors. I know it’s not ideal to use outdated versions, but I have a few Raspberry Pi devices with TSDuck installed, which I manage remotely, and updating them is not easy.

I was hoping the issue would be solved with the latest version of Raspberry Pi OS (Bookworm). However, the problem persisted: while compiling, the system would freeze without any error message, and the green read/write LED stayed on. The Raspberry Pi was unresponsive, as if it had locked up.
I replaced the SD card thinking it might be the cause, but the problem happened again.

Finally, I restarted the compilation while monitoring the system with htop. Thanks to this, I discovered that the swap file was too small, and the Raspberry Pi froze when it ran out of space during the compilation! After increasing the swap size, the issue was resolved! Now I just need to find a way to upgrade the Os in other devices remotely.
Hardware: Raspberry 3A+

closed 2025-02-09 by github-actions[bot]