tsduck/tsduck #1508: [Q] Problem when switching from vatek to dektec

Ravnholt7507, 2024-08-30:
Hi

I'm encountering an issue with the TSP command.
I have 2 different modulators, a vatek and a dektec. Individually they work fine when running the tsp command, but once i have run the tsp command for vatek one time, i can no longer run it with dektec, untill i have unplugged my vatek device. 

It does not give me any errors when i run it, but no output stream is produced to my tv.

it seems dmesg is pointing at the issue, but im not quite sure what it is.

`[20767.685946] usb 3-7: Product: VAT-Device-103
[20767.685947] usb 3-7: Manufacturer: VATek
[20767.685948] usb 3-7: SerialNumber: B103140916AB
[20769.309903] usb 3-1: new high-speed USB device number 8 using xhci_hcd
[20769.436048] usb 3-1: New USB device found, idVendor=1297, idProduct=020f, bcdDevice= 0.00
[20769.436058] usb 3-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[20769.465277] usb 3-1: USB disconnect, device number 8
[20771.203759] usb 3-1: new high-speed USB device number 9 using xhci_hcd
[20771.330682] usb 3-1: New USB device found, idVendor=1297, idProduct=020f, bcdDevice= 0.00
[20771.330696] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[20771.330700] usb 3-1: Product: DTU-215 USB-2 Multi-Standard Modulator
[20771.330703] usb 3-1: Manufacturer: DekTec
[20771.989337] Dtu: < DtIoConfigReadFromIniFile > Error reading config file /etc/DekTec/Dtu/215/Port1.ini. Error: fffffffeh
[20771.989346] Dtu: < DtIoConfigReadFromIniFile > Error reading config file /etc/DekTec/Dtu/4215005198/Port1.ini. Error: fffffffeh
[20806.300434] Dtu: < DtTableGet > Max. number of entries to small. Needed:2, Space for:1
[20806.300454] Dtu: < DtTableGet > Max. number of entries to small. Needed:96, Space for:2
`

Has anyone encountered this issue before?

lelegard, 2024-08-30:
Hi @Ravnholt7507 

Could you remove the two devices, then plug the Vatek modulator, wait a few second, unplug it, and report the corresponding log from `dmesg`? Then the same thing with the Dektec device. From your log, it is difficult to understand which line comes from which device.

Ravnholt7507, 2024-08-30:
hi @lelegard 

The error messages about the max number of entries only appear after i reproduce the issue. If i simply unplug and plugin the devices there are no issues.

The issue is that when i first run a command on my vatek, like "tsp -I file generic-HbbTV-Teststream.ts -i -O vatek --device 0 --frequency 474000000 --modulation DVB-T --bandwidth 8",  it works fine, and i can keep running this command without any trouble. But when i then run exactly the same command, but with dektec instead, it still seems to run fine, but no stream is displayed to the screen.

dmesg log for vatek after i unplugged it and plugged it in looks like:

[24682.404338] usb 3-7: USB disconnect, device number 7
[24682.934699] usb 3-1: USB disconnect, device number 9
[24688.468571] usb 3-7: new high-speed USB device number 10 using xhci_hcd
[24688.597993] usb 3-7: New USB device found, idVendor=2c42, idProduct=1031, bcdDevice= 1.00
[24688.598000] usb 3-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[24688.598001] usb 3-7: Product: VAT-Device-103
[24688.598002] usb 3-7: Manufacturer: VATek
[24688.598003] usb 3-7: SerialNumber: B103140916AB

and for dectek it looks like:

[24733.128964] usb 3-1: new high-speed USB device number 11 using xhci_hcd
[24733.259357] usb 3-1: New USB device found, idVendor=1297, idProduct=020f, bcdDevice= 0.00
[24733.259363] usb 3-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[24733.296439] usb 3-1: USB disconnect, device number 11
[24735.054027] usb 3-1: new high-speed USB device number 12 using xhci_hcd
[24735.182086] usb 3-1: New USB device found, idVendor=1297, idProduct=020f, bcdDevice= 0.00
[24735.182110] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[24735.182117] usb 3-1: Product: DTU-215 USB-2 Multi-Standard Modulator
[24735.182123] usb 3-1: Manufacturer: DekTec
[24735.878956] Dtu: < DtIoConfigReadFromIniFile > Error reading config file /etc/DekTec/Dtu/215/Port1.ini. Error: fffffffeh
[24735.878974] Dtu: < DtIoConfigReadFromIniFile > Error reading config file /etc/DekTec/Dtu/4215005198/Port1.ini. Error: fffffffeh

lelegard, 2024-08-30:
> But when i then run exactly the same command, but with dektec instead, it still seems to run fine, but no stream is displayed to the screen.

In your first post, you mentioned it worked "individually".

Could you please clarify and list all configurations which work and which don't, with the precise commands?

lelegard, 2024-08-30:
We need the help of someone from Dektec.

@EricDeJong, could you please help troubleshooting what can go wrong with a DTU-215 when another USB device is connected on another USB port?

The other device is a Vatek modulator. FYI, this device has no device driver. It is used from userspace using libusb. So, there is no associated driver in kernel space.

Ravnholt7507, 2024-11-11:
@lelegard  This issue has been fixed by #1522 - Thanks a lot

closed 2024-11-11 by Ravnholt7507