tsduck/tsduck #1509: [FEAT] Continuity Counters log options

enricovittorini, 2024-09-01:
It would be great to add the following two options for the Continuity Counter logs:
Continuity Counters logs show PID value in hex format. Add an option to log PIDs in decimal format
Continuity Counters logs show by default packets index. Add an option to remove packets index

lelegard, 2024-09-02:
Hi @enricovittorini 

What is the purpose of the request? If you want to process the output with some automatic system, the best way is to use `--json` which produces logs like this one:

~~~
* continuity: { "index": 962728, "packets": 2, "pid": 230, "type": "missing" }
~~~

enricovittorini, 2024-09-02:
@lelegard is to have something easier to read on the console output. I am not a developer and basically no knowledge of C++ but I just submitted a pull request for this. Tried to mimic what you do in your code.... it seems ok 

```
* continuity: packet index: 756,805, PID: 2552, missing 2 packets
* continuity: packet index: 757,412, PID: 2555, missing 3 packets
* continuity: packet index: 757,416, PID: 2554, missing 3 packets
* continuity: packet index: 759,006, PID: 2553, missing 3 packets
* continuity: packet index: 760,299, PID: 2552, missing 10 packets
* continuity: packet index: 760,772, PID: 2556, missing 1 packets
* continuity: packet index: 762,517, PID: 2553, missing 6 packet
```

lelegard, 2024-09-02:
If this is only for the sake of clarity, no need for a new option. Let's switch to the hex/dec format which is used in most other tools: `0xHHHH (DDD)`

enricovittorini, 2024-09-02:
I guess this might break someone code that is parsing the PID number form PID: to the next ","

I was thinking about using an option to keep in the log only what is needed and eventually add another option: "no_index" to remove the "packet index: 967,812". But if you prefer your approach... it is fine too...

lelegard, 2024-09-02:
> I guess this might break someone code that is parsing the PID number form PID: to the next ","

Applications should not use the "human readable" output, they should parse the `--json` output. That's why JSON exists, a parseable structured format.

closed 2025-02-09 by github-actions[bot]