tsduck/tsduck #1514: [Q] tsRec application using tsduck does not resume recording when Source multicast UDP stream stops and restarts again

anup-chhetri, 2024-09-07:
Hi, 

I have installed tsRec + tsduck applications on Rocky Linux 9 and have been using it to record UDP multicast streams. Manual start and stop works normally as expected. However, when the stream is disabled/stops from the source = recording stops and if the source stream resumes again after a while, it does not resume recording again, remains stopped.
Can some one please advise? Appreciate any suggestions. Below is the 'recording start' configuration on the tsRec.py file.

#############Used to start the recordings##########

def start_recording(data, is_loading=False):
    logging.info(f"start_recording called. is_loading={is_loading}")  # Debug log
    multicast_address = data['multicast_address']
    port = data.get('port', 2001)
    durationsplit = data.get('durationsplit', 300)
    fileprefix = data.get('fileprefix', 'ip_')[:6]
    base_filename = f"{fileprefix}_{multicast_address}_{port}"
    logging.info(f"Prepared to start recording with base_filename={base_filename}")  # Debug log

    with ongoing_recordings_lock:
        # Check if a recording is already ongoing for this stream
        if (multicast_address, port) in ongoing_recordings:
            logging.info(f"Stream {multicast_address}:{port} is already being recorded.")
            return "failure"

        # Check if the base_filename is already in use
        if any(meta['base_filename'] == base_filename for meta in ongoing_recordings.values()):
            logging.info(f"Base filename {base_filename} is already in use.")
            return "failure"

        # Get the current time in GMT if not loading from saved state
        start_time_gmt = data.get('start_time', datetime.utcnow().strftime('%Y-%m-%d %H:%M:%S') if not is_loading else None)

        logging.debug("Initializing TSDuck report and TSProcessor for recording")
        rep = Logger()
        tsp = tsduck.TSProcessor(rep)

        tsp.input = ['ip', f"{multicast_address}:{port}", '--receive-timeout', '6000']
        tsp.output = ['file', '--max-duration', f'{durationsplit}', f'{OUTPUT_DIR}/{base_filename}_.ts']

        retry_attempts = 0
        max_retries = 10  # Maximum number of retries before giving up
        retry_interval = 10  # Seconds to wait between retries

        while retry_attempts < max_retries:
            try:
                logging.info(tsp.input)
                logging.info(tsp.output)
                tsp.start()
                logging.info("TSP started successfully.")  # Debug log
                ongoing_recordings[(multicast_address, port)] = {
                    'tsp': tsp,
                    'rep': rep,
                    'start_time': start_time_gmt,
                    'base_filename': base_filename,
                    'fileprefix': fileprefix
                }
                if not is_loading:
                    save_state()
                return "success"
            except tsduck.TSPStartError:
                logging.warning(f"Stream recording could not be started. Retry attempt {retry_attempts + 1}/{max_retries}")
                retry_attempts += 1
                time.sleep(retry_interval)

        logging.error("Failed to start the recording after maximum retry attempts.")
        return "failure"

Thank you
Andy

lelegard, 2024-09-08:
Hi @anup-chhetri 

I have no idea on what tsRec does.

To debug an issue with two products, first try one by one only.

Let's have a rational troubleshooting approach:

- Could you reproduce a problem with TSDuck only, so that we can analyse it without falling into the usual trap of "this is the other one's fault"?
- If you can't, please explain in details what tsRec does and how it is supposed to interact with TSDuck.
- If you can, try to reproduce the issue with a `tsp` command.
  - If you can, we can concentrate or reception issues.
  - If you can't and have two identical configurations, one which works with a `tsp` command and one which does not from Python, we can concentrate on Python issues.

anup-chhetri, 2024-09-29:
Hi Lelegard,

1. I tried using TsDuck only to record as below and stopped the source stream. Recording stopped and resumed again when i enabled the source stream again. So looks like it not an issue with TsDuck implementation.
tsp --synchronous-log -I ip  239.200.100.9:2001 -O file --max-duration 300 J2K69_.ts

2. However, the issue arises when we do the same thing via https (using nginx). The stream does not restart when source stream is enabled again.

Thanks
Anup Chhetri

lelegard, 2024-09-29:
Hi @anup-chhetri 

As I mentioned in my previous post, you need to have a rational troubleshooting approach if you want to have a chance to analyze your problem. 

Please be precise. Describe exact commands, in chronological order, with an exact description of what happens, on each side. Expressions such as "stop the source stream", "same thing with nginx" are not precise descriptions and do not help.

closed 2025-03-02 by github-actions[bot]