tsduck/tsduck #1515: [BUG] Raspberry Pi TV HAT does not work with DVB-T2 Signals

rainerfritz, 2024-09-07:
Hi!
TSDuck is really great tool and love to use it. Thx for all the work done!
I want to use it to monitor the TS tables and specially there the AIT and hbbTV signalling reading the http link.
In my region there are only DVB-T2 transmitters anymore. I have signals on various channels like UHF 24, 34, and so on.

#### Bug description:
When I do a scan with following commands I can get no output of the scan tool:
tsscan -g --uhf-band
knowing that there is a signal on C 24:
tsscan --hf-band-region europe --use-offsets --delivery-system DVB-T2 --first-channel 22 --last-channel 26 -u
tsscan --hf-band-region europe --use-offsets --delivery-system DVB-T2 -u

I then compiled t2scan: https://github.com/mighty-p/t2scan
With this tool I can get all the expected scan results:

t2scan -t2 

498000 (CH24): (time: 00:06.607) 
   plp id -1:   signal ok:	QAM_AUTO f = 498000 kHz I999B8C999D999T999G999Y0P-1 (0:0:0)
        QAM_AUTO f = 498000 kHz I999B8C999D0T32G16Y0P1 (8232:13101:101) : scanning for services
	service = ORF1 HD (ORF)
	service = ORF2 W HD (ORF)
	service = ORF2 N HD (ORF)
	service = ORF2 B HD (ORF)
	service = ORF III HD (ORF)
	service = ORF Sport+ HD (ORF)
	service = ORF1 (ORF)
	service = ORF2 W (ORF)
	service = Radio Österreich 1 (ORF)
	service = Radio Wien (ORF)
	service = Radio Niederösterreich (ORF)
	service = Radio Burgenland (ORF)
	service = Hitradio Ö3 (ORF)
	service = radio FM4 (ORF)
	service = Ö3-Visual-Radio (ORF)
	service = SRT8506 OTA (ORS)

ORF1 HD;ORF:498000:B8D0G16S1T32Y0P1:T:27500:1010=27:0;1011,1012:1015:69C:14101:8232:101:0
ORF2 W HD;ORF:498000:B8D0G16S1T32Y0P1:T:27500:1000=27:0;1001,1002:1005:69C:14102:8232:101:0
ORF2 N HD;ORF:498000:B8D0G16S1T32Y0P1:T:27500:1000=27:0;1001,1002:1005:69C:14103:8232:101:0
ORF2 B HD;ORF:498000:B8D0G16S1T32Y0P1:T:27500:1000=27:0;1001,1002:1005:69C:14104:8232:101:0
ORF III HD;ORF:498000:B8D0G16S1T32Y0P1:T:27500:1050=27:0;1051,1052:1055:69C:14105:8232:101:0
ORF Sport+ HD;ORF:498000:B8D0G16S1T32Y0P1:T:27500:1060=27:0;1061,1062:1065:69C:14106:8232:101:0
ORF1;ORF:498000:B8D0G16S1T32Y0P1:T:27500:1070=27:0;1071:1015:0:14107:8232:101:0
ORF2 W;ORF:498000:B8D0G16S1T32Y0P1:T:27500:1080=27:0;1081:1005:0:14108:8232:101:0
Radio Österreich 1;ORF:498000:B8D0G16S1T32Y0P1:T:27500:0:0;1091:0:0:14109:8232:101:0
Radio Wien;ORF:498000:B8D0G16S1T32Y0P1:T:27500:0:0;1100:0:0:14110:8232:101:0
Radio Niederösterreich;ORF:498000:B8D0G16S1T32Y0P1:T:27500:0:0;1110:0:0:14111:8232:101:0
Radio Burgenland;ORF:498000:B8D0G16S1T32Y0P1:T:27500:0:0;1120:0:0:14112:8232:101:0
Hitradio Ö3;ORF:498000:B8D0G16S1T32Y0P1:T:27500:0:0;1131:0:0:14113:8232:101:0
radio FM4;ORF:498000:B8D0G16S1T32Y0P1:T:27500:0:0;1141:0:0:14114:8232:101:0
Ö3-Visual-Radio;ORF:498000:B8D0G16S1T32Y0P1:T:27500:1150=27:0;1131:0:0:14115:8232:101:0

#### How to reproduce:
use commands above

#### Expected behavior:
should give scan results of received channels

#### Errors and logs:
uname -a
Linux raspilora 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

~~~
[be sure to surround verbatim log lines with "~~~" lines]
~~~

#### Environment:
 - OS: latest RaspiOS 64bit bookworm
 - OS version: see above
 - TSDuck full version: 3.38-3822
 - Installation type: manually installed from website https://github.com/tsduck/tsduck/releases/download/v3.38-3822/tsduck_3.38-3822.debian12_arm64.deb
 - installed libsrt1.5-openssl and librist4 manually

Any help would be much appreciated!
Can you recommend a USB Dongle which defintely works with DVB-T2 on the raspberry or linux??

Thx
Rainer

lelegard, 2024-09-08:
Hi @rainerfritz 

Please use `tsscan --delivery-system dvb-t2`. Let us know if it works.

As documented with `tsscan --help`, it uses the "default delivery system" for the tuner, which is DVB-T in that case: DVB-T comes first in "Sony CXD2880, DVB-T, DVB-T2". If you want to scan another delivery system, you have to specify it.

There is recent issue #1505 about the fact that `tsscan` could find DVB-T streams only because `tsscan` only uses one delivery system. Originally, this was done to save time. Scanning two delivery systems doubles the scan time because you have to try each of them, one by one, on each frequency. As a solution, I added the possibility to specify `--delivery-system` several times to list all systems to try, at the expense of the scan time of course.

With the general switch to DVB-T2, it may make sense to use DVB-T2 as default on all tuners which support T and T2. But it would break existing scripts which scan networks all over the place.

lelegard, 2024-09-08:
Just to be sure, since you also mention `--delivery-system DVB-T2`, does it work or not?

Could you try this:
~~~
tsp -v -I dvb --delivery-system DVB-T2 --frequency 498,000,000 -P dump -O drop
~~~

rainerfritz, 2024-09-08:
Hi Thierry!

Thanks for that really fast reply, much appreciated!

I don't know what the problem was, maybe a typo or whatever, but it works now. Thanks a lot for the fast help!!

May I ask you one more question. I am currently recording the TS and then analysing it with tstables to get a JSON output. I want to do that on the fly and send the tables in JSON format via localhost to another application. Think that it is possible, as we can send the TS from tsp to tstables directly via a pipe? 

As mentioned I want to monitor the AIT tables specially.

lelegard, 2024-09-08:
> I don't know what the problem was, maybe a typo or whatever, but it works now.

Glad to read that it works.

> I am currently recording the TS and then analysing it with tstables to get a JSON output. I want to do that on the fly and send the tables in JSON format via localhost to another application. Think that it is possible, as we can send the TS from tsp to tstables directly via a pipe?

First, there is no need to pipe `tsp` into `tstables`. The plugin `tables` is identical to `tstables` (same options, same code) and works inside the `tsp` pipeline.

Then, the plugin `tables` (and `tstables`) can send the extracted sections over UDP. See the [reference documentation for plugin tables](https://tsduck.io/download/docs/tsduck.html#_tables), especially the section "UDP/IP logging options".

Currently, the sections are sent in binary format over UDP. If you want to convert them into JSON (or XML) on reception, I would suggest a shell script with `nc` (netcat) for reception, piped into `tstabdump`. Don't forget to use `--no-encapsulation` with `-P tables --ip-udp` to send the raw sections, without encapsulation.

> If all works well and the project gets accepted on my side, I will ask for permission to promote our broadcasting station on tsduck page.

Of course, you are welcome, this is mutual publicity. There is a [references page here](https://tsduck.io/references/).

lelegard, 2024-09-08:
And because it can be useful and it is easy to implement, I just added an option `--udp-format` to `tstables` and plugin `tables`. To send tables in JSON format over UDP, use:

~~~
tsp .... -P tables --ip-udp address:port --udp-format json ...
~~~

rainerfritz, 2024-09-08:
Great! Thats a really useful option! Thx a lot for that!

closed 2025-02-09 by github-actions[bot]