tsduck/tsduck #1516: [BUG] RIST output plugin crashes after calling rist_sender_create

jheliker, 2024-09-09:
#### Bug description:
RIST output plugin crashes after calling rist_sender_create.

#### How to reproduce:
Any tsp command line with -O rist seems to crash on my system.

#### Expected behavior:
I would expect to be able to connect to other RIST-compatible systems such as AWS MediaConnect.

#### Errors and logs:
~~~
tsp --debug=10 -I file "C:\473.ts" -O rist url rist://127.0.0.1:1024
tsp: Debug[10]: debug level set to 10
tsp: Debug: ====> tsp --debug=10
* Debug[10]: debug level set to 10
* Debug[10]: file: debug level set to 10
* Debug: file: ====> tsp -I file 'C:\\473.ts'
* Debug[10]: rist: debug level set to 10
* Debug: rist: ====> tsp -O rist url rist://127.0.0.1:1024
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug[10]: file: initBuffer(..., pkt_first = 0, pkt_cnt = 89,240, input_end = false, aborted = false, bitrate = 0)
* Debug: file: detected TS file format TS
* Debug: file: initial buffer load: 44,620 packets, 8,388,560 bytes
* file: initial input bitrate is 19,392,946 b/s
* Debug[10]: rist: initBuffer(..., pkt_first = 0, pkt_cnt = 44,620, input_end = false, aborted = false, bitrate = 19,392,946)
* Debug[10]: file: initBuffer(..., pkt_first = 44,620, pkt_cnt = 44,620, input_end = false, aborted = false, bitrate = 19,392,946)
* Debug: rist: calling rist_sender_create, profile: 1
~~~

#### Environment:
 - OS: Windows 
 - OS version: Windows 11 Business 22H2 64-bit
 - TSDuck full version: 3.38-3822
 - Installation type: official binary, TSDuck-Win64-3.38-3822-Portable.zip

#### Additional information:
~~~
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.38-3822
Built Aug 21 2024 - 12:39:02
Using MSVC 19.41.34120.00, C++ std 2017.03
System: Windows (Windows 6.2 Build 9200), on Intel x86-64, 64-bit, little-endian, page size: 4096 bytes
Acceleration: CRC32: no
Bitrate: 64-bit floating-point value
Dektec: DTAPI: 6.5.0.229
VATek: libvatek version 3.01
Cryptographic library: Microsoft BCrypt
Web library: WinInet
SRT library: libsrt version 1.5.3
RIST library: librist version v0.2.10, API version 4.4.0
~~~

lelegard, 2024-09-10:
Hi @jheliker 

There is a known issue with libsrt on Windows. The libsrt library is natively developed on / for Linux systems. Its support for Windows and Visual Studio is not perfect. If you want to use RIST, I recommend Linux or macOS.

lelegard, 2024-09-10:
I just checked if there was something new in librist, maybe a fix. I was surprised to see that the [librist repository](https://code.videolan.org/rist/librist) had not even a single commit in the last 10 months. The last commit in the repo is tagged v0.2.10. This was the official release 0.2.10, in October 2023. Nothing happened since then.

By contrast, the last commit in the [libsrt repository](https://github.com/Haivision/srt) occured 30 minutes ago, 6 commits were pushed yesterday, etc.

I am neutral in the SRT / RIST competition, I have no opinion. However, there is obviously much more activity in the former.

closed 2025-03-30 by github-actions[bot]