tsduck/tsduck #1517: [BUG] tsscan --verbose --japan --save-channels channels.xml --uhf-band --debug defaults to using DVB-T delsys

trollcop, 2024-09-11:
#### Bug description:
running `tsscan --verbose --japan --save-channels channels.xml --uhf-band --debug` will fail in Japan because for some reason it will default to DVB-T delivery system.

#### How to reproduce:
run the above command on a dvb-api supported ISDB-T tuner

#### Expected behavior:
Scanning UHF band using ISDB-T

#### Errors and logs:
~~~
tsscan: Debug: debug level set to 1
tsscan: Debug: ====> tsscan --verbose --japan --save-channels channels2.xml --uhf-band --debug
tsscan: Debug: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb0.frontend0/device/idVendor = "" (failure)
tsscan: Debug: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb0.frontend0/device/idProduct = "" (failure)
tsscan: Debug: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb0.frontend0/device/manufacturer = "" (failure)
tsscan: Debug: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb0.frontend0/device/product = "" (failure)
tsscan: Debug: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb0.frontend0/device/version = "" (failure)
tsscan: Debug: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb0.frontend0/device/serial = "" (failure)
scanning channel 13, 473,142,857 Hz
tsscan: Debug: trying offset 0
tsscan: Debug: using DVB-T as default delivery system
tsscan: Debug: adding standards DVB to ISDB, Japan
tsscan: Debug: starting discarding frontend events
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 1 DTVProperties:
tsscan: Debug: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 11 DTVProperties:
tsscan: Debug: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 3 (0x00000003)
tsscan: Debug: [1] cmd = 3 (DTV_FREQUENCY), data = 473142857 (0x1C339649)
tsscan: Debug: [2] cmd = 4 (DTV_MODULATION), data = 3 (0x00000003)
tsscan: Debug: [3] cmd = 6 (DTV_INVERSION), data = 2 (0x00000002)
tsscan: Debug: [4] cmd = 5 (DTV_BANDWIDTH_HZ), data = 8000000 (0x007A1200)
tsscan: Debug: [5] cmd = 36 (DTV_CODE_RATE_HP), data = 9 (0x00000009)
tsscan: Debug: [6] cmd = 37 (DTV_CODE_RATE_LP), data = 9 (0x00000009)
tsscan: Debug: [7] cmd = 39 (DTV_TRANSMISSION_MODE), data = 1 (0x00000001)
tsscan: Debug: [8] cmd = 38 (DTV_GUARD_INTERVAL), data = 0 (0x00000000)
tsscan: Debug: [9] cmd = 40 (DTV_HIERARCHY), data = 0 (0x00000000)
tsscan: Debug: [10] cmd = 1 (DTV_TUNE), data = 4294967295 (0xFFFFFFFF)
~~~

#### Environment:
 - OS: Linux
 - OS version: any
 - TSDuck full version: 3.38-3822
 - Installation type: official binary ubuntu 64

lelegard, 2024-09-11:
Hi @trollcop 

When `--delivery-system` is not specified, `tsscan` uses the "default" delivery system of the tuner. What is your tuner exactly? Could you post the output of `tslsdvb -v`?

If ISDB-T is the first (or only) supported system in the tuner, `tsscan` will default to ISDB-T. If it is a mixed tuner, with DVB-T and ISDB-T support, and DVB-T comes first, then the default will be DVB-T.

In the [tsscan documentation](https://tsduck.io/download/docs/tsduck.html#_tsscan), `--japan` is documented as a synonym for `--isdb --default-charset ARIB-STD-B24 --hf-band-region japan`. There is no `--delivery-system`.

Here, `--isdb` means ISDB signalization, not modulation. As [documented here](https://tsduck.io/download/docs/tsduck.html#opt-default-std), the ISDB signalization is not an original, independent one. It is a mixture of DVB signalization, with some modified syntax and/or semantics, and some additional tables. Therefore, if the first table which is received in a stream is a DVB table and not one of the additional ISDB table, it will be initially interpreted as DVB, with some possible misinterpretation of the fields for which ISDB modified the semantics. The option `--isdb` (contained in `--japan`) specifies to interpret all tables as ISDB from the beginning. There is no modulation here.

If your tuner has DVB-T as default, this is not a bug.

It could be a request for enhancement. For instance to add `--delivery-system isdb-t` in `--japan`. But what about ISDB-S? If you want to scan a satellite with an ISDB-S tuner (with ISDB-S as default system), using `--japan` would force `--delivery-system isdb-t` as default and it wouldn't work. So, you would be forced to explicitly set  `--delivery-system isdb-s` even though it could be the default or even only supported system by the tuner.

We could request that `--japan` would force `--delivery-system` to some supported variant of ISDB that will be found in the tuner. But that does not fit in the command argument resolution system of TSDuck. The command line argument processing comes first, independently of the commands and plugins. The tuners are very specific beasts which are only known by the plugin `dvb` (historically misnamed, I know) and `tsscan`.

trollcop, 2024-09-11:
Hi,

I see, your explanation makes sense. The card i'm using is a TBS multistandard card, which does DVB-T/T2/C/ISDB-T/etc. TBS6209SE. You are right, it probably reports DVB-T as default delsys.

I'll go ahead and close this, understood.

lelegard, 2024-09-11:
@trollcop, I assume that I can add TBS6209SE in the [list of tested tuners](https://tsduck.io/download/docs/tsduck.html#_tested_devices) on Linux. Not tested on Windows, I presume...

trollcop, 2024-09-11:
Here's output of `tslsdvb -v`
[dump.txt](https://github.com/user-attachments/files/16971058/dump.txt)

> I assume that I can add TBS6209SE in the [list of tested tuners](https://tsduck.io/download/docs/tsduck.html#_tested_devices) on Linux. Not tested on Windows, I presume...

I've only tested DVBC/B and ISDB-T but yes, those 2 work fine ( once we get the DVBC/B thing figured out in #1518 ).

> Does it really have 8 tuners?
Yes, it has 8x of Sony CXD6802GL modules which are SiP of tuner + demod in a smallish package, plus a RF splitter, as well as the usual TBS ECP3 TS DMA engine FPGA.

lelegard, 2024-09-12:
Nice beast. Above $400, quite expensive for a tuner, but probably worth it.
I have added it in the list of tested tuners.
Thanks.

trollcop, 2024-10-02:
Closing as resolved, thanks!

closed 2024-10-02 by trollcop