tsduck/tsduck #1518: [Q] How to tune DVB-C/B aka J.83B 256-QAM transponder over terrestrial?

trollcop, 2024-09-11:
```
tsscan --nit-scan --delivery-system DVB-C/B -f 112750000 --debug -m 256-qam -s 5900000
```
doesn't lock

```
dvbv5-zap -vvv -c channel.conf test
```
where `channel.conf` contains the following, works:
```
[test]
        DELIVERY_SYSTEM = DVBC/ANNEX_B
        FREQUENCY = 112750000
        MODULATION = QAM/256
```

I see dvb-c/b has unsupported stuff in a bunch of places, what does that actually "mean"? Is there anything needed to support other than passing this tuning parameter to dvb-api?

lelegard, 2024-09-11:
Hi @trollcop 

I have never used DVB-C, /A or /B, with TSDuck because I never had access to a real cable network. So, I cannot tell by my own experience.

As a troubleshooting method, I suggest:

1. Try direct reception before full scanning: `tsp -d -I dvb ... -O drop`
2. If you receive nothing, make sure to specify all modulation parameters, typically `--fec-inner`. Each command uses its own defaults and maybe `dvbv5-zap` and TSDuck don't use the same defaults for the unspecified options.

In the TSDuck user's guide, DVB-C/B is marked as "unsupported", unlike /A and /C. I cannot remember what it means. Probably, at some point, the Linux DVB API did not support it. It has probably changed if you can successfully tune using `dvbv5-zap`. Fixing this in TSDuck requires a developer with access to a real DVB-C/B network with real streams. I do not have that access.

trollcop, 2024-09-11:
i can try to add support and test if you point out where i should start. I'll try pre-tuning and running tsp when I'm back at computer.

I've added test support for tuning this in a simple tuning/capture app and it works without any specific fec/etc settings:

https://github.com/trollcop/recskapa/commit/649ecc5c42617b6241d1b5b88ccc8a1daae203bc

I've got a network full of dvbc/b streams and would have no problem giving you access to it as needed.

trollcop, 2024-09-11:
Here's output of `tslsdvb -v` when one of the tuners is tuned + locked:

~~~
0: /dev/dvb/adapter0 ("Turbosight TBS 6209SE(Octa DVB-T/T2/C/C2/ISDB-T/C/A", DVB-T, DVB-T2, ATSC, ISDB-T, DVB-C, 2, DVB-C/C)
   Device: /dev/dvb/adapter0/frontend0

     Status: has signal, has carrier, has viterbi, has sync, has lock

     Signal: locked: yes, strength: -32.41 dB, SNR: 34.496 dB

     Delivery system ................................ 2
     Frequencies:
       Current ............................ 112,750,000 Hz
       Min ................................. 47,000,000 Hz
       Max ................................ 862,000,000 Hz
       Step .................................... 62,500 Hz
       Tolerance .................................... 0 Hz
     Symbol rates:
       Current .............................. 5,600,000 sym/s
       Min .................................. 1,700,000 sym/s
       Max .................................. 7,200,000 sym/s
       Tolerance .................................... 0 sym/s
     Spectral inversion .......................... auto
     FEC(inner) .................................. auto
     Modulation ............................... 256-QAM

     Capabilities: inversion auto, FEC 1/2, FEC 2/3, FEC 3/4, FEC 4/5, FEC 5/6,
       FEC 7/8, FEC auto, 16-QAM, 32-QAM, 64-QAM, 128-QAM, 256-QAM, QAM auto,
       transmission mode auto, guard interval auto, 2nd generation, recover,
       mute TS
~~~

It seems fec-inner should just work with auto - and since I've written drivers for Sony CXD28xx based hardware before, I can tell you that it doesn't even support setting FEC etc. since its all auto. Tuning parameters for DVB-C/J.83 is just frequency and symbol rate.

It is curious that delsys printed in above log is `2`, I think that's due to a typo @ tsDeliverySystem.cpp:98 - will change and rebuild and see.

I will poke at the rest of the sores you pointed out and see if I can make something work.

trollcop, 2024-09-11:
Hey uhhh, just fixing 
https://github.com/tsduck/tsduck/blob/2ea222a3cb4ab6c7ce6f9fcd9a299b4f67618cee/src/libtsduck/dtv/broadcast/tsDeliverySystem.cpp#L98

to DVB-C/B makes it tune and work in dvb plugin

~~~
./tsp -d -I dvb --delivery-system DVB-C/B --frequency 112750000 -m 256-qam -s 5600000 -O drop
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -d
* Debug: debug level set to 1
* Debug: trying to load "/home/timecop/dev/tsduck/bin/release-x86_64-epgstation/tsplugin_dvb.so"
* Debug: registering input plugin "dvb", status: ok
* Debug: dvb: debug level set to 1
* Debug: dvb: ====> tsp -I dvb --delivery-system DVB-C/B --frequency 112750000 -m 256-qam -s 5600000
* Debug: drop: debug level set to 1
* Debug: drop: ====> tsp -O drop
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: dvb: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb0.frontend0/device/idVendor = "" (failure)
* Debug: dvb: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb0.frontend0/device/idProduct = "" (failure)
* Debug: dvb: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb0.frontend0/device/manufacturer = "" (failure)
* Debug: dvb: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb0.frontend0/device/product = "" (failure)
* Debug: dvb: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb0.frontend0/device/version = "" (failure)
* Debug: dvb: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb0.frontend0/device/serial = "" (failure)
* * dvb: using /dev/dvb/adapter0 (DVB-T, DVB-T2, ATSC, ISDB-T, DVB-C, DVB-C/B, DVB-C/C)
Debug: loading names file tsduck.dtv.names
* Debug: dvb: adding standards DVB to none
* Debug: dvb: starting discarding frontend events
* Debug: dvb: finished discarding frontend events
* Debug: dvb: tuning on /dev/dvb/adapter0/frontend0
* Debug: dvb: 1 DTVProperties:
* Debug: dvb: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
* Debug: dvb: tuning on /dev/dvb/adapter0/frontend0
* Debug: dvb: 6 DTVProperties:
* Debug: dvb: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 2 (0x00000002)
* Debug: dvb: [1] cmd = 3 (DTV_FREQUENCY), data = 112750000 (0x06B86DB0)
* Debug: dvb: [2] cmd = 4 (DTV_MODULATION), data = 5 (0x00000005)
* Debug: dvb: [3] cmd = 6 (DTV_INVERSION), data = 2 (0x00000002)
* Debug: dvb: [4] cmd = 8 (DTV_SYMBOL_RATE), data = 5600000 (0x00557300)
* Debug: dvb: [5] cmd = 1 (DTV_TUNE), data = 4294967295 (0xFFFFFFFF)
* dvb: tuned to transponder --delivery-system DVB-C/B --frequency 112,750,000 --symbol-rate 5,600,000 --fec-inner auto --modulation 256-QAM
* Debug: dvb: starting tuner reception
* Debug: dvb: tuner reception started
* Debug: dvb: 6 DTVProperties (statistics result):
* Debug: dvb: [0] cmd = 62 (DTV_STAT_SIGNAL_STRENGTH), count = 2, stat = {scale: 1, value: -32410}, {scale: 2, value: 63500}
* Debug: dvb: [1] cmd = 63 (DTV_STAT_CNR), count = 2, stat = {scale: 1, value: 34781}, {scale: 2, value: 53136}
* Debug: dvb: [2] cmd = 66 (DTV_STAT_POST_ERROR_BIT_COUNT), count = 1, stat = {scale: 3, value: 0}
* Debug: dvb: [3] cmd = 67 (DTV_STAT_POST_TOTAL_BIT_COUNT), count = 1, stat = {scale: 3, value: 0}
* Debug: dvb: [4] cmd = 68 (DTV_STAT_ERROR_BLOCK_COUNT), count = 0, stat =
* Debug: dvb: [5] cmd = 69 (DTV_STAT_TOTAL_BLOCK_COUNT), count = 0, stat =
* dvb: locked: yes, strength: -32.41 dB, SNR: 34.781 dB
* Debug: dvb: initial buffer load: 44,620 packets, 8,388,560 bytes
* dvb: initial input bitrate is 40,544,293 b/s
* Debug: dvb: input thread started
* Debug: control commands: found 0 packet processor plugins
* Debug: drop: output thread started
* Debug: dvb: input: got bitrate 40,544,293 b/s
~~~

lelegard, 2024-09-12:
Damn it! Thanks a lot for finding this. Another wonderful copy/paste story...

I will fix it right away. Because this enum/string table is used everywhere, including in the definition of option `--delivery-system`, specifying `--delivery-system DVB-C/B` actually resulted in using DVB-C/A.

See the definition of option:
https://github.com/tsduck/tsduck/blob/2ea222a3cb4ab6c7ce6f9fcd9a299b4f67618cee/src/libtsduck/dtv/broadcast/tsModulationArgs.cpp#L1443

trollcop, 2024-09-12:
> including in the definition of option --delivery-system, specifying --delivery-system DVB-C/B actually resulted in using DVB-C/A.

Yes, I think that's what happened. it was tuning A despite specifying B in cli.

~~~
(old version)
tsp -d -I dvb --delivery-system DVB-C/B --frequency 112750000 -m 256-qam -s 5600000 -O drop
...
* Debug: dvb: 7 DTVProperties:
* Debug: dvb: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 1 (0x00000001)
~~~

sets it to 1 which is incorrect. with the copypaste fix, it's correct at
~~~
* Debug: dvb: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 2 (0x00000002)
~~~

trollcop, 2024-09-13:
okay, so, tuning got fixed... but nit-scan isn't doing too well. It keeps defaulting to DVB-C, probably needs some extra hacks for ISDB... here's some logs

~~~
567750000 --debug -m 256-qam -s 5600000 --adapter 1
tsscan: Debug: debug level set to 1
tsscan: Debug: ====> tsscan --nit-scan --delivery-system DVB-C/B -f 567750000 --debug -m 256-qam -s 5600000 --adapter 1
tsscan: Debug: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb1.frontend0/device/idVendor = "" (failure)
tsscan: Debug: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb1.frontend0/device/idProduct = "" (failure)
tsscan: Debug: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb1.frontend0/device/manufacturer = "" (failure)
tsscan: Debug: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb1.frontend0/device/product = "" (failure)
tsscan: Debug: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb1.frontend0/device/version = "" (failure)
tsscan: Debug: /sys/devices/pci0000:00/0000:00:16.0/0000:0b:00.0/dvb/dvb1.frontend0/device/serial = "" (failure)
tsscan: Debug: adding standards DVB to none
tsscan: Debug: starting discarding frontend events
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter1/frontend0
tsscan: Debug: 1 DTVProperties:
tsscan: Debug: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: tuning on /dev/dvb/adapter1/frontend0
tsscan: Debug: 6 DTVProperties:
tsscan: Debug: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 2 (0x00000002)
tsscan: Debug: [1] cmd = 3 (DTV_FREQUENCY), data = 567750000 (0x21D72D70)
tsscan: Debug: [2] cmd = 4 (DTV_MODULATION), data = 5 (0x00000005)
tsscan: Debug: [3] cmd = 6 (DTV_INVERSION), data = 2 (0x00000002)
tsscan: Debug: [4] cmd = 8 (DTV_SYMBOL_RATE), data = 5600000 (0x00557300)
tsscan: Debug: [5] cmd = 1 (DTV_TUNE), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: got 10000 packets
tsscan: Debug: adding standards MPEG to DVB
tsscan: Debug: got table id 0x00 on PID 0x0000
tsscan: Debug: got 10000 packets
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got table id 0x40 on PID 0x0010
tsscan: Debug: got table id 0x41 on PID 0x0010
tsscan: Debug: got 10000 packets
tsscan: Debug: got 10000 packets
tsscan: Debug: got table id 0x42 on PID 0x0011
tsscan: Debug: * tuning to --delivery-system DVB-C --frequency 567,750,000 --symbol-rate 5,600,000 --fec-inner 8/9 --modulation 256-QAM
tsscan: Debug: starting discarding frontend events
tsscan: Debug: one frontend event discarded
tsscan: Debug: one frontend event discarded
tsscan: Debug: one frontend event discarded
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter1/frontend0
tsscan: Debug: 1 DTVProperties:
tsscan: Debug: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: tuning on /dev/dvb/adapter1/frontend0
tsscan: Debug: 7 DTVProperties:
tsscan: Debug: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 1 (0x00000001)
tsscan: Debug: [1] cmd = 3 (DTV_FREQUENCY), data = 567750000 (0x21D72D70)
tsscan: Debug: [2] cmd = 4 (DTV_MODULATION), data = 5 (0x00000005)
tsscan: Debug: [3] cmd = 6 (DTV_INVERSION), data = 2 (0x00000002)
tsscan: Debug: [4] cmd = 9 (DTV_INNER_FEC), data = 8 (0x00000008)
tsscan: Debug: [5] cmd = 8 (DTV_SYMBOL_RATE), data = 5600000 (0x00557300)
tsscan: Debug: [6] cmd = 1 (DTV_TUNE), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: 6 DTVProperties (statistics result):
tsscan: Debug: [0] cmd = 62 (DTV_STAT_SIGNAL_STRENGTH), count = 2, stat = {scale: 1, value: -41000}, {scale: 2, value: 62230}
tsscan: Debug: [1] cmd = 63 (DTV_STAT_CNR), count = 2, stat = {scale: 1, value: 33926}, {scale: 2, value: 51824}
tsscan: Debug: [2] cmd = 66 (DTV_STAT_POST_ERROR_BIT_COUNT), count = 1, stat = {scale: 3, value: 0}
tsscan: Debug: [3] cmd = 67 (DTV_STAT_POST_TOTAL_BIT_COUNT), count = 1, stat = {scale: 3, value: 0}
tsscan: Debug: [4] cmd = 68 (DTV_STAT_ERROR_BLOCK_COUNT), count = 0, stat =
tsscan: Debug: [5] cmd = 69 (DTV_STAT_TOTAL_BLOCK_COUNT), count = 0, stat =
* Frequency: 567,750,000 Hz, locked: no, strength: -41 dB, SNR: 33.926 dB
~~~

here's NIT:

~~~
* NIT Actual, TID 0x40 (64), PID 0x0010 (16)
  Version: 21, sections: 1, total size: 870 bytes
[... 267 lines cut ...]
      Service id: 0x8267 (33383), Type: 0x01 (Digital television service)
      Service id: 0x8275 (33397), Type: 0x01 (Digital television service)
      Service id: 0x827A (33402), Type: 0x01 (Digital television service)
      Service id: 0x8296 (33430), Type: 0x01 (Digital television service)
      Service id: 0x82A8 (33448), Type: 0x01 (Digital television service)
    - Descriptor 1: Cable Delivery System (0x44, 68), 11 bytes
      Frequency: 119.2500 MHz
      Symbol rate: 5.6000 Msymbol/s
      Modulation: 256-QAM
      Outer FEC: code 3 (reserved), Inner FEC: 8/9 conv. code rate
    Transport Stream Id: 0x4032 (16434), Original Network Id: 0x000A (10)
    - Descriptor 0: Service List (0x41, 65), 15 bytes
      Service id: 0x8209 (33289), Type: 0x01 (Digital television service)
      Service id: 0x8222 (33314), Type: 0x01 (Digital television service)
      Service id: 0x8238 (33336), Type: 0x01 (Digital television service)
      Service id: 0x8245 (33349), Type: 0x01 (Digital television service)
      Service id: 0x82B4 (33460), Type: 0x01 (Digital television service)
    - Descriptor 1: Cable Delivery System (0x44, 68), 11 bytes
      Frequency: 320.7500 MHz
      Symbol rate: 5.6000 Msymbol/s
      Modulation: 256-QAM
      Outer FEC: code 3 (reserved), Inner FEC: 8/9 conv. code rate
~~~

I guess it tries to tune it as DVB-C/A instead of DVB-C/B, or something idk.

lelegard, 2024-09-13:
> I guess it tries to tune it as DVB-C/A instead of DVB-C/B, or something idk.

Exactly. A NIT-scan is based on the system delivery descriptor for each TS. These descriptors are defined by DVB.

ISDB redefines the satellite delivery descriptor for ISDB-S with same tag as the DVB one but slightly different layout (a standardization nightmare). ISDB also defines a specific descriptor for ISDB-T, with a distinct tag from the DVB terrestrial delivery descriptor (so this one is OK).

But there is nothing for cable. There is no ISDB-C, isn't it? And the DVB-defined cable delivery descriptor does not provide any information to discriminate between DVB-C/A/B/C.

So, currently, when the tuning parameters are extracted from a cable system delivery descriptor, the type is assumed to be DVB-C (meaning /A by default).

https://github.com/tsduck/tsduck/blob/1c16295ef9bcc7969a5bc1c7648fec7615cdc989/src/libtsduck/dtv/broadcast/tsModulationArgs.cpp#L586

We should add something external to the signalization to force the interpretation of a cable delivery descriptor as DVB-C/B (or /C for generalization). There are several possibilities.

1. This can be an option in `tsscan` meaning "in a NIT scan, force all delivery systems to this value". This will work but this is limited to `tsscan`, not used in other contexts.
2. This can be an ISDB discriminator such as used in the satellite delivery descriptor which uses the same tag but the interpretation of which varies depends on the TS standard, DVB or ISDB.
 
The problem with the second solution is that the satellite interpretation is well-defined in ARIB STD-B10 and can be safely considered as part of ISDB signalization. AFAIK, the ARIB standards say nothing about cable. There is no strictly defined ISDB signalization on the cable delivery descriptor (unlike satellite). So, assuming DVB-C/B in a cable descriptor as soon as we are in an ISDB context is maybe not legitimate. Are we sure that all Japanese cable networks use DVB-C/B? What about other ISDB countries in South America, the Philippines and Africa?

lelegard, 2024-09-13:
Maybe there is a more automatic way. 

Can we consider that a cable network is always homogeneous? All DVB-C/A or /B or /C, but never a mixture? It sounds rational but is this true? Satellite and terrestrial are different. You can have DVB-T and -T2, or -S and -S2, on the same network. But you usually do not have DVB-S and ISDB-S on the same network.

If we assume that any DVB-C network always uses the same variant A, B or C, considering that the cable delivery descriptor has no data to discriminate between them, `tsscan` can safely assume that the initial system delivery can be forced on all TS when it is one of DVB-C/x.

If we cannot assume that all DVB-C network are homogeneous, then we need to add some option such as `--homogeneous-cable-network` to `tsscan`.

Of course, this only applies to DVB-C/x. In the case of a DVB-C2 TS, there is a specific C2 delivery descriptor.

trollcop, 2024-09-13:
> If we cannot assume that all DVB-C network are homogeneous, then we need to add some option such as --homogeneous-cable-network to tsscan.

there's a nit-other that links to isdb-t channel that's used for discovery of cable network (it has nit-other with the dvb-c/b entries but from what i can see in the current qam nit they're all the same.

i don't have a lot of computer access during the next week so I'm not sure how much time I'll have to dig through the arib specs to look for dvbc specific descriptors.

i like your suggestion to force homogeneous network with a switch, but i wouldn't add it as default under --isdb as this is one specific network (flets tv, catv over fiber) which isn't IP based, and i don't have an easy way to scan others like jcom or similar.

lelegard, 2024-09-13:
> there's a nit-other that links to isdb-t channel 

Currently, `tsscan` uses the NIT Actual only. The idea is to scan a network starting from a TS of _that_ network. Therefore, on the initial TS, the target NIT is the actual one.

> dig through the arib specs to look for dvbc specific descriptors.

AFAIK, the only ARIB specs containing signalisation are STD-B10 and STD-B21 (see [bibliography](https://tsduck.io/download/docs/tsduck.html#bibliography)) and they do not mention a cable descriptor. 

The only spec I do not have access to is ARIB STD-B16 (ISDB Standard Digital Receiver for Satellite) which is not free and not available in English. Given the reference to satellite, I don't think we could find anything about cable.

The only delivery system for which there is no signalisation is DVB-C (/A, /B, /C). So, the only case with a problem is a TS description containing a cable_delivery_system_descriptor. All the rest is addressed using other delivery descriptors.

We face only two cases:

1. On a cable, inside one "network" (in the DVB sense), all TS use the same Annex A, B or C. In that case, in the presence of a cable_delivery_system_descriptor, we scan that TS using the same variant of DVB-C that was used to scan the initial NIT.
2. On the same cable, several variants of A, B or C coexist. Since there is no way to signal which variant is used on which TS, I consider this case as unrealistic and unpractical because no official cable set-top box could successfully scan the network.

Therefore, I will implement solution 1. We will see how it behaves. We count on you @trollcop to provide some feedback, when you are available, because you are probably the only TSDuck user who uses it on non-/A DVB-C networks.

trollcop, 2024-09-13:
> On a cable, inside one "network" (in the DVB sense), all TS use the same Annex A, B or C. In that case, in the presence of a cable_delivery_system_descriptor, we scan that TS using the same variant of DVB-C that was used to scan the initial NIT.

That sounds reasonable. As you say in variant 2, it would make no sense to mix and match delivery systems as no box would be able to scan unless hard coded, which i really wouldn't put past a Japanese system, but is probably unlikely for anywhere else. I guess this really only affects cable because even in satellite you'd have a delivery descriptor that would say dvbs or s2 etc, leading to no ambiguity.

and yeah no issues with testing. i still have access to various things remotely even not in front of PC.

lelegard, 2024-09-14:
I pushed an enhancement of the NIT scan to propagate the original DVB-C/x annex to the scanned transport streams, when they are signalled using a cable_delivery_system_descriptor.

This had other consequences that were not previously considered, in the plugin `nitscan` and in the [format of the channel configuration files](https://tsduck.io/download/docs/tsduck.html#chap-chanconfig).

Let us know if you can now successfully scan a DVB-C/B network.

lelegard, 2024-09-14:
In the test suite, I added test-161 which uses the TSDuck builtin tuner emulator to scan a DVB-C/B network with two TS. The scan fails before the fix (`tsscan` tries to use DVB-C on the scanned TS) and passes after the fix. However, a test on a real DVB-C/B network remains necessary.

trollcop, 2024-09-15:
I've pulled latest revision today and built.
here's output log and generated xml from:
`./tsscan --japan --nit-scan --delivery-system DVB-C/B -f 112750000 --debug -m 256-qam -s 5600000 --adapter 0 -l --update-channels skapa2.xml`

[output.log](https://github.com/user-attachments/files/17005552/output.log)
[skapa2.zip](https://github.com/user-attachments/files/17005555/skapa2.zip)

.zip is xml zipped since I can't attach it.

Other than "locked: no" during scan i think it looks ok? I'm not sure what the locked flag there signifies because its actually locking and receiving ts?

lelegard, 2024-09-15:
Thanks for the feedback. It looks good.

The service names in output.log look corrupted but it's probably a character representation issue during a transfer because they look correct in the XML file. Normally, the output is forced in UTF-8 representation. Are the service names correctly displayed by tsscan?

Concerning the signal lock status, TSDuck only displays what the driver reports. The status is obtained on the front end device using an `ioctl` with `FE_READ_STATUS` command. In the returned value, the lock status is obtained from bit mask `FE_HAS_LOCK`. If the driver reports crap, TSDuck reports the same crap.

The accuracy of reporting in LinuxTV drivers is decreasing over time. It seems that developers no longer care about doing everything correctly. When the signal is tuned and the TS is read, they stop there and don't care about accurate reporting. When I started with Linux tuners, around 2005, the reporting was accurate: status, BER, SNR, modulation parameters. Getting the latter was especially useful with DVB-T where you can tune on a given frequency without accurate parameters and when the signal is locked the drivers used to report the actual parameters. After a couple of years, I ported TSDuck to Windows and realised that Windows BDA drivers were completely unreliable and reported crap. After nearly two decades of tuning, I would have hoped that the Windows drivers would improve up to the Linux level. However, I realise that this is the opposite: the Linux drivers worsen down to the Windows level...

trollcop, 2024-09-15:
> The service names in output.log look corrupted but it's probably a character representation issue during a transfer because they look correct in the XML file. Normally, the output is forced in UTF-8 representation. Are the service names correctly displayed by tsscan?

I used putty log since `> logfile` didn't work and only did 'printable' output which skipped non-ascii stuff. yes, it looked fine on console as UTF-8. ARIB encoding was properly picked up.

Regarding lock, it's definitely locked (since my other rec app depends on waiting for lock before going to get TS), so I'm not sure why it would report unlocked. Maybe it needs to poll a little bit for lock status to change? idk. sony frontend hardware has pretty granular lock reporting, for example in ISDB-T they'd have TMCC (1seg/some layers locked) followed by TS lock (actually error-free TS available on all segments), so it's capable of reporting TS lock on DVB-C for sure.

But yeah, overall scan seems to be working correctly. There are indeed 20 transponders and service IDs seem to match. btw, fun (?) fact, service ID determines the visible channel # but with top bit removed. i.e. home drama channel SID 0x8295 = ch 0x295 = ch661 on UI.

Regarding signal reporting accuracy, I think dvbv5 actually improved on things by allowing to store scale/dbm based values so that you can get RF input levels and C/N etc as proper fractions and not just random numbers. I remember last I touched dvb you'd just get a random uint16 as "SNR" and no further explanation of its range or anything.

trollcop, 2024-10-02:
Closing as resolved, thanks!

closed 2024-10-02 by trollcop