tsduck/tsduck #1519: [BUG] Program_splice_flag is set 0 when component tag is missing in xml

gauravv5, 2024-09-12:
#### Bug description:

#### How to reproduce:
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
    <splice_information_table
        protocol_version="1"
        pts_adjustment="0"
        tier="0x0FFF">

        <splice_insert
            splice_event_id="1002"
            splice_event_cancel="false"
            out_of_network="true"
            unique_program_id="1234"
            avail_num="1"
            avails_expected="3">

            <break_duration
                auto_return="true" 
                duration="0x19BFCC0"/>  

        </splice_insert>

    </splice_information_table>
</tsduck>

My splice.xml 

#### Expected behavior:
program splice flag is set 0 whereas as per documentation as it should set to 1 when component tag is not in xml

#### Environment:
 - OS: 
 - OS version: 
 - TSDuck full version:

lelegard, 2024-09-12:
Hi @gauravv5 

Thanks for reporting this. There is a bug indeed, but not exactly this one.

The problem was the missing `pts_time` attribute in your `<splice_insert>` structure. When there is no component, then `program_splice_flag` should be 1. However, in that case, the program-wide splice time (attribute `pts_time`) must be set.

From SCTE 35 spec:
~~~
if(splice_event_cancel_indicator == ‘0’) {
    ...
    if((program_splice_flag == ‘1’) && (splice_immediate_flag == ‘0’))
        splice_time()
    if(program_splice_flag == ‘0’) {
        component_count
        ...
~~~

The bug in the code was to not check the consistency of canceled + immediate + splice_time + components. The code erroneously assumed that the absence of splice_time meant `program_splice_flag` was zero, with a component count equals to zero.

I fixed this. Now, the provided XML reports an error because the attribute `pts_time` is missing.

gauravv5, 2024-09-12:
Thanks for the support mate

gauravv5, 2024-09-26:
tsp --verbose -I ip 235.2.1.1:5004 -P pmt --service 1 --add-programinfo-id 0x43554549 --add-pid 1026/0x86 -P spliceinject --inject-interval 1000 --service 1 --files ./static/udp_scte/Melted_Test/splice.xml -O ip 238.1.1.1:9001
 
 Hey mate when i am using this command to create output multicast it has lot of errors in it. can you please help me with this command.
 
 
[image]

lelegard, 2024-09-27:
Hi @gauravv5,

The receiver is probably unable to absorb the bursts which were created by the `tsp` command. The receiver loses UDP datagrams, resulting in in video "corruption".

As [explained here](https://tsduck.io/download/docs/tsduck.html#tsp-reference), `tsp` is a generic framework. It works on offline TS files as well as real-time streams. Each plugin is fully independent from the others and ignore the constraints of the others. There are incompatibilities between performance, throughput, latency. Specific tuning may be necessary, case by case. By default, `tsp` tries to favor performance and to reduce its own overhead. This means using large buffers and minimum switching between the plugins' threads. The consequence is a higher latency with bursts.

That is appropriate for batch processing. In the case of a UDP-to-UDP, you must reduce the latency to avoid bursts.

I suggest to use
```
tsp --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 ...
    ...
    -O ip --enforce-burst ...
```

closed 2025-03-02 by github-actions[bot]