tsduck/tsduck #1520: [BUG] tscmp: option --search-reorder failed with error Aborted (core dumped)

nsaadi, 2024-09-13:
#### Bug description:
Using following cmd tscmp --continue --json-line --search-reorder stream_sent.ts stream_acq.ts
Output:
Packet 0 in /home/bgarnier/pytest-output/TitanTsOverIpInput/stream_sent.ts, missing 70 packets in /home/.......
tscmp: tscmp.cpp:393: void ts::FileToCompare::fillBuffer(): Assertion `_packet_index % _packets_buffer.size() == 0' failed.
Aborted (core dumped)

#### How to reproduce:
Systematic

#### Expected behavior:
No error

#### Errors and logs:
Packet 0 in /home/bgarnier/pytest-output/TitanTsOverIpInput/stream_sent.ts, missing 70 packets in /home/.......
tscmp: tscmp.cpp:393: void ts::FileToCompare::fillBuffer(): Assertion `_packet_index % _packets_buffer.size() == 0' failed.
Aborted (core dumped)

#### Environment:
 - OS: Linux
 - OS version: Ubuntu22.04
 - TSDuck full version: 3.35-3258 and 3.38-3822
 - Installation type: Official binary

#### Additional information:
ts used for testing added.
[Downloads.zip](https://github.com/user-attachments/files/16995629/Downloads.zip)

lelegard, 2024-09-13:
Hi @nsaadi 

Thank you for reporting this.
I just fixed it.

nsaadi, 2024-09-16:
Hi
Thank you for youy fix. Do you when new relase will be available?

lelegard, 2024-09-16:
There is no fixed plan. A new release is published every 3 or 4 months. The latest one was released on August 21st, so do not expect a new release before 2 or 3 months.

If you use Ubuntu 22, you may use the [latest nightly builds](https://tsduck.io/download/prerelease/). Otherwise, you can [rebuild from sources](https://tsduck.io/download/docs/tsduck-dev.html#unixbuild).

closed 2025-02-16 by github-actions[bot]