tsduck/tsduck #1522: [BUG]  Not possiple to run multiple tsp processes with different vatek devices.

Ravnholt7507, 2024-09-18:
#### Bug description:
Hi all and @lelegard 

I am currently using a setup where i have 2 TV's and 2 vatek modulators to send a transport stream to my tv's. Both of these vatek devices are connected to my laptop and when i list the devices using `tsvatek -a` i get the following output:
`Found 2 VATek devices
 - Device 0: .//4-0//vatek-usb
 - Device 1: .//4-0//vatek-usb`

I use the following tsp command: `tsp -I file generic-HbbTV-Teststream.ts -i -O vatek --device 0 --frequency 474000000 --modulation DVB-T --bandwidth 8`. This command works perfectly fine with both devices, when run by itself.

The problem happens when i try to start multiple tsp processes, by running the above tsp command with both device 0 and device 1.

What happens is that once i have run the command `tsp -I file generic-HbbTV-Teststream.ts -i -O vatek --device 0 --frequency 474000000 --modulation DVB-T --bandwidth 8` , it runs fine, and i then move on to the second device by running, `tsp -I file generic-HbbTV-Teststream.ts -i -O vatek --device 1 --frequency 474000000 --modulation DVB-T --bandwidth 8`. This is where the error happens and it tells me that it is unable to detect  any vatek devices. It seems like the first tsp process is locking both vatek devices for some reason.

#### Expected behavior:
I expected to be able to have 2 tsp processes running simultaneously, using 2 different vatek modulators. 

#### Errors and logs:
The error i get when i run the second tsp command is the following:
~~~
* Error: vatek: no VATek device found
~~~

#### Environment:
 - OS: Kubuntu
 - OS version: Ubuntu 22.04.4 
 - TSDuck full version: 3.38-3820
 - Installation type:  I build the one from github.

lelegard, 2024-09-19:
Hi @Ravnholt7507, thank you for reporting this.

Richie, @a83912a, could you help on this ?

Ravnholt7507, 2024-09-19:
Hi @lelegard and @a83912a ,
I'm wondering if this has anything to do with another issue i'm encountering (https://github.com/tsduck/tsduck/issues/1508), all though i'm not sure it's the same problem.

phil-net, 2024-09-24:
Hi Thierry,
Hi all,

I have encoutered the same as @Ravnholt7507:
- first tsp instance with vatek output lauch in background OK
- second tsp instance with vatek output lauch and quit immediatly NOK

Briefly, the issue seem related with with idx of vatek_device_open(hvatek_devices hdevices, int32_t idx, hvatek_chip* hchip) API.

Correct me if i ' wrong thierry but idx is device number value starting at 0 right?
To highlight this i use  sample.c  program  of vatek_sdk 2-3.06 which send file to modulator.
I have 4 modulators and i have changed program to put idx = 0,1,2,3 (initially 0 in the code) and the result is:
idx = 0 => OK - the file is modulated 
idx =1,2,3 => NOK - error - open device fail : -3

it's seems idx values are wrong are it is the wrong way to pass adapter number ie 0,1,2,3,4.
unfornatly there is no example in the sdk with passing device number as argument.

another issue maybe related with tsvatek -a  i would like i to mention is:

i have config with 4 modulators:
Expected behavior:
 tsvatek -a : 
 Found 4 VATek devices
 - Device 0: .//6-0//vatek-usb
 - Device 1: .//6-0//vatek-usb
 - Device 2: .//6-0//vatek-usb
 - Device 3: .//6-0//vatek-usb

I've got :
 Found 2 VATek devices
 - Device 0: .//6-0//vatek-usb
 - Device 1: .//6-0//vatek-usb

devices 2,3 not listed.

briefly It's because in tsVatekControl.cpp the code rely return nres value of vatek_device_list_enum(DEVICE_BUS_USB, service_transform, &hdevices) API to count the device number, but nres value returned is always 0 or 1 not 4.
If i in ugly way put i<4 in the loop later and compile i 've have now the 4 devices listed but the 2 lasts devices 2 and 3 are displayed as unknown might be .//6-0//vatek-usb.

hope this help, i'm ready for any additionnals tests if needed.

lelegard, 2024-09-24:
@Ravnholt7507, @phil-net, the `vatek` plugin is a code contribution from VATek. They developed and maintain the code. Let's wait for Richie (@a83912a) to provide a feedback on this one.

phil-net, 2024-09-25:
hi all, just to tell you that Vatek may release a sdk on next friday which support multiple devices

a83912a, 2024-09-26:
Hi All, due to some reasons, I am unable to perform a complete test. I have uploaded the latest VATek binary file. Please recompile TSDuck to apply the latest update. After compilation, kindly help check if the issue has been resolved. Thank you.

lelegard, 2024-09-26:
Thank you @a83912a for the new release.
@Ravnholt7507 and @phil-net, could you please confirm if the libvatek update fixes your problems, including #1508 ?

phil-net, 2024-09-26:
hi Thierry,
hi Richie,

put new 3.11, and recompile TSDUCK (v3.35-3258) and it doesn't fix:
- send first tsp instance in background, second instance quit immediatly (same bug)
- tsvatek -a  display now "No VATek device found" (i have 3 A3 devices connected on usb port).

@a83912, it could be very helpful for debug to have an updated version of SDK sample.c with A3 number (0,1,2,3..) pass as argument of command line.

phil-net, 2024-09-26:
SDK app_stream.c sorry

lelegard, 2024-09-27:
Hi Ritchie, @a83912a, I think that there some git issue with the the fixes you did in libvatek. They are not in the version 3.11. The only difference between 3.10 and 3.11 is a small compilation fix (due to latest version of gcc) for which I submitted a pull request in April. If you recently did a functional fix in libvatek, it is not in v3.11. Some commit wasn't pushed maybe.

a83912a, 2024-09-28:
Hi @lelegard I will check and fix the bug, then commit the correct source code to Git.

Hi @phil-net @Ravnholt7507 I have obtained two devices, and I will conduct a full test to confirm the bug is fixed before proceeding with the update.

lelegard, 2024-10-04:
@a83912a, please have a look at issue https://github.com/VisionAdvanceTechnologyInc/vatek_sdk_2/issues/9 that I just opened.

Ravnholt7507, 2024-10-09:
Hi @lelegard and @a83912a ,

It seems a fix was committed to the tsduck github, but when i pull the latest nightly build (`version 3.39-3889`), i still encounter the issue.

Has the issue been implemented in the nightly builds?

lelegard, 2024-10-09:
@Ravnholt7507, there are almost daily updates in the TSDuck code base. It's about 400,000 lines of code and many different areas may get updates. There was nothing recent about this issue.

Richie (@a83912a) said that this was an issue with `libvatek` and he needed some time to fix it. To see which version of `libvatek` is included into TSDuck:
~~~
$ tsp --version=vatek
libvatek version 3.10
~~~

As long as you see 3.10 or 3.11, there is no change. Richie's fix will be implemented in something higher than 3.11.

a83912a, 2024-10-09:
@Ravnholt7507 I apologize for the delay due to my busy schedule recently, which has caused the task to remain incomplete. Once the issue is fixed, I will upload a version higher than 3.11 and pull a new version of the TSDuck source code to update the tsvatek plugin. I will inform you when it's ready for functionality verification.

@lelegard Thank you for your suggestion. I will make the modifications based on your advice and complete the testing.

lelegard, 2024-10-22:
@phil-net, there is still an issue with libvatek (see issue https://github.com/VisionAdvanceTechnologyInc/vatek_sdk_2/issues/9), up to the point that an unfortunate side effect of its last version broke the compatibility and broke TSDuck builds. I had to temporarily disable VATek support in TSDuck until a definitive fix is available in libvatek.

If you want to use one VATek device, use TSDuck version 3.38. If you want to simultaneously use several VATek devices, wait for the libvatek fix. In intermediate versions, VATek support is completely disabled.

lelegard, 2024-10-31:
@Ravnholt7507, @phil-net, a new release of the libvatek and a fix in TSDuck have been provided by @a83912a.

Could you please test the new release from the repository and report here if your problems are fixed?

phil-net, 2024-11-04:
Hi Thierry, i have tested with last 3.12.1 - still the issue get "000d1ea3:error - cross_device_create - index 0"
@a83912a when can we expected a definitive fix for this -tks

a83912a, 2024-11-04:
Hi @phil-net , I tested on Linux using the latest TSDuck TS_COMMIT 3922 from GitHub, and no errors occurred. Could you please provide the environment you are using (OS, OS version, full TSDuck version) and the command you used so I can perform a detailed test? Thank you.

lelegard, 2024-11-04:
Adding to @a83912a's reply, the command to use is `tsversion --version=all`. It gives everything.

lelegard, 2024-11-09:
@phil-net, @Ravnholt7507, could you provide your feedback to @a83912a, including `tsversion --version=all` for the tested version?

This issue was opened two months ago and has generated a lot of work for @a83912a. So, we need to conclude.

Ravnholt7507, 2024-11-11:
Hi @lelegard and @a83912a,
Sorry i have not responded, i have been ill all last week.

I have now tested the newest version, and it seems the bug has been completely fixed. 

I have tested playing two different streams on 2 tv's with 2 vateks, and it works perfectly.

furthermore when the streams are playing, i can still see the devices with `tsvatek -a`, which i was not able to do before.

I therefor consider the bug fixed (but since it is an external bug i'll let you close the issue, when you consider the bug fixed)

For good measure, here is my `tsversion --version=all`:

`tsversion: TSDuck - The MPEG Transport Stream Toolkit - version 3.39-3940
Built Nov 11 2024 - 08:05:20
Using GCC 11.4.0, C++ std 2017.03
System: Ubuntu (Ubuntu 22.04.5 LTS), on Intel x86-64, 64-bit, little-endian, page size: 4096 bytes
Acceleration: CRC32: no
Bitrate: 64-bit floating-point value
Dektec: DTAPI: 6.6.0.233
VATek: libvatek version 3.12.10
Cryptographic library: OpenSSL 3.0.2 (CPUINFO: OPENSSL_ia32cap=0x7ffaf3ffffebffff:0x98c027bc239c27eb)
Web library: libcurl: 7.81.0, ssl: OpenSSL/3.0.2, libz: 1.2.11
SRT library: libsrt version 1.4.4
RIST library: This version of TSDuck was compiled without RIST support
`

closed 2024-11-11 by lelegard