tsduck/tsduck #1523: [FEAT] Add inject TDT and TOT like eitinject  (tdtinject)

TVforME, 2024-09-23:
#### Is your feature request related to a problem?
No, feature  request
#### Proposed solution:
Hi @lelegard,

Has there been any thought on creating a plugin similar to eitinject in inject eit xlm into the ts.

I understand, if the transport doesn't contain a TDT or TOT there is a  need to create a TDT.xml/bin and TOT.xml/bin and inject these before using timeref to modify the time and dates from there on.

It would be nice to have a timeinject to create or insert a TDT and TOT to the transport then use timeref to keep the TDT updated from there on.

#### Alternatives:
No alternatives, ffmpeg doesn't add TDT or TOT as yet. A TDT and TOT is optional for DVB however, some TS analysers complain it missing from the stream.   To use EIT, time needs to be available to the end device TV/STB etc to correct sync EIT events. (EPG)

#### Additional information:
Is there a way to use plugins without having to create these xml or bin files? currently, I need to create fresh XML for my NIT and TDT and TOT and patch SDT in code then inject from external files.  Be nice to be able to do all if this an inline process to make integration less reliant on external files and easy to do with string passing from a config.

eg,
timeinject  -t  13:06:14 -o 10/10/2026  -d 660  -c AUS

-t = time of day etc  13:06:14  
-d offset  date  23/04/2024 -
-o offset minutes 660
-c country AUS

DONE for both TDT and TOT

TVforME, 2024-09-23:
> #### Is your feature request related to a problem?
> No, Just a feature request
> 
> #### Proposed solution:
> Hi @lelegard,
> 
> Has there been any thought on creating a plugin similar to eitinject in inject eit xlm into the ts.
> 
> I understand, if the transport doesn't contain a TDT or TOT there is a need to create a TDT.xml/bin and TOT.xml/bin and inject these before using timeref to modify the time and dates from there on.
> 
> It would be nice to have a timeinject to create or insert a TDT and TOT to the transport then use timeref to keep the TDT updated from there on.
> 
> #### Alternatives:
> No alternatives, ffmpeg doesn't add TDT or TOT as yet. A TDT and TOT is optional for DVB however, some TS analysers complain it missing from the stream. To use EIT, time needs to be available to the end device TV/STB etc to correct sync EIT events. (EPG)
> 
> #### Additional information:
> Is there a way to use plugins without having to create these xml or bin files? currently, I need to create fresh XML for my NIT and TDT and TOT and patch SDT in code then inject from external files. Be nice to be able to do all if this an inline process to make integration less reliant on external files and easy to do with string passing from a config.
> 
> eg, timeinject -t 13:06:14 -o 10/10/2026 -d 660 -c AUS
> 
> -t = time of day etc 13:06:14 -d offset date 23/04/2024 - -o offset minutes 660 -c country AUS
> 
> DONE for both TDT and TOT

lelegard, 2024-09-23:
Hi @TVforME 

There is no need to create an external XML or binary file to inject a table. You can use [inline XML](https://tsduck.io/download/docs/tsduck.html#inline-xml).

Working example:
~~~
$ tsp --bitrate 1000000 \
      -I null \
      -P regulate \
      -P inject "<?xml version='1.0' encoding='UTF-8'?><tsduck><TDT UTC_time='2000-01-01 00:00:00'/></tsduck>" --pid 20 --bitrate 150 --stuffing \
      -P timeref --system \
      -P tables -x 3 \
      -O drop

* TDT, TID 0x70 (112), PID 0x0014 (20)
  Short section, total size: 8 bytes
  - Section 0:
    UTC time: 2024/09/23 08:31:14

* TDT, TID 0x70 (112), PID 0x0014 (20)
  Short section, total size: 8 bytes
  - Section 0:
    UTC time: 2024/09/23 08:31:24

* TDT, TID 0x70 (112), PID 0x0014 (20)
  Short section, total size: 8 bytes
  - Section 0:
    UTC time: 2024/09/23 08:31:34

$ 
~~~

Creating a new plugin would be redundant.

TVforME, 2024-10-12:
Thankyou @lelegard, I'll take your explanation and work with using text.

Does this method work with EIT inject?

lelegard, 2024-10-12:
> Does this method work with EIT inject?

What do you mean by that? Using direct inline XML text works everywhere a XML file name can be specified. However, in the case of EIT injection, you rarely inject one or two event. Events are renewed on a regular basis. The idea is to specify names of XML files, including wildcard if necessary, and renew the content of these files regularly. Or create new files which match the wildcard specification. The plugin `eitinject` polls the creation of new files or modifications of existing files and reload them, continuously.

Specifying inline XML text in commands line is only good for scripts which inject constant tables on a given cycle. Or tables which are then updated by a subsequent plugin, as in the example above.

EIT injection is a different use case.

SonnyWalkman, 2025-02-10:
Hi @lelegard.
Sorry to reply late,
What I was asking since TDT and TOT arecases to use inline XML  switch is small in content.
I assume the same is possible with `eitinject ` to script the EPG metadata in as XML text without needing to save as a file and have `eitinject` poll for a change.

Do you recommend the later save as XML file OR use `    -P inject "<?xml version='1.0' encoding='UTF-8'?><tsduck><EIT blah blah.. /></tsduck>" --pid 18 --bitrate 3000 --stuffing \`

OR do I save EIT as XML to file and be done with it?

lelegard, 2025-02-10:
EIT cannot be used as other tables. EIT's are sparse sections, almost never complete tables. Therefore, it is not possible to create an EIT in XML and inject it directly (section numbers are not contiguous and incomplete by design).

This is why `eitinject` does not read EIT's, it loads _events_. In practice, you specify events inside an EIT container for convenience, just because the format existed. Keep in mind that any EIT you set in an input file for `eitinject` will _never_ be injected as is. The events are extracted, stored, and later new EIT's will be generated with the events inside them.

Additionally, an EPG is usually large and evolves over time. This is why the option is defined as `--files 'file-wildcard'`. The string is not a file which is directly loaded (in which case inline XML would work). The string is a wildcard which is explicitly resolved every time. If you use inline XML, `glob()` will not understand.

closed 2025-05-01 by TVforME