tsduck/tsduck #1527: [Q] How do I convert a SCTE message and reinsert the converted SCTE message to my output stream

arhankundu99, 2024-09-30:
Hi, 
For context: I am using ffmpeg libraries to read an incoming rtp stream which has 3 streams: video, audio and SCTE-35 stream. Now my goal is to replace the SCTE messages (For eg., If the scte message is "SPLICE_INSERT", convert it into "SPLICE_NULL") and reinsert the new SCTE message in my output stream. Other streams such as video and audio will remain same in the output stream.

Can anyone share code examples on how to do this?

Basically I want this function implementation

```cpp
std::shared_ptr<AVPacket> SCTEHandler::processAndGenerateSCTEPacket(const AVPacket &avPacket) {
        // read the scte message

       // if the splice command type is splice insert, then convert into splice null

       // return the converted SCTE message

}
```

I also want the other properties such as pts, dts etc to be same so that I don't face any packet loss in my output stream.

Thank you!

lelegard, 2024-09-30:
Hi @arhankundu99 

This is TSDuck's forum, not the ffmpeg one. We cannot help on ffmpeg library. I don't know what an `AVPacket` contains. This is a ffmpeg library type.

TSDuck works on transport stream layer while ffmpeg works on A/V streams. This is why the two tools are complementary.

SCTE-35 sections are carried on their own PID. What you want to do could be quite easy if you implement a TSDuck plugin. You only have to replace SCTE-35 sections in their PID and that's all. All the video and audio containing PCR, PTS, DTS are in other PID's that you simply don't modify.

arhankundu99, 2024-09-30:
Hii
AVpacket contains hex value of scte 35 message. I just wanted to know if you could provide me any code examples of converting the message using tsduck library.

Thank you and sorry for the inconvenience.

lelegard, 2024-09-30:
> AVpacket contains hex value of scte 35 message.

I assume you mean raw binary content, not an ascii string of hex digits, right?

I also assume that this raw binary content is the exact content of a SCTE-35 section, header included, without encapsulation. If this is not the case, then you have to adapt.

Using TSDuck library, you can decode this raw binary content as a `SpliceInformationTable` instance, modify the object and reserialize it as raw content. How to reinsert the raw binary content in replacement of the input SCTE-35 section is a matter of ffmpeg library.

The code would look like this (untested):
~~~
ts::DuckContext duck;
ts::BinaryTable table;
table.addNewSection(section_data_address, section_data_size);
ts::SpliceInformationTable scte35(duck, table);
if (scte35.isValid() && scte35.splice_command_type == ts::SPLICE_INSERT) {
    scte35.splice_command_type = ts::SPLICE_NULL;
    scte35.descs.clear();
    if (scte35.serialize(duck, table) && table.isValid() && table.sectionCount() == 1) {
        const ts::SectionPtr sec(table.sectionAt(0));
        // raw section content at sec->content() / sec->size()
        // to replace the previous SCTE-35 section,
        // according to what ffmpeg can do in terms of replacement
    }
}
~~~

lelegard, 2024-10-06:
Also note that, with the latest state of the TSDuck repo (version 3.39-3885 or higher), you can achieve this without writing code, only using the plugin `sections` and patch files.

Consider the following XML patch file.

~~~
<tsduck>
  <splice_information_table x-undefine="NULLIFY">
    <splice_insert x-define="NULLIFY" x-node="delete"/>
    <splice_null x-condition="NULLIFY" x-node="add"/>
  </splice_information_table>
</tsduck>
~~~

In all `<splice_information_table>`, it replaces any `<splice_insert>` with a `<splice_null>`.

See [this section of the user guide](https://tsduck.io/download/docs/tsduck.html#xml-patch) for more details on XML patch files.

Sample usage using a TS file from the [TSDuck test suite](https://github.com/tsduck/tsduck-test). This file contains _splice_information_table_ with either _splice_insert_ and _splice_null_ commands. The tables are displayed like this:

~~~
tsp -I file input/test-149.ts \
    -P tables --pid 0x0632 \
    -O drop
~~~

To replace all _splice_insert_ with _splice_null_ commands, using the XML patch file above and display the result:
~~~
tsp -I file input/test-149.ts \
    -P sections --pid 0x0632 --patch-xml p.xml \
    -P tables --pid 0x0632 \
    -O drop
~~~

closed 2025-03-23 by github-actions[bot]