tsduck/tsduck #1528: [Q] Unchecked PCR packets with pcrverify

lgu-appear, 2024-09-30:
Hello there!

I am using TSDuck as follow:
``` tsp -I pcap .\a_pcap_file.pcap -P pcrverify -O drop```

And I get 
```pcrverify: 27 PCR OK, 0 with jitter > 27,000 (1,000 micro-seconds), 33 unchecked```

Is there a way to know why 33 PCR were unchecked?I also tried with `pcrverify -j 1` and didn't change anything. I looked at my .pcap file and I only see one RTP stream, I have exported my file to .ts and looked at it with DVB Inspector and see 60 PCR packets that look healthy. Any idea of what is happening?

Thanks in advance

lgu-appear, 2024-10-01:
Hello @lelegard,

Indeed, using `--input-synchronous` solved it! So it means that it is hard to know the bitrate and we can't check the PCR due to that reason? So what it does is that it compares the input timestamp (recording time, RTP) with the PCR and checks that it aligns? Let me know if my understanding is wrong. Which input timestamp does it chose by the way?

lelegard, 2024-10-01:
> Indeed, using --input-synchronous solved it! So it means that it is hard to know the bitrate and we can't check the PCR due to that reason?

Yes, checking the validity of PCR's in a VBR stream without external clock reference is impossible by definition.

Additionally, it is impossible to check if a recorded stream is VBR or CBR. Therefore, checking the validity of PCR's is never a magical operation. It either relies on default assumptions or directives from the user. If you don't provide any directive and the tool's default assumptions match the reality of a specific use case, you have the impression that the tool works. And if they don't, you have the impression that the tools does not work. However, these are just impressions, not a technical base.

BTW, this also means that after extracting the TS as a file of TS packets (and it is a VBR one), the PCR cannot look "healthy". They don't look anything at all. Nobody can evaluate if the PCR as correct or not.

> So what it does is that it compares the input timestamp (recording time, RTP) with the PCR and checks that it aligns? Let me know if my understanding is wrong. Which input timestamp does it chose by the way?

That is correct. In the case of the `pcap` input plugin, the time source is the PCAP capture timestamp. If the difference in time between two PCR's of the same PID is exactly equal to the difference of the PCAP capture timestamps of their respective UDP datagrams, there is no jitter. If they differ by a certain amount, this is the jitter value.

Each input plugin chooses its own most reliable timestamp source. In the case of the `pcap` plugin, this is the capture timestamp because it is the real clock at the time of the capture. In the case of [the `ip` plugin](https://tsduck.io/download/docs/tsduck.html#_ip_input) (realtime capture), several timestamp sources may be available, depending on the operating system and received packets. The user may choose in which order these sources are used using option `--timestamp-priority`.

lgu-appear, 2024-10-01:
The recording is a CBR stream actually, with null packets stuffing and so on

Thanks a lot for the detailed explanations! Also, I can see the --timestamp-priority in the `pcap` plugin, so I guess I can also choose my preference there!

You are doing a great work with this software and the maintainance you do with it, that's amazing!

lelegard, 2024-10-01:
> Also, I can see the --timestamp-priority in the pcap plugin, so I guess I can also choose my preference there!

Yes, you are right, I forgot this one, you may choose between PCAP and RTP timestamp (if RTP is used)

closed 2024-10-04 by lgu-appear