tsduck/tsduck #1529: [FEAT] Add a VLAN flag to the pcap plugin to ignore it

lgu-appear, 2024-10-01:
#### Is your feature request related to a problem?
While parsing a pcap with the `pcap` plugin, nothing comes out. I checked with `tspcap` command and I get that result:
```File summary:
  Total packets in file: 380,355
  Total IPv4 packets:    0
  File size:             528,692,084 bytes
  Total packets size:    522,606,396 bytes
  Total IPv4 size:       0 bytes

Filtered packets summary:
  Packets:               0
  Packets size:          0
  Payload data size:     0```

The issue is because I have a VLAN tag in the packet, confirmed with Wireshark. That is at least my guess for the root cause of the problem.

#### Proposed solution:
Add a --remove-vlan flag or something similar to both `pcap` plugin and `tspcap` command to remove the VLAN tag or maybe add a warning that a VLAN tag was detected and that does not support parsing them, so they will be removed by default. This way, the user knows that there might be more consideration needed to fully use their pcap recording.

lelegard, 2024-10-01:
Could you please provide a sample pcap file?
Not necessarily a big one, just to reproduce that behaviour.

lgu-appear, 2024-10-01:
Sure, I will do so ASAP!

lelegard, 2024-10-01:
I would like to get a sample file anyway, in order to add it in the test suite and make sure no regression appears later on VLAN support in pcap files.

lgu-appear, 2024-10-01:
I'll provide you one tomorrow during the day

lgu-appear, 2024-10-02:
Hello @lelegard,
I couldn't drop the file in here, so here is a Swisstransfer link instead: https://www.swisstransfer.com/d/32182a3a-c91a-431f-af19-c39f907e617a

Let me know if there is anything else I can do!

lgu-appear, 2024-10-02:
I built locally and I confirm that it works 🎉

lelegard, 2024-10-02:
Hi @lgu-appear 

I confirmed it worked on my side as well with your file. I have used it (only the first 16 IP packets) to create test-162 in the non-regression test suite.

Since we are now able to detect VLAN's (including nested VLAN using IEEE 802.1ad and 802.1ah), it would have been a pity to ignore them. Therefore, I added an option `--vlan-id` to filter packets from the specified VLAN only. By default, without this option, all VLAN's are used (meaning that the VLAN ids are skipped and ignored). When `--vlan-id` is specified, only packets from that VLAN are used.

You need to pull and rebuild to get that new option.

lgu-appear, 2024-10-02:
I saw your commit on master this morning so I built it with that latest version. Works like a charm, amazing job!

closed 2024-10-04 by lgu-appear