tsduck/tsduck #1532: [Q] extend an existing transport stream

ToniTram81, 2024-10-07:
I want to extend an existing transport stream 239.0.0.1:5000 (including video and audio) with teletext from another TS 239.0.0.2:5000 (change PID 100 to PID 60) and EIT also from another TS 239.0.0.3:5000 (PID 18).
Is „-P merge“ the right way for that?
Will the tables be adjusted automatically or do I have to do this manually with replaces (xml injects)?

lelegard, 2024-10-07:
Hi @ToniTram81 

By design, each plugin does a very specific task, nothing more. The plugin `merge` only merges raw TS packets from a command output into the main transport stream, as those packets arrive from the the external command, nothing more.

Important: "inserting packets" means replacing null packets in the main TS (see TSDuck user's guide). Therefore, in practice, "inserting as packets arrive" means "buffering packets and inserting them when null packets arrive in the main TS". So, 1) you need enough null packets (aka "free space") in the main TS to accommodate the stream to merge, 2) if these null packets are scarce you may have some delay in the insertion.

Also, inserting packets as they arrive and when null packets are available means that there is no timing synchronization between the elementary streams from the main TS and the merge TS. If you merge data, this is not an issue. If you merge components of a service which need synchronization with video and audio, it won't work well. To do this synchronization, you need a MUX and TSDuck is not a MUX.

About Teletext, it depends on what's inside these Teletext streams. If this is only data pages, that should be fine. It this is subtitles, then you need a MUX to synchronize with audio.

About the signalization, you have to update it in another plugin (the plugin `merge` only inserts packets, it has no way to know that these packets carry Teletext - or any other type of data - and in which service of the main TS they should be included if necessary). The plugin `inject` injects complete tables. It creates or replaces a PID. If you simply want to add in a PMT a stream component with a Teletext descriptor, it is easier to use the plugin `pmt` and use an XML patch file to add the component.

About the EIT's, it depends whether the main TS already contains an EIT PID or not. If the main TS does not contain any EIT PID, then `merge` is fine. It simply merges a complete EIT PID. However, if the main TS already contains an EIT PID, then you must merge the EIT sections from the two TS and recreate a new EIT PID. The best way to do this is 1) rename the EIT PID in the merge stream before insertion, 2) after plugin `merge`, use plugin `sections` to merge the EIT sections from the two PID and replace them with one common EIT PID.

ToniTram81, 2024-10-12:
Thank you Thierry, do you have a code example for any merge? Is this only in two or more separate command lines possible or also as one-liner? The diagram at "merging and forking" (user guide page 105) is clear, but a code example would be great...

ToniTram81, 2024-10-14:
I want to merge the teletext PID 100 from 232.144.130.1  into another TS. There it should have the PID 25. How can I only pic PID 100 and rename it to 25?
-P merge "tsp -I ip 232.144.130.1:6666 -l 10.10.10.10 ..."
Thanks a lot!

lelegard, 2024-10-14:
> How can I only pic PID 100 and rename it to 25?

~~~
-P merge "tsp -I ip 232.144.130.1:6666 -l 10.10.10.10 -P filter -p 100 -P remap -n 100=25"
~~~

When you look at something to do, please [browse through the list of plugins](https://tsduck.io/download/docs/tsduck.html#chap-plugins) and their documentation, just following the table of plugins.

closed 2025-03-23 by github-actions[bot]